MarlinFirmware/Marlin #18147: DISTINCT_E_FACTOR caused Build error

sipqis, 2020-05-29:
Hi,

I've got the following build error after modified the DISTINCT_E_FACTOR in the latest 2.0.5.3 fw:

In file included from sketch\src\lcd\menu\menu_advanced.cpp:31:0:
sketch\src\lcd\menu\menu_advanced.cpp: In lambda function:
sketch\src\lcd\menu\menu_advanced.cpp:488:113: error: '_planner_refresh_e_positioning' was not declared in this scope

       EDIT_ITEM_FAST_N(float51, n, MSG_EN_STEPS, &planner.settings.axis_steps_per_mm[E_AXIS_N(n)], 5, 9999, []{ _planner_refresh_e_positioning(MenuItemBase::itemIndex); });
                                                                                                                 ^
sketch\src\lcd\menu\menu.h:387:39: note: in definition of macro '_MENU_INNER_P'

     MenuItem_##TYPE::action(plabel, ##V);                     \
                                       ^
sketch\src\lcd\menu\menu.h:476:52: note: in expansion of macro '_MENU_ITEM_N_P'

 #define EDIT_ITEM_FAST_N_P(TYPE, N, PLABEL, V...)  _MENU_ITEM_N_P(TYPE, N, true, PLABEL, ##V)
                                                    ^~~~~~~~~~~~~~
sketch\src\lcd\menu\menu.h:477:48: note: in expansion of macro 'EDIT_ITEM_FAST_N_P'

 #define EDIT_ITEM_FAST_N(TYPE, N, LABEL, V...) EDIT_ITEM_FAST_N_P(TYPE, N, GET_TEXT(LABEL), ##V)
                                                ^~~~~~~~~~~~~~~~~~
sketch\src\lcd\menu\menu_advanced.cpp:488:7: note: in expansion of macro 'EDIT_ITEM_FAST_N'

       EDIT_ITEM_FAST_N(float51, n, MSG_EN_STEPS, &planner.settings.axis_steps_per_mm[E_AXIS_N(n)], 5, 9999, []{ _planner_refresh_e_positioning(MenuItemBase::itemIndex); });
       ^~~~~~~~~~~~~~~~
sketch\src\lcd\menu\menu_advanced.cpp: In lambda function:
sketch\src\lcd\menu\menu_advanced.cpp:488:113: error: '_planner_refresh_e_positioning' was not declared in this scope

       EDIT_ITEM_FAST_N(float51, n, MSG_EN_STEPS, &planner.settings.axis_steps_per_mm[E_AXIS_N(n)], 5, 9999, []{ _planner_refresh_e_positioning(MenuItemBase::itemIndex); });
                                                                                                                 ^
sketch\src\lcd\menu\menu.h:392:58: note: in definition of macro '_MENU_INNER_P'

       (encoderLine == _thisItemNr, _lcdLineNr, plabel, ##V);  \
                                                          ^
sketch\src\lcd\menu\menu.h:476:52: note: in expansion of macro '_MENU_ITEM_N_P'
 #define EDIT_ITEM_FAST_N_P(TYPE, N, PLABEL, V...)  _MENU_ITEM_N_P(TYPE, N, true, PLABEL, ##V)
                                                    ^~~~~~~~~~~~~~
sketch\src\lcd\menu\menu.h:477:48: note: in expansion of macro 'EDIT_ITEM_FAST_N_P'
 #define EDIT_ITEM_FAST_N(TYPE, N, LABEL, V...) EDIT_ITEM_FAST_N_P(TYPE, N, GET_TEXT(LABEL), ##V)
                                                ^~~~~~~~~~~~~~~~~~
sketch\src\lcd\menu\menu_advanced.cpp:488:7: note: in expansion of macro 'EDIT_ITEM_FAST_N'

       EDIT_ITEM_FAST_N(float51, n, MSG_EN_STEPS, &planner.settings.axis_steps_per_mm[E_AXIS_N(n)], 5, 9999, []{ _planner_refresh_e_positioning(MenuItemBase::itemIndex); });
       ^~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

The modifications in the configuration.h was performed and the 2nd extruders settings was added to the array of values (as the attached picture shows). 
The error message refers a lambda operator in files belongs to the LCD display/controller I am using (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER).

Does anyone any idea what can I do?

Many thanks in advance!
[image]

sjasonsmith, 2020-06-20:
@sipqis can you test this from the bugfix-2.0.x branch, and report whether the issue exists there as well?

sipqis, 2020-06-21:
@sjasonsmith I found the issue weeks ago - there was not documented, or only I did not found, setting to be changed in case of enable the DISTINCT_E_FACTORS and have a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. Found in the menu_advanced.cpp that the SLIM_LCD_MENUS must be disable to declare the "_planner_refresh_e_positioning" variable.
[image]
 So just commented out the "#define SLIM_LCD_MENUS" line in the Configuration.h and this resolved my problem and was able to build and upload the firmware.
[image]
 And the multiple extruders settings just displayed well on the smart controller. So I think this dependency should be added to the configuration guide - I know this is a conditional thing that you should have distinct extruders and REPRAP discount full graph smart controller but - to highlight this configuration and refer the SLIM_LCD_MENUS definition removal.

boelle, 2020-06-23:
So to be clear, the error is still in here? [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle