MarlinFirmware/Marlin #18151: [FR] (Update EEPROM with MBL active state [on/off])

xaviervilla, 2020-05-29:
### Description

Currently, when mesh bed leveling is enabled and active, rebooting the printer defaults leveling_active to OFF. The feature being requested is that this status variable be saved to EEPROM when M500 is called, that way when the printer reboots, mesh bed leveling is still active. This allows user to always print with mesh bed leveling without needing to call M420 S1 in the G code (assuming user has RESTORE_LEVELING_AFTER_G28 is enabled).

### Feature Workflow

1. User creates bed mesh for the first time with their desired method.
2. User calls M420 S1 to enable bed leveling.
3. User calls M500 to store the mesh and also the leveling_active variable to EEPROM. 
4. User reboots the printer.
5. Upon checking mesh with G29, the mesh exists, ** and bed leveling is active. **

#### Additional Information

I actually think I am close with my own implementation of this feature. I studied the code a lot yesterday and I'm certain this would need to be implemented in `marlin/module/configuration_store.cpp`. I believe the variable that needs to be stored is `planner.leveling_active`. I added some code to store and to load near the mesh bed leveling features for storing and loading the mesh. I kept the order for loading and storing. However I keep getting 
```
EEPROM datasize error.
Index: 691 Size: 690
Hardcoded Default Settings Loaded
```
I'm not sure if I need to manually update the size or index, it doesn't appear that the other implementations in the file are doing that. I tried adjusting the offset at the top of the file but this didn't help. If somebody could advise I'm willing to try again and do a pull request to the bugfix 2.0 branch.

Thank you!

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Marlor, 2020-05-30:
The mesh is gone after reboot, so the state would always be off.
Have a look at UBL https://marlinfw.org/docs/gcode/G029-ubl.html insted, this saves the Mesh and the on/off state in EEPROM.

borland1, 2020-05-31:
For Manual Mesh Bed Leveling, I am interested in why MBL is not active from power-up?

Could add g-codes to Start from slicer to set active state ON, but there is also a Marlin 'Startup Commands' feature that can be set in `Configuration_adv.h` which allows sending g-code. 

```
/**
 * Startup commands
 *
 * Execute certain G-code commands immediately after power-on.
 */
//#define STARTUP_COMMANDS "M17 Z"
```

xaviervilla, 2020-06-01:
> The mesh is gone after reboot, so the state would always be off.
> Have a look at UBL https://marlinfw.org/docs/gcode/G029-ubl.html insted, this saves the Mesh and the on/off state in EEPROM.

I'm confused by "mesh is gone after reboot so state is off".

I'm already using my ender 3 with my pull request and I can confirm it always saves the mesh and keeps mesh bed leveling on and active after many reboots.

xaviervilla, 2020-06-01:
> For Manual Mesh Bed Leveling, I am interested in why MBL is not active from power-up?
>
> Could add g-codes to Start from slicer to set active state ON, but there is also a Marlin 'Startup Commands' [...]

Nice, I didn't know about this. So theoretically one could add M420 S1 to to this section and they would always have MBL enabled at boot?

Marlor, 2020-06-01:
UBL is the feature you are asking for:
one time high resolution (up to 15x15) probing of the bed 
save the mesh in eeprom
never think about your bed ever again

xaviervilla, 2020-06-02:
> UBL is the feature you are asking for:
> one time high resolution (up to 15x15) probing of the bed
> save the mesh in eeprom
> never think about your bed ever again

Wow, I wish this was easier to discover. Thank you, clearly I had a fundamental misunderstanding of the purpose of UBL. I thought a z-probe was needed for UBL.

closed 2020-06-02 by xaviervilla