MarlinFirmware/Marlin #18153: [BUG] Inverted extruder on second print after power on and stop

Exalik, 2020-05-29:
### Bug Description

The extruder is inverted (purge line) on the second print after a stop of the first.

### My Configurations

[Marlin Bugfix-2.0.x Config.zip](https://github.com/MarlinFirmware/Marlin/files/4701258/Marlin.Bugfix-2.0.x.Config.zip)

### Steps to Reproduce

1- Launch a print with purge line or purge bucket
2- Stop it at the end of purge
3- Launch again the print
4- Look the extruder turning on the opposite way

**Expected behavior:** The extruder should not unload the filament

**Actual behavior:** The extruder unload the filament

#### Additional Information

ellensp, 2020-05-30:
Not official marlin. 
You have to get help from one of the following.
// Youtube             "https://youtube.com/verta"
// Discord             "https://discord.gg/brq79WQ"
// Forum               "http://www.vertshobbies.com/" 
// GitHub              "https://github.com/Vertabreak/Marlin"
// Thingivese          "https://www.thingiverse.com/Vertabreaker"
// Patreon             "https://www.patreon.com/vertabreaker" 
// Buy me a coffee     "https://ko-fi.com/vertabreaker"

Exalik, 2020-05-30:
Hello ellensp,

Thank's for your answer.

It is the official Marlin Bugfix-2.0.x The configurations files have just been optimized for some simplifications but the code has not been modified.

Regards,

Olivier

thisiskeithb, 2020-05-30:
> It is the official Marlin Bugfix-2.0.x 

Sure isn't.

If we supported every fork of Marlin, slight config changes or not, this issue queue would be way longer than it is now. @ellensp already pasted the support links for @Vertabreak's firmware, so please visit those or download a clean copy of [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) (not a fork) to see if your issue persists.

Exalik, 2020-05-31:
Hello,

The source code used for building have been downloaded from the Marlin Bugfix-2.0.x
Only the configuration.h, configuration_adv.h, and StartHere.h are from @Vertabreak 's firmware. I could remake all the configuration but I am sure it won't solve this issue as I use theses configurations files since many versions without this issue. I had just to make some adaptations cause of upgrades but I know what I do and it was on a previous version.

You can make the choice to not help me but it is your choice...

Regards,

Olivier

thisiskeithb, 2020-05-31:
Please close the issue and post in the support links provided above.

You’re using a fork of Marlin/mixing and matching configs from other sources when you’ve been asked to start with a clean copy from this repo (this includes configs).

There simply isn’t enough resources to support forks or code from other repos.

Exalik, 2020-05-31:
Ok, I am going to do it

closed 2020-05-31 by Exalik