MarlinFirmware/Marlin #18155: [BUG] Buttons in print menu Anet A8

chepo92, 2020-05-30:
### Bug Description
Printer: Anet A8 Stock board
Buttons work correctly except in menu with horizontal options (eg. Cancel print confirmation, start print confirmation): to move the cursor to the option to to right you have to press UP button, and no button make the cursor move left, LEFT button makes it go back to menu (kind of a cancel)

### My Configurations
[Marlin 5.3.zip](https://github.com/MarlinFirmware/Marlin/files/4704849/Marlin.5.3.zip)

[Marlin bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/4704852/Marlin.bugfix.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Go to print menu confirmation 
2. try to move cursor to "print"
3. 
**Expected behavior:**  RIGHT button moves right

**Actual behavior:** UP button, moves to the left

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

drackp2m, 2020-05-30:
You can try to comment `SD_MENU_CONFIRM_START` in `Configuration_adv.h` while you wait for the error to be fixed..

chepo92, 2020-05-31:
I know but I want to help, and learn and maybe as documentation for the working of marlin (and keep it in the same repo.
As far as i read the code, `SD_MENU_CONFIRM_START` in  `Configuration_adv.h`  creates a MenuItem_confirm::select_screen()

```
      #if ENABLED(SD_MENU_CONFIRM_START)
        MenuItem_submenu::action(pstr, []{
          char * const longest = card.longest_filename();
          char buffer[strlen(longest) + 2];
          buffer[0] = ' ';
          strcpy(buffer + 1, longest);
          MenuItem_confirm::select_screen(
            GET_TEXT(MSG_BUTTON_PRINT), GET_TEXT(MSG_BUTTON_CANCEL),
            sdcard_start_selected_file, ui.goto_previous_screen,
            GET_TEXT(MSG_START_PRINT), buffer, PSTR("?")
          );
        });
```

This gets to `menu.cpp` file 

```
void MenuItem_confirm::select_screen(PGM_P const yes, PGM_P const no, selectFunc_t yesFunc, selectFunc_t noFunc, PGM_P const pref, const char * const string/*=nullptr*/, PGM_P const suff/*=nullptr*/) {
  const bool ui_selection = ui.update_selection(), got_click = ui.use_click();
  if (got_click || ui.should_draw()) {
    draw_select_screen(yes, no, ui_selection, pref, string, suff);
    if (got_click) { ui_selection ? yesFunc() : noFunc(); }
  }
}
```

assuming the problem is with `ui.update_selection()`, which is defined as

```
bool MarlinUI::update_selection() {
  encoder_direction_select();
  if (encoderPosition) {
    selection = int16_t(encoderPosition) > 0;
    encoderPosition = 0;
  }
  return selection;
}
```

I don't thinks it is assuming the LCD has an encoder? or is it?

boelle, 2020-06-21:
@chepo92 still an issue?

chepo92, 2020-06-21:
yes

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle