MarlinFirmware/Marlin #18156: ANET ET4 DISPLAY ST7789V SUPPORT

0NEHITKILL, 2020-05-30:
**ANET ET4 DISPLAY SUPPORT I DON'T KNOW HOW TO MAKE IT WORK COULD HELP ME**
how is it connected - https://imgur.com/a/CmGcxwR
everything is blank and nothing happens already reversing the connectors
skr 1.3
MARLIN 2.X BUGFIX
**### Sorry Translate ifrom brazil**
//
// Touch UI for the FTDI Embedded Video Engine (EVE)
//
#if ENABLED(TOUCH_UI_FTDI_EVE)
  // Display board used
  #define LCD_FTDI_VM800B35A        // FTDI 3.5" with FT800 (320x240)
  //#define LCD_4DSYSTEMS_4DLCD_FT843 // 4D Systems 4.3" (480x272)
  //#define LCD_HAOYU_FT800CB         // Haoyu with 4.3" or 5" (480x272)
  //#define LCD_HAOYU_FT810CB         // Haoyu with 5" (800x480)
  //#define LCD_ALEPHOBJECTS_CLCD_UI  // Aleph Objects Color LCD UI
  //#define LCD_FYSETC_TFT81050       // FYSETC with 5" (800x480)

  // Correct the resolution if not using the stock TFT panel.
  #define TOUCH_UI_320x240
  //#define TOUCH_UI_480x272
  //#define TOUCH_UI_800x480

  // Mappings for boards with a standard RepRapDiscount Display connector
  //#define AO_EXP1_PINMAP      // AlephObjects CLCD UI EXP1 mapping
  //#define AO_EXP2_PINMAP      // AlephObjects CLCD UI EXP2 mapping
  //#define CR10_TFT_PINMAP     // Rudolph Riedel's CR10 pin mapping
  //#define S6_TFT_PINMAP       // FYSETC S6 pin mapping

  //#define OTHER_PIN_LAYOUT  // Define pins manually below
  #if ENABLED(OTHER_PIN_LAYOUT)
    // Pins for CS and MOD_RESET (PD) must be chosen
    #define CLCD_MOD_RESET  9
    #define CLCD_SPI_CS    10

    // If using software SPI, specify pins for SCLK, MOSI, MISO
    #define CLCD_USE_SOFT_SPI
    #if ENABLED(CLCD_USE_SOFT_SPI)
      #define CLCD_SOFT_SPI_MOSI P1_20
      #define CLCD_SOFT_SPI_MISO P1_19
      #define CLCD_SOFT_SPI_SCLK P1_21

  // UTF8 processing and rendering.
  // Unsupported characters are shown as '?'.
  #define TOUCH_UI_USE_UTF8
  #if ENABLED(TOUCH_UI_USE_UTF8)
    // Western accents support. These accented characters use
    // combined bitmaps and require relatively little storage.
    #define TOUCH_UI_UTF8_WESTERN_CHARSET

Layakovi, 2021-02-25:
hello succeeded because i have a problem with that too.
Write if you come up with something.
Thanks

RazzerX7, 2021-03-25:
I check the lcd pinout, anet is using a difrenft communication format, isn't spi, I don't remember the name but what's 8 bit something, so if you want to use this panel you need to make custom board for the lcd, or  re map the current one if you see closely they are using a lot of connections for the display not Just clk, vsyn, hsyn, etc.

thisiskeithb, 2021-03-25:
> I check the lcd pinout, anet is using a difrenft communication format, isn't spi, I don't remember the name but what's 8 bit something

FSMC.

These TFTs are not going to work on other boards without heavy custom wiring or daughterboard/adapter.

thisiskeithb, 2021-03-25:
I’m closing this request since it’s not something that can be handled via code.

closed 2021-03-25 by thisiskeithb