MarlinFirmware/Marlin #18159: [FR] Firmware retract and z lift while moving

paulnewall, 2020-05-30:
### Description
During non-printing moves ooze causes strings, and blobs when arriving at the destination. Retract reduces strings and blobs. Fast moves reduce blobs but they still occur on long moves (which take more time). Z lift reduces the impact of blobs by depositing them on top of the print instead of on the side of the print. But z lift is slow which makes strings worse and allows more time for oozing.
If z lift could be done during the move we could leave the start position at high speed (good for string reduction) reduce the duration of the move (good for blob reduction) and still deposit the blob on top of the print (reduces the impact of the blob)

### Feature Workflow
G10 would:
1. Do the existing extruder retract action (without any z lift)
2. Do the next move as XYZ instead of XY, where the Z is the z lift. (either G10 looks ahead to the next move, or it sets a flag to tell the next move to do the z lift as well as XY)

G11 would:
3. Do a vertical z un-lift and extruder un-retract as it does now.

This feature could be switched on/off (with an M code?) to keep it backward compatible

#### Additional Information
I tested this by modifying prusaslicer and it seems very helpful when printing petg.
If we wish to avoid the dependence on the following move, an alternative implementation could be to allow G10 to have optional X Y coordinates. But I think that would mean slicers need changing to use this feature.

qwewer0, 2020-06-06:
You mean [this](https://marlinfw.org/docs/features/fwretract.html)?

paulnewall, 2020-06-06:
I want to modify the existing fw retract so that instead of doing: Up, Horizontal, Down. It does: Diagonal-(up&horizontal moves combined), Down. I'll add some pictures later.

qwewer0, 2020-06-06:
Ok, get it now.
What do you consider a slow z movement?

paulnewall, 2020-06-06:
[image]
[image]
Here are pictures: A sketch of the motion showing the sloping move & z lift, followed by the vertical unlift. A photo of some PETG trials. The move is 100mm. These all have 3mm filament retraction at 20mm/s. The left sample has no z lift - there is some stringing and significant blobs on the side of the part where the nozzle arrives at the end of each move and the blobs are wiped off on the side of the part. The middle sample has normal z lift with z axis speed 20mm/s - the blobbing is better because it goes onto the top of each layer but the stringing is worse. The string starts larger due to the vertical z movement being slower than the horizontal movement. The right hand sample has the proposed sloping z lift - it has the same stringing as no lift, but still has the improvement in blobbing.

qwewer0, 2020-06-07:
On the 3rd part, did you notice a slower XY movement in the diagonal Z-hop movement?

paulnewall, 2020-06-07:
This is what I expect, and it's what I think I am seeing: The effect on the XY speed depends on: the XY and Z speed limits and on the XY and Z distances. When the XY distance is long compared with the Z distance, the Z axis can keep up with X and Y and the speed is not reduced. When the XY distance is short Z can't keep up and so the XY speed gets reduced. But that does not make the move+zlift  take longer than it would if the Z lift happened before the XY move and it does not make the speed of departure from the surface slower than it would be if the Z lift happened first.

There's one thing that might be making the moves slower: I think the normal (up-along-down) Z lift is done at DEFAULT_MAX_FEEDRATE for Z. I think my edit is using the current Z (and XY) speed limits set by the slicer which could be lower than DEFAULT_MAX_FEEDRATE. It would probably be possible to correct that by having G10 set the Z speed limit to maximum and G11 set it back to what it was.

thinkyhead, 2020-07-08:
It seems rather risky and a bit out of phase with the way G-code works to make every next move after a `G10` combine with the Z lift, while not doing the Z lift at all during the `G10`. This kind of fancy multiple movement voodoo would be better handled by the slicer.

paulnewall, 2020-07-10:
I had the same kind of feeling, but two things made me interested in having it in Marlin.
One was that when I suggested it was added to a slicer, I got comments saying - what about firmware z lift?
The other was that implementing this in Marlin is incredibly simple, you just don't do the vertical move. The inclusion of the z lift offset into subsequent moves is already part of the existing z lift implementation because the z axis must stay up until a G11 occurs.

ikjl, 2021-05-22:
This sounds like a cool feature. Usually z-lifts are very small compared to retraction distances and in cases where retractions are short there is little chance for oozing.