MarlinFirmware/Marlin #18161: [BUG]Z_STEPPER_AUTO_ALIGN wrong direction

chenxiaowei107, 2020-05-31:
I have corexy machine,I used the double Z,

i define Z_STEPPER_ALIGN_XY { { 20, 155 }, { 290, 155 }} in the Configuration_adv.h

when I used G34,the probe move to 
point 1:20,155
point 2:290,155

point 3:290,155
point 4:20,155

point 5:20,155
point 6 290,155

ecv: ITERATION: 1
Recv: DIFFERENCE Z1-Z2=1.40
Recv: ITERATION: 2
Recv: DIFFERENCE Z1-Z2=-0.12
Recv: Decreasing accuracy detected.
Recv: ITERATION: 3
Recv: DIFFERENCE Z1-Z2=-0.15 
Recv: ITERATION: 4
Recv: DIFFERENCE Z1-Z2=-0.32  **Wrong direction** 
Recv: Decreasing accuracy detected.
Recv: ITERATION: 5
Recv: DIFFERENCE Z1-Z2=0.06
Recv: Decreasing accuracy detected.
Recv: ITERATION: 6
Recv: DIFFERENCE Z1-Z2=0.09 **Wrong direction** 
Recv: Decreasing accuracy detected.
Recv: ITERATION: 7
Recv: DIFFERENCE Z1-Z2=-0.01
Recv: Decreasing accuracy detected.
Recv: Target accuracy achieved.
Recv: Did 7 of 10
Recv: Accuracy: 0.01

bill-orange, 2020-06-01:
I can confirm this on my Tronxy X3A except for me its worse.  See below. 

first  Diff = 0,04
second Diff = 0.06
third and final  Dif = 0.08

Running auto-align repeatedly results in visible tilting of the gantry.

chenxiaowei107, 2020-06-01:
> I confirmed this on my Tronxy X3A, except for me its worse. See below.
> 
> first Diff = 0,04
> second Diff = 0.06
> third and final Dif = 0.08
> 
> Running auto-align repeatedly results in visible tilting of the gantry.

I think you can try change z1 to z2 and z2 to z1 in the pin files

bill-orange, 2020-06-01:
@chenxiaowei107 I bet I could just switch the  Z1 and Z2 cables.  Perhaps there should be a 

`#define Auto_Align_Exchange_Steppers false`  

Change to true and it would be the same thing changing the pin files.  I mean, how could Marlin know which is which?  The #define statement above could also be worded:

`#define Z1_nearest_X_home true`

@thinkyhead does this all make sense or am I missing something?

chenxiaowei107, 2020-06-02:
> @chenxiaowei107 I bet I could just switch the Z1 and Z2 cables. Perhaps there should be a
> 
> `#define Auto_Align_Exchange_Steppers false`
> 
> Change to true and it would be the same thing changing the pin files. I mean, how could Marlin know which is which? The #define statement above could also be worded:
> 
> `#define Z1_nearest_X_home true`
> 
> @thinkyhead does this all make sense or am I missing something?
> 
> **EDIT - Switching the Z stepper cables did the trick. Diff = 0.01. We still need a software solution, I think.

you used marlin 1.0X?

bill-orange, 2020-06-02:
2.0.5.3

Marlor, 2020-06-02:
In 2.0.5.3 you can choose an orientation and see how Z1 and Z2 should be connected:
```
  /**
   * Orientation for the automatically-calculated probe positions.
   * Override Z stepper align points with 'M422 S<index> X<pos> Y<pos>'
   *
   * 2 Steppers:  (0)     (1)
   *               |       |   2   |
   *               | 1   2 |       |
   *               |       |   1   |
   *
   * 3 Steppers:  (0)     (1)     (2)     (3)
   *               |   3   | 1     | 2   1 |     2 |
   *               |       |     3 |       | 3     |
   *               | 1   2 | 2     |   3   |     1 |
   *
   * 4 Steppers:  (0)     (1)     (2)     (3)
   *               | 4   3 | 1   4 | 2   1 | 3   2 |
   *               |       |       |       |       |
   *               | 1   2 | 2   3 | 3   4 | 4   1 |
   *
   */
  #ifndef Z_STEPPER_ALIGN_XY
    //#define Z_STEPPERS_ORIENTATION 0
  #endif
```
Maybe easier just to switch the connectors, than supporting the many many extra combinations in Marlin

bill-orange, 2020-06-02:
I will leave that decision to greater minds than my own. Since I switched the cables, my installation is fine now, in any case.

AnHardt, 2020-06-02:
An other alternative is to swap the pin definitions for the steppers in the pin-file.

bill-orange, 2020-06-02:
Agreed, but memory is short. On the next major release of Marlin a new release of pins would load and the change would be lost.

chenxiaowei107, 2020-06-03:
Please note that
My Z1 and Z2 settings are right（I have change z1 and z2 define in the pin file）, because they are in the right direction at the first time of AUTO_ALIGN, but please note that bold words ，there are 2 times of adjustment in the wrong direction。
I also have #define Z_STEPPERS_ORIENTATION 0
But I think code have some wrong.

ITERATION: 1
point 1 Z1 point Z2 dif 1.40
ITERATION: 2
**pint 1 Z2 point Z1 dif -0.12** (It's mean the ITERATION: 1 align is right)
ITERATION: 3
**pint 1 Z1 point Z2 dif -0.15** (It's mean the ITERATION: 2 align is **Wrong**)
ITERATION: 4
**pint 1 Z2 point Z1 dif -0.32 ** (It's mean the ITERATION: 3 align is **Wrong**)
ITERATION: 5
**pint 1 Z2 point Z1 dif -0.06 ** (It's mean the ITERATION: 4 align is **Right**)
.....

Marlor, 2020-06-03:
Remember not to set `Z_STEPPER_ALIGN_XY` if you are using `Z_STEPPERS_ORIENTATION`
How precise is your probe? Try a Probe Accuracy Test `M48`

chenxiaowei107, 2020-06-03:
I have test probe precise ,it about 0.03.I set Z_STEPPER_ALIGN_ACC = 0.03
I Will try to not define Z_STEPPER_ALIGN_XY or Z_STEPPERS_ORIENTATION or
 Z_STEPPER_ALIGN_XY and Z_STEPPERS_ORIENTATION

Marlor, 2020-06-03:
Standard deviation of 0.03 is not very good, thats alot of wrong steps.
But not as bad as it would explain the wrong alignment.

chenxiaowei107, 2020-06-03:
It is working correctly in dual Z,Thank you very much.
Recv: ITERATION: 1
[...]
Recv:
Recv: DIFFERENCE Z1-Z2=18.03
[...]
Recv:
Recv: ITERATION: 2
[...]
Recv:
Recv: DIFFERENCE Z1-Z2=1.99
Recv:
Recv: ITERATION: 3
[...]
Recv:
Recv: DIFFERENCE Z1-Z2=2.36
Recv:

closed 2020-06-03 by chenxiaowei107