MarlinFirmware/Marlin #18162: [BUG] Anycubic i3 Mega BLTouch 3.1 original not deploying on home

alphastrem, 2020-05-31:
### Bug Description

The pin on my original bltouch 3.1 isnt deploying when trying to abl from the print menu. But the pin does deploy and retract in an auto test when powering on as well as sending gcode commands does deploy and retract also. The printer does home X and Y and then starts to drive down on the Z but with no pin deploy and the Z pcb endstops unplugged just goes down and down until i GUESS would crash into the bed. 

Also the home button in cura for Z doesnt work with this fw

### My Configurations

I have tested loads of peoples configs but the closest that seems to work is MNieddu91 - Marlin-AI3M-BLTouch-master

config files are here
[](https://www.dropbox.com/s/npa4nqkcdfqpfuu/Marlin.zip?dl=0)

#### Additional Information

youtube video here  - [](https://youtu.be/S2S_CDAGo7o)

alphastrem, 2020-06-03:
Bump - Please can someone help me - im wracking my brains over this one and now my brains are mash potato

boelle, 2020-06-21:
@alphastrem still an issue?

alphastrem, 2020-06-21:
Yes unfortunately still no further forward

thinkyhead, 2020-06-21:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

alphastrem, 2020-06-21:
Ok no problem: I think I did try that in my many many tests but I’ll happily try again and test any suggestions. I’ve even bought a new main board to see if that’s the issue which I know it’s not but we will see when that gets delivered too

alphastrem, 2020-06-24:
This is my code... ive removed 99% of the commented lines

`

#pragma once

#define CONFIGURATION_H_VERSION 020005

#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define SHOW_BOOTSCREEN
#define SERIAL_PORT 0
#define BAUDRATE 250000
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_TRIGORILLA_14
#endif
#define CUSTOM_MACHINE_NAME "Anycubic i3"
#define EXTRUDERS 1
#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
#if ENABLED(SINGLENOZZLE)
#endif
#if ENABLED(MK2_MULTIPLEXER)
#endif
#if ENABLED(SWITCHING_EXTRUDER)
  #define SWITCHING_EXTRUDER_SERVO_NR 0
  #define SWITCHING_EXTRUDER_SERVO_ANGLES { 0, 90 } // Angles for E0, E1[, E2, E3]
  #if EXTRUDERS > 3
    #define SWITCHING_EXTRUDER_E23_SERVO_NR 1
  #endif
#endif
#if ENABLED(SWITCHING_NOZZLE)
  #define SWITCHING_NOZZLE_SERVO_NR 0
  #define SWITCHING_NOZZLE_SERVO_ANGLES { 0, 90 }   // Angles for E0, E1 (single servo) or lowered/raised (dual servo)
#endif
#if EITHER(PARKING_EXTRUDER, MAGNETIC_PARKING_EXTRUDER)
  #define PARKING_EXTRUDER_PARKING_X { -78, 184 }     // X positions for parking the extruders
  #define PARKING_EXTRUDER_GRAB_DISTANCE 1            // (mm) Distance to move beyond the parking point to grab the extruder
  #if ENABLED(PARKING_EXTRUDER)
    #define PARKING_EXTRUDER_SOLENOIDS_INVERT           // If enabled, the solenoid is NOT magnetized with applied voltage
    #define PARKING_EXTRUDER_SOLENOIDS_PINS_ACTIVE LOW  // LOW or HIGH pin signal energizes the coil
    #define PARKING_EXTRUDER_SOLENOIDS_DELAY 250        // (ms) Delay for magnetic field. No delay if 0 or not defined.
  #elif ENABLED(MAGNETIC_PARKING_EXTRUDER)
    #define MPE_FAST_SPEED      9000      // (mm/m) Speed for travel before last distance point
    #define MPE_SLOW_SPEED      4500      // (mm/m) Speed for last distance travel to park and couple
    #define MPE_TRAVEL_DISTANCE   10      // (mm) Last distance point
    #define MPE_COMPENSATION       0      // Offset Compensation -1 , 0 , 1 (multiplier) only for coupling
  #endif
#endif
#if ANY(SWITCHING_TOOLHEAD, MAGNETIC_SWITCHING_TOOLHEAD, ELECTROMAGNETIC_SWITCHING_TOOLHEAD)
  #define SWITCHING_TOOLHEAD_Y_POS          235         // (mm) Y position of the toolhead dock
  #define SWITCHING_TOOLHEAD_Y_SECURITY      10         // (mm) Security distance Y axis
  #define SWITCHING_TOOLHEAD_Y_CLEAR         60         // (mm) Minimum distance from dock for unobstructed X axis
  #define SWITCHING_TOOLHEAD_X_POS          { 215, 0 }  // (mm) X positions for parking the extruders
  #if ENABLED(SWITCHING_TOOLHEAD)
    #define SWITCHING_TOOLHEAD_SERVO_NR       2         // Index of the servo connector
    #define SWITCHING_TOOLHEAD_SERVO_ANGLES { 0, 180 }  // (degrees) Angles for Lock, Unlock
  #elif ENABLED(MAGNETIC_SWITCHING_TOOLHEAD)
    #define SWITCHING_TOOLHEAD_Y_RELEASE      5         // (mm) Security distance Y axis
    #define SWITCHING_TOOLHEAD_X_SECURITY   { 90, 150 } // (mm) Security distance X axis (T0,T1)
    #if ENABLED(PRIME_BEFORE_REMOVE)
      #define SWITCHING_TOOLHEAD_PRIME_MM           20  // (mm)   Extruder prime length
      #define SWITCHING_TOOLHEAD_RETRACT_MM         10  // (mm)   Retract after priming length
      #define SWITCHING_TOOLHEAD_PRIME_FEEDRATE    300  // (mm/m) Extruder prime feedrate
      #define SWITCHING_TOOLHEAD_RETRACT_FEEDRATE 2400  // (mm/m) Extruder retract feedrate
    #endif
  #elif ENABLED(ELECTROMAGNETIC_SWITCHING_TOOLHEAD)
    #define SWITCHING_TOOLHEAD_Z_HOP          2         // (mm) Z raise for switching
  #endif
#endif
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 2        // Number of steppers in your mixing extruder
[... 472 lines cut ...]
  #define XPT2046_X_OFFSET       -43
  #define XPT2046_Y_OFFSET        257
#endif

#define SOFT_PWM_SCALE 0

#if EITHER(RGB_LED, RGBW_LED)
#endif

#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRBW // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     4       // LED driving pin
  #define NEOPIXEL_PIXELS 30       // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
#endif

#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif

#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command

#define SERVO_DELAY { 300 }
`

boelle, 2020-06-30:
@alphastrem code is to long to read, so was the problem fixed after retesting?

alphastrem, 2020-06-30:
No still doesn’t work it homes with the physical endstops and the Z-up a few mil ... bltouch pin extends ... moves to the first mesh point ... z-down ... the pin hits the bed but doesn’t ‘flick’ back up inside itself. And then starts flashing red.

boelle, 2020-06-30:
and the distance from the nozzle tip to bl touch is correct?

[image]

alphastrem, 2020-06-30:
Actually that is pretty spot on you can see my YouTube video here https://youtu.be/S2S_CDAGo7o

At about 35 seconds in you can see how level itnis

[image]<https://user-images.githubusercontent.com/3832239/86147219-69085c00-baf9-11ea-83b3-a85b94d636e3.png>

closed 2020-08-05 by github-actions[bot]