MarlinFirmware/Marlin #18163: SD Firmware Update issue with STM32103

ModMike, 2020-05-31:
**Bug Description**

I am trying to use the SD Firmware update function with a Fysetc Cheetah 1.1b board. I installed the new Fystec SD firmware updater capable boot loader on the board and used the latest official Marlin bug fix branch from yesterday.

As per the Fysetc instructions I went to add their code but found similar code already there. When I uncomment the existing code and build, it fails.

When I use the Fysetc code using [their instructions](https://github.com/FYSETC/Bootloader-STM32F103), Marlin builds successfully. When flashed via regular USB, it will not boot. re-flashing without the code allows Marlin to boot with the new boot loader.

**Hardware Used**

1) Ender 3
2) Fysetc Cheetah 1.1b (new improved version of board)with new boot-loader from their [Github](https://github.com/FYSETC/Bootloader-STM32F103)
3) Proven SD card (did many prints with it)

**Steps to reproduce for Marlin:**

Test 1

1) Uncomment similar code in Marlin\Marlin\buildroot\share\PlatformIO\scripts\fysetc_STM32F1.py:

`# Relocate firmware from 0x08000000 to 0x08010000
# for define in env['CPPDEFINES']:
#    if define[0] == "VECT_TAB_ADDR":
#        env['CPPDEFINES'].remove(define)
#env['CPPDEFINES'].remove(("VECT_TAB_ADDR", 134217728))
#env['CPPDEFINES'].append(("VECT_TAB_ADDR", "0x08010000"))`

2) Build using STM32103RC_Fysetc Env

3) Build fails with following error:

`ValueError: list.remove(x): x not in list:
  File "/Users/Michael/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 176:
    env.SConscript(item, exports="env")
  File "/Users/Michael/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/Michael/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "/Users/Michael/Documents/GitHub/Marlin-bugfix-2.0.x/buildroot/share/PlatformIO/scripts/STM32F103RC_fysetc.py", line 9:
    env['CPPDEFINES'].remove(("VECT_TAB_ADDR", 134217728))`

Test 2

1) Do not uncomment code
2) Addd Fysetc code

`define in env['CPPDEFINES']:
    if define[0] == "VECT_TAB_ADDR":
        env['CPPDEFINES'].remove(define)
env['CPPDEFINES'].append(("VECT_TAB_ADDR", "0x08010000"))
env.Replace(LDSCRIPT_PATH="buildroot/share/PlatformIO/ldscripts/fysetc_aio_ii.ld")`

3) Build using STM32103RC_Fysetc Env
4) Flash to Fysetc Cheetah

No display, presumably Marlin did not boot.

**Notes**

 - I did not enable #define SD_FIRMWARE_UPDATE because it says it is for ATMega 2560 ONLY.
- Configs are attached 

**Questions**

1) Was I supposed to #define SD_FIRMWARE_UPDATE anyway?
2) IS the filename supposed to be changed from **firmware.bin** to **old.bin** after a successful flash?

sl1pkn07, 2020-05-31:
https://github.com/FYSETC/Bootloader-STM32F103/issues/3

xC0000005, 2020-06-03:
I don't have a cheetah but the line setting the VECT_TAB_ADDR to 134217728 shouldn't be uncommented. The one that appends is fine, since the previous loop stripped off VECT_TAB_ADDR entries.

ModMike, 2020-06-03:
There are a few more changes as well, a new pull request by Fysetc was [just merged](https://github.com/MarlinFirmware/Marlin/pull/18179/commits) 

Unfortunately my tool chain is messed up so I can't test. Been trying to fix it for a week.

xC0000005, 2020-06-03:
Well, when you can build a binary, share it. I’ll drop it in binary ninja and see if it’s set for the right addresses. Most of those changes don’t look like they would fix it, unless the LD file wasn’t getting set.

ModMike, 2020-06-03:
Will do. I am on OSX and mistakenly installed a few versions of Python that I can't seem to uninstall. It's totally messed up. 

I reinstalled on another Mac and it worked beautifully. What a pain.

xC0000005, 2020-06-03:
Same - my primary macbook can’t even build anymore. I have another where I test PIO builds. Let me know if you figure out how to fix it.

ModMike, 2020-06-03:
Type the following in terminal to list all your python install files:

ls -ls /usr/bin/python*

Many of them are system files and can't be removed. That being said, Apple is deprecating Python 2.7 and it supposedly removes the directory. Mine is still there.

ModMike, 2020-06-05:
coo

> Same - my primary macbook can’t even build anymore. I have another where I test PIO builds. Let me know if you figure out how to fix it.
> […](#)

I fixed it and it was a lot of work, nearly 30 hours of hunting and piecing things together. I am not sure what is or is not necessary but try this and let me know if you got it up and running:

Cleanup:

1) Go to your home folder and use Command + Shift + Period to unhide hidden files. You can check [this post](https://osxdaily.com/2018/02/12/show-hidden-files-mac-keyboard-shortcut/) if unsure how
2) Erase the hidden .platformio folder in your home folder
3) Open terminal and type in
`sudo pip3 uninstall platformio` 
You will probably get a message that it is not installed. The issue is that PIP can no longer install PlatformIO in 10.15.5, nor does it work with Python 2.7 (at least for me)
4) Use home-brew to install PlatformIO. If you don't have home-brew  you can install it by opening terminal and typing or copy pasting the following text
`/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"`
5) In terminal type or paste the following to install PlatformIO
`brew install platformio`
6) open up VSCode and install the python extension from microsoft
7) When it finishes installing use Command + Shift + P to open the command palette
8) Type in "Python: select interpreter" and choose Python 3.7.3 for command line tools (both options should work)
9) Double check you work by opening the .platformio folder in your home holder by double clicking. You should not see anything relating to Python 2.7 or any version for that matter.
10) Downlaod latest Marlin an compile as is to test.

I am sure I have a lot of unnecessary step in here but thats' what finally did it for me.

Good luck!

xC0000005, 2020-06-05:
> … <x-msg://10/#>

So, can you boot with the bootloader now?

ModMike, 2020-06-05:
The pull request I referenced was merged but give me 10 minutes to try it. I was just so happy to fix VSCode.

ModMike, 2020-06-05:
So it builds perfectly but when I flash it it I lose the screen. Will work on it tomorrow, need to sleep.

xC0000005, 2020-06-05:
Zip up the bin and mail it to me at my github handle at gmail dot com. I’ll see if it’s at the right offset and setting the VTOR.

ModMike, 2020-06-05:
Ok but I just did a quick test so I used the Ender 3 sample config, set the board and baud and that was it.

I uncommented all the code that was suggested at https://github.com/FYSETC/Bootloader-STM32F103

[firmware.bin.zip](https://github.com/MarlinFirmware/Marlin/files/4733991/firmware.bin.zip)

ModMike, 2020-06-05:
I suspect I need to compile their boot loader and refalsh. It looks like they made a few changes in the last 2 weeks but the release version is 9 months old.

xC0000005, 2020-06-05:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/buildroot/share/PlatformIO/scripts/STM32F103RC_fysetc.py#L35

Shouldn’t the offset be adjusted for the upload tool, to 0x8010000?  If you’ve been uploading from PIO, I think it’s been flashing to 0x8000000.

ModMike, 2020-06-05:
No, it starts it at 0X801000 on m’y machine. I see it as it uploads.

sl1pkn07, 2020-06-05:
now works for me. tested with AIOII board

ModMike, 2020-06-05:
Still doesn't work for me but I am probably overthinking it. This is what I did:

1) Flashed the board with the hex boot loader form the Fysetc Github
2) Wrote a regular version of Marlin
3) Uncommented the 2 sections in the newly merged Fysetc code in Marlin
4) Put Firmware.bin on sd card
5) Inserted card in SD slot with Machine powered off
6) Powered on Machine

And Nada.

Am I supposed to enable anything else?

sl1pkn07, 2020-06-05:
step 2 and 3 should be switched

1) burn the bootloader
2) uncoment the two sections
3) build the marlin
4) put the firmware.bin in the SD
5) poweroff or reset the machine

ModMike, 2020-06-05:
I think my mistake was loading regular Marlin after I burnt the boot loader. I will re-burn and try.

ModMike, 2020-06-05:
Worked perfectly! It was critical for me to get this to work because I used the 4 header pins to connect a Wemos Mini Pro to the 4 header pins on the front, which takes over the USB port. I didn't want a separate power switch to turn off the board.

Now I have a wireless printer with ESP3d-WebUI (an Octoprint like web interface). I didn't want to have to wire in a power switch every time I wanted to use the port to update. Last step is to see if I can do something about the file transfer speed. Hoping that new Binary mode is useful.

ModMike, 2020-06-05:
So ran into another issue. When I fully configured my printer, I ran in to `.rodata' will not fit in region `rom'. Does the SD Flash code significantly reduce program? Never had issues before.

xC0000005, 2020-06-05:
The bootloader cuts down on available space. How much did it overflow by?

ModMike, 2020-06-05:
8764 bytes

ModMike, 2020-06-05:
I thought I saw somewhere that the BTT STM32 is actually 512K and you needed to build for that size. Any chance that is true for Cheetah?

sl1pkn07, 2020-06-05:
not true at all, some btt smt32 have RET6 chips, which have real 512K. the RCT6 chip, like Cheetahs, have 256K. but some RCT6 chips is a RET6 with some sectors damaged, and renamed as RCT6. so you need use with carefull. i personal recomend use the certified size for this chips. 256K

greetings

ModMike, 2020-06-05:
Whats the worst that can happen? Reinstall the boot loader? Lets hope their yields improved. Should I try this: 512K - 28K - 4K or this 512k- 64?

sl1pkn07, 2020-06-05:
you can try use the BTT E3/DIP or MKS (E3/E3D) botloader and cross the fingers. btt starts the firmware in 0x08007000 and MKS E3/E3D in 0x08006000 (this last needs a "encrypted" firmware)

greetings

ModMike, 2020-06-05:
I used 320k and lost my LCD menus. Do you think they are hardwired to 256K or have you heard of  so I guess it's physically locked at 256K or its just a config thing? Should I have used 512K?

xC0000005, 2020-06-05:
64k is a hell of a lot of space for a bootloader. I’d be tempted to either use openblt or recompile the existing one for something that’s not ridiculously large. If all it’s providing is IAP, malyan’s does it in 8k (and so does the open bootloader). Is this thing doing anything else useful?

sl1pkn07, 2020-06-05:
nothing more flash the firmware and start it

i'm interest in you say @xC0000005 , because i make a botloader for my BTT E3 DIP with genuine RET6 chip (512K), but not works at all (only flash the first time, but not when reflash when update the firmware.bin)

greetings

xC0000005, 2020-06-05:
If you want to use the existing one, the offset is just a #define. Make sure it’s a multiple of the sector size and larger than the .bin and you can recompile the bootloader for that (then adjust Marlin’s offset in the script the same way). But again, I have to wonder what this provides over OpenBLT.

sl1pkn07, 2020-06-05:
with "exixtin" you mean the Malyan? link plox?

greetings

xC0000005, 2020-06-05:
No, just recompile the fytsec one and change the #define for the offset. It should be a multiple of whatever the board’s sector size is, and larger than the binary for your bootloader. So if the bootloader .bin is 6k and you have 4k pages, an 8k offset would work.

sl1pkn07, 2020-06-05:
how build the fysetc bootloader? with what toolchain/platform/IDE?

greetings

ModMike, 2020-06-05:
> how build the fysetc bootloader? with what toolchain/platform/IDE?
> 
> greetings

No. They it's in Keil project format and their Ultravision is a commercial product. There is a free one but I am not sure what if it's enough. Keil is very popular STM32 developers because of their powerful debugging tools.

ModMike, 2020-06-05:
> No, just recompile the fytsec one and change the #define for the offset. It should be a multiple of whatever the board’s sector size is, and larger than the binary for your bootloader. So if the bootloader .bin is 6k and you have 4k pages, an 8k offset would work.
> […](#)

How do I confirm sector size?

ModMike, 2020-06-05:
> If you want to use the existing one, the offset is just a #define. Make sure it’s a multiple of the sector size and larger than the .bin and you can recompile the bootloader for that (then adjust [...]

Looking at OpenBLT. The advantage for Fysetc building their own boot loader is that they don't have to pay for it. OpenBLT does have a commercial license but not sure if that means Fysetc would have to pay. Luckily it's free for non-commercial.

ModMike, 2020-06-08:
@xC0000005

> No, just recompile the fytsec one and change the #define for the offset. It should be a multiple of whatever the board’s sector size is, and larger than the binary for your bootloader. So if the bootloader .bin is 6k and you have 4k pages, an 8k offset would work.
> […](#)

Apparently all we need to change in the fysetc bootloader is:

**Bootloader-STM32F103/CODE/APP/bootloaders.h change**
#define BOOTLOADER_FLASH_ADDR ( 0x08010000 )
to
#define BOOTLOADER_FLASH_ADDR ( 0x08070000 )

I tried to install the free version of Keil but the project is in 4.0. I found a version and tried to install but keep getting a tool chain error. Do you have Keil installed?

ModMike, 2020-06-09:
Project update:

I installed the latest version of Keil uVision. The license is free for uploads under 32K which was perfect for this project. A few notes:

1) When starting Keil, you will have a device not found or similar error, ignore it
2) You will be told this is a legacy project, choose option 1 to bring it into Keils new format
3) Close the project and then Keil
4) Restart Keil and reload the project, it should load with no errors.
5) Make the changes to the boot loader and Marlin as referenced in the [Fysetc github readme](https://github.com/FYSETC/Bootloader-STM32F103)
6) I used the Fysetc upload tool as referenced in their Github
7) Make the changes to Marlin (I will submit a pull request so this may not be necessary, depending on when it is mreged)
8) Good news is that the bootloader occupies half it's original 64k space and clocks in at a nice 32K of reserved space.

Note that you will need to install any new Marlin firmware by SD card, I don't think the USB method works anymore, at least in my limited testing. Let me know your experiences.

Here is the bootloader that I compiled with the change. You will most likely get a blank screen when you install it. Don't panic, just put the marlin firmware.bin on the sdcard and insert into your machine, power on and give it at least 60 seconds. If the display is still blank, power off and power back on. Your menu should appear.

[Bootloader-STM32F103.zip](https://github.com/MarlinFirmware/Marlin/files/4752502/Bootloader-STM32F103.zip)

closed 2020-06-09 by ModMike