MarlinFirmware/Marlin #18164: [BUG] TMC2209 crashes into frame if using low HYBRID_THRESHOLD value, not staying in stealthchop while homing

govido, 2020-05-31:
Hi,

I wanted to change my printer electronics from SKR 1.3 to SKR Pro with TMC2209 drivers.
After a few runs of tweaking I ran in to a problem of setting a 'low' HYBRID_THRESHOLD for the TMC2209. Because I was getting layer shifts in certain speed changes I decided to play it safe on my default settings:

#define X_HYBRID_THRESHOLD      50 
#define Y_HYBRID_THRESHOLD      40

with this default homing speed:

#define HOMING_FEEDRATE_XY (50*60)

After many attempts of various testing I found following problem:
If the bed or the extruder are alle the way to the front or the right respectively, they have enough time to accelerate to left and the back. 
If I understand the TMC2209 correctly, they HAVE to use stealthchop for homing (TMC2130 need spreadcycle if I recall correctly).
This way, depending on where the extruder / bed stops, there is sometimes enough acceleration time to switch so spreadcycle mode, which can't be used for homing:
This way the axis crash in the frame until timeout or shutoff.

Why am I even using HYBRID_THRESHOLD this low?
Because I want to use my printer as silent and reliable as possible, in the past with TMC2130 I got various layer shift depending on model shape or slicer speeds, even with different slicers on the same model. This way I want to make sure the machine gets more reliable.
If I can lift those limits after a decent number of successful prints I will do so.

What I would expect?

On a homing move HYBRID_THRESHOLD should not be allowed in any way for TMC2209, the drivers have to use stealthchop for this (there was also an closed issue for this, not homing without using stealthchop at all).

### Steps to Reproduce

1. Lower HYBRID_THRESHOLD below your homing speeds
2. Move X / Y axis at the opposite of your homing position
3. Let the machine home, listen for spreadcycle noise change while travel happens
4. See the machine crash without triggering an endstop

**Expected behavior:**

No matter what 'safe' HYBRID_THRESHOLD is defined, the homing moves should (for TMC2209) ONLY be made with stealthchop, maybe resulting in lost steps if homing to fast, but never endlessly crashing and not triggering stallguard.
In the end ignoring HYBRID_THRESHOLD completely for homing moves (maybe put a short warning  in the HYBRID_THRESHOLD notes).

Would be great if someone could test and confirm this issue.
(all tests using the latest marlin bugfix 2.x from may 31st, same issue on latest stable release)

swilkens, 2020-06-01:
There used to be a note in the config files stating TMC2209's should be in stealthChop mode when using sensorless homing, as StallGuard4 is built to work in stealthChop mode.

https://github.com/MarlinFirmware/Marlin/pull/15658

For some reason, this is gone. @teemuatlut @thinkyhead ?

Maybe hybrid_threshold needs some logic to stop it during a homing move

SonBroku, 2020-06-09:
Just as an aside, your TMC2130 had Stealtchop1 that was a bit more susceptible to step loss at higher RPM. I believe stealthchop 2, which is what the 2209s have, is less prone to that. I've never seen a step loss from my 2209s even at very high printing speeds.

rhukster, 2020-06-11:
Will pile on this.  I turned on hybrid mode to stop the horrible screeching steppers at higher speeds.  I set my hybrid threshold to 75mm/s and my homing feedrate is 50mm/s.  However, sensorless homing will not work while hybrid is enabled.  As soon as I disable this, sensorless homing works again. FYI, it seems no matter what i set the threshold to, sensorless homing doesn't work and it just crashes into the frame.  I would guess it's stuck in spreadcycle during homing rather than the required stealthchop.

I'm also using 2209s in UART mode, only X and Y for sensorless homing, while my triple Z is homed and leveled by probe.

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle