MarlinFirmware/Marlin #18167: Starting Print With SD Card Triggers G26 Heating Nozzle Instead

inventabuild, 2020-06-01:
Marlin 2.0.5.3
When I leave my hotend and bed heated idly for a while then go to start a print job via the SD Card Reader on my TAZ printer by scrolling to and clicking on the print job then clicking on Print, the "G26 Heating Nozzle" algorithm gets activated instead of my print job.

When I reboot my printer, then I can start my desired print job successfully via the SD Card Reader.  The only pattern I seem to recognize so far is that this issue mainly occurs when I leave my hotend and bed heated for a while before trying to start a print job although I cannot verify for sure that idly heating the hotend and bed for a period of time are root causes.  Rebooting my printer always solves the problem.

"G26 Heating Nozzle" is not even an option on my SD Card so I have no idea why it gets triggered. 
 Any idea if this might be a bug?

sjasonsmith, 2020-06-20:
If you are still having issues, please update this according to the new issue template.
It is important to have configuration files and exact steps to reproduce prior to anybody being able to try to help.

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

inventabuild, 2020-06-24:
The problem persists.  I found out the problem occurs if I don't reboot my printer after a print job is completed before starting a new print.  Here's a video showing the problem in action.  I am not double clicking to start the new print job, the two or three clicks you hear are firmware generated when it false starts the G26 algorithm.

https://youtu.be/mlZJQdhnuEI

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4822597/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4822598/Configuration_adv.zip)

thinkyhead, 2020-06-24:
Marlin starts a `G26` when it receives the command "G26" from a serial port or G-code file. Showing us that the behavior exists is fine, but it would be better if you did more testing and logging to narrow down the cause and figure out what makes it happen more often or less often.

To help with your testing you can send `M111 S3` before you take whatever action causes the `G26` to occur and the firmware will log every command it receives. You might see a "G26" coming from some unexpected source.

You should also make a very minimal G-code file that causes the issue to appear and provide that for testing.

inventabuild, 2020-06-27:
Lulzbot's start gcode caused the problem.  I customized it for Marlin 2.0.5.3 and everything's good now.