MarlinFirmware/Marlin #18168: [BUG] Ender 5 Pro, Auto home & prints off center (10mm low), possible culprit for ABL issues

txt4nk, 2020-06-01:
This is my first time posting an issue like this, so in advance, if I have not done something right or provided enough information, I apologize. 

Bug Description:
Auto home puts the probe\nozzle 10mm lower than the center of the bed. I have also had absolute hell getting my bltouch to actually level the bed and now I'm wondering if the two issues are related. I have messed around with changed the z safe homing x and y points to the older "((x_bed_size) / 2)" to see if that worked or not, still not centered. The configuration files I've created below are bugfix 2.0.x & example configs downloaded on 05/27/20 at 11:30AM CST. 

My Configurations:
Firstly, my printer and all hardware changes.
Ender 5 Pro
E3D Hemera
Authentic Antclabs BLTouch 3.1
BTT SKR E3 MINI V1.2 

My current firmware state, after all self troubleshooting before waving the white flag.
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4709007/Marlin-bugfix-2.0.x.zip)

Steps to Reproduce:
1. Compile firmware w/ marlin supplied Ender 5 Pro configurations (string config author shows @thisiskeithb)
   a. Changed all necessary items for the SKR E3 Mini V1.2
      aa. #define SERIAL_PORT 2 (configuration.h)
      ab. #define SERIAL_PORT_2 -1 (configuration.h)
      ac. #define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2 (configuration.h)
      ad. //#define SPEAKER (configuration.h)
      ae. default_envs = STM32F103RC_btt_512K (platformio.ini)
   b. Changed all necessary items for the E3D Hemera (just esteps)
   c. Uncommented BLTouch (-46, -2, -4.1), multiple probing (2), HSmode not enabled.
   d. Uncommented Auto Bed Leveling UBL
   e. Uncommented restore level after G28
   f. Uncommented z safe homing
2. Flash to printer
   a. Init EEPROM
3. Auto home
4. measure distance from BLTouch probe to the top of the bed
5. measure distance from BLTouch probe to bottom of the bed

Expected behavior:
steps 4 and 5 should both have a measurement around 117.5mm

Actual behavior:
step 4 results in 127.5mm
step 5 results in 107.5mm

Additional Information:
As mentioned in the title, my ABL mesh is so horribly off that the one and only way I've been able to get my printer to run a "decent" print, which only lasted 3 prints, was to go UBL, run 15 points, and spend 2-3 hours adjusting the mesh manually. 

Thanks in advance.

firestrife23, 2020-06-02:
Just want to chime in because I'm having same issue with TH3D's EZABL as well. It seems to be increasing by 0.020 - 0.040 mm after each z-offset calibration and still drive right into the bed.

txt4nk, 2020-06-03:
I've rolled back all the way to Marlin 2.0.3 and still have an issue. 

I am also still fighting Z Safe Homing, it will not home in the center of my bed. It's always 10mm low.

txt4nk, 2020-06-03:
So I got my printer to home in the center of the bed finally.. Had to make the adjustment below to the z safe offset
#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT X_CENTER  // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT **_(Y_CENTER + 10)_** // Y point for Z homing when homing all axes (G28).

closed 2020-06-03 by txt4nk