MarlinFirmware/Marlin #18169: [BUG] Spindle Speed Percentage onscreen incorrect when max speed is over 255

Patronics, 2020-06-01:
### Bug Description

The spindle speed (or presumably laser intensity too, but untested) displayed as a percentage on the display is incorrect, effectively showing the speed modulo 256 divided by 2.56, instead of a percentage of the actual maximum speed (in my case, 30000). 
### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce
Configure Marlin with a spindle tool, and a maximum speed for the tool over 255, such as the RPM of the tool. 

Install marlin, and turn on the spindle, and experiment with adjusting its speed. 
**Expected behavior:** [What you expect to happen]
Percentage of speed indicated on screen linearly increases with actual set speed.
**Actual behavior:** [What actually happens]
Indicated speed will be incorrect, and at first glance appear entirely random (but consistent). It does in fact follow the pattern I described above.
#### Additional Information
I managed to find a fix to this issue in the code in the current 2.0.x branch, as implemented by a simple function overload to use 16 bit calculations when the max is over 255. This is implemented sucessfully in my fork, [here](https://github.com/Patronics/Marlin/commit/a7f28ae67a3d6bb19853f9700ccfd92acdacf390). However, in the current bugfix-2.0.x this file's code has been changed substantially, and I'm not sure how to translate this fix over. However, I confirmed that this bug is still an issue in the latest form of that branch (although the behavior is somewhat different (different wrong numbers show up, not wrong percentages), it still indicates entirely wrong values when configured with `#define CUTTER_POWER_DISPLAY RPM`)

GMagician, 2020-06-01:
Have you tested bugfix version? it seems that Marlin is now using float so this should not be an issue anymore.

GMagician, 2020-06-01:
```c++
  static cutter_setPower_t interpret_power(const float pwr) {     // convert speed/power to configured PWM, Percentage or RPM in relative or normal range
    #if CUTTER_DISPLAY_IS(PERCENT)
      return (pwr / SPEED_POWER_MAX) * 100;                                               // to percent
    #elif CUTTER_DISPLAY_IS(RPM)                                                          // to RPM is unaltered
      return pwr;
    #else                                                                                 // to PWM
      #if ENABLED(CUTTER_POWER_RELATIVE)
        return (pwr - SPEED_POWER_MIN) / (SPEED_POWER_MAX - SPEED_POWER_MIN) * 255;     // using rpm range as relative percentage
      #else
        return (pwr / SPEED_POWER_MAX) * 255;
      #endif
    #endif
  }
```

Patronics, 2020-06-01:
I did test the bugfix version, it seemed to still give wrong results when in RPM mode. I'll test it more for more details of its misbehavior, and report back.

Patronics, 2020-06-01:
Ok, here's a bunch of set speeds, and the speeds it displays, still very wrong, as it's only showing the 3 least significant digits, and appending a "K" to them:

- Set : Shown

- 7000 : 000K
- 7500 : 500K
- 8700 : 700K
- 10000 : 000K
- 15300 : 300K

GMagician, 2020-06-01:
Then it may be only a display issued, I'll check

GMagician, 2020-06-02:
Don't know where K is coming....when SPEED_MAX_POWER > 255 variable used is `uint16_t`. Function to convert to number is:
```c++
const char* ui16tostr5rj(const uint16_t xx) {
  conv[2] = RJDIGIT(xx, 10000);
  conv[3] = RJDIGIT(xx, 1000);
  conv[4] = RJDIGIT(xx, 100);
  conv[5] = RJDIGIT(xx, 10);
  conv[6] = DIGIMOD(xx, 1);
  return &conv[2];
}
```
then just numeric digits and value may be up to 65535.. can you attach your cnf files?

Patronics, 2020-06-03:
Sure, here's the relevant configuration files: 
[configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/4721273/configuration_files.zip)

Patronics, 2020-06-03:
Aha, I found the origin of the "K" in the source code! https://github.com/MarlinFirmware/Marlin/blob/8477dbf65db4572d6aac07f5c732130607f31545/Marlin/src/lcd/dogm/status_screen_DOGM.cpp#L549 
Haven't investigated there too thoroughly yet, but I suspect there's a good chance the truncated number display issue is caused by something near there too.

ellensp, 2020-06-03:
Give this PR https://github.com/MarlinFirmware/Marlin/pull/18184 a try.

Patronics, 2020-06-03:
@ellensp I've also made a fix that works ( https://github.com/Patronics/Marlin/commit/de334ea04eb178afb375c15b74166d5a8240ca57 ), but looks like you beat me to the pull request. I'll try your variant, and let you know how it goes. The one difference with my version is that it still uses the 3-digit variant for percentages (though I guess it's fine without that change, as it hides leading zeroes), and also my variant removes the "K" from the end (As far as I can tell there's not much need for the K to indicate RPM, and it adds more clutter to an already pretty long number).

ellensp, 2020-06-03:
lol, I just added that also...  3 digits for percentage or PWM, but 5 for RPM

Patronics, 2020-06-03:
Huh, that's odd. Your version seems to have completely broken communication to the display (including the reset button, beeper, etc) somehow? And possibly other things, such as homing sensors? I'm recompiling my version to rule out some weird hardware bug or failure, but it's getting pretty late here, and my computer takes ages to recompile Marlin, so I won't be able to do much more testing until the morning (and, assuming you don't find the cause of this issue (probably something messed up when adding those other number-to-string functions, if I had to guess), I suppose I'll submit the pull request for my variant, which has been tested and verified to work.)

Patronics, 2020-06-03:
Yep, reinstalled my variant, and all the issues went away. Not sure what it is that's up with your variant of the code.

GMagician, 2020-06-03:
> ha, I found the origin of the "K" in the source code! 

That's ok...I was locking into menu item, not display

GMagician, 2020-06-03:
LCD uses i16tostr3rj instead of ui16tostr5rj. @ellensp PR should not be so wrong...

GMagician, 2020-06-03:
@Patronics please test my PR, not so different from @ellensp.
Menu items is working (as it was used to work before) but I'm not able to show value you are refering to.

ellensp, 2020-06-03:
only shows value on status screen when spindle is on.

Patronics, 2020-06-03:
@GMagician, I hope you don't mind, I've submitted my own pull request for this issue with my fix (which was practically identical to ellensp's variant, but didn't change some other things that seemed to cause issues), given the time it takes marlin to recompile, and the fact I need to do so at least 3 time to properly test this fix (ensuring that PWM255 and Percent mode weren't broken), and that I've gotten a working fix on my own, I'm not particularly eager to spend a long time testing your variant redundantly at the moment. If your implementation is generally considered to be a better, or more consistent with marlin's coding style, I'm happy to try it eventually, but if not I don't really see the point when mine is already tested and works reliably.

GMagician, 2020-06-04:
> If your implementation is generally considered to be a better, or more consistent with marlin's coding style, I'm happy to try it eventually, but if not I don't really see the point when mine is already tested and works reliably.

That's a @thinkyhead choice. I don't have problem with your fix, I was not able to test your cnf so for sure yours will work

shitcreek, 2020-06-14:
There has been some changes to that section of code recently. Namely:
```
lcd_put_u8str(STATUS_CUTTER_TEXT_X - 2, STATUS_CUTTER_TEXT_Y, ftostr51rj(float(cutter.unitPower) / 1000));
          lcd_put_wchar('K');
```

Please close the issue if you feel it has been resolved. Thanks.

sjasonsmith, 2020-06-20:
@Patronics, what's the state of this issue? Did you go ahead and try the change that was merged from @GMagician?

Patronics, 2020-06-20:
I believe I did try it successfully, but unfortunately I had an unrelated hardware issue, I'm waiting on a replacement part before I can do any further testing. I'm pretty sure it did work using GMagician's variant until I had that issue. Sorry I don't remember more details.

sjasonsmith, 2020-06-20:
@Patronics based on that I will go ahead and close this issue, since there doesn't seem to be anything else to be done right now.

Once your hardware issue is resolved, this could be re-opened if the issue persists.

PR #18189 (2020-06-04): Fix insufficient digits displayed in spindle RPM
This should fix #18169
I changed old `i16tostr3rj` to `ui8tostr3rj` to reflect real variable type. Don't know if it's ok

To be tested
closed 2020-06-20 by sjasonsmith