MarlinFirmware/Marlin #18170: Dual Extruder 30 purges backwards in Start Gcode

toophast2catch, 2020-06-01:
# NO SUPPORT REQUESTS PLEASE
I just added a dual Extruder Chimera to my Tevo Tornado. 

When I purge the T0 Nozzle it purges fine, but when I switch to T1 and try to purge it- it retracts instead of purging. After that it will print just fine but seems to be something wrong with the logic that Marlin uses to invert the stepper direction for T1. 

Here is my start GCODE:
M117 Getting the bed up to temp!
M140 S[bed0_temperature] ; Set Heat Bed temperature
M190 S[bed0_temperature] ; Wait for Heat Bed temperature
M117 Pre-heating the extruder!
M104 T0 S175; start warming extruder 0 to 175
M104 T1 S175; start warming extruder 1 to 175
G28 ; Home all axes
M117 Auto bed-level GO!
G29 ; Auto bed-level (BL-Touch)
G92 E0 ; Reset Extruder0
G92 E1 ; Reset Extruder1
M117 Getting the extruder up to temp!
M104  T0 S[extruder0_temperature] ; Set Extruder 0 temperature
M104 T1 S[extruder0_temperature] ; Set Extruder 1 temperature
M109 T0 S[extruder0_temperature] ; Wait for T0 Extruder temperature
M109 T1 S[extruder0_temperature] ; Wait for T1 Extruder temperature
G1 X0.1 Y225 Z0.3 F5000.0 ; move to start-line position
M117 Let the Purge Begin!
G1 X0.1 Y180.0 Z0.3 F500.0 E15 ; draw 1st line
G1 X0.4 Y180.0 Z0.3 F5000.0 ; move to side a little
G1 X0.4 Y225 Z0.3 F500.0 E30 ; draw 2nd line
T1;
G1 X5.1 Y180.0 Z0.3 F500.0 E15 ; draw 1st line
G1 X5.4 Y180.0 Z0.3 F5000.0 ; move to side a little
G1 X5.4 Y225 Z0.3 F500.0 E30 ; draw 2nd line
G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
G92 E0 ; Reset Extruder0
G92 E1 ; Reset Extruder1
M117 Autobots! Roll Out!

I will also attach my config files as well. 
Support Requests posted here will be automatically closed!

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. See https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question.

For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org Marlin Forum http://forums.reprap.org/list.php?415
- Tom's 3D Forums https://discuss.toms3d.org/
- Facebook Group "Marlin Firmware" https://www.facebook.com/groups/1049718498464482/
- Facebook Group "Marlin Firmware for 3D Printers" https://www.facebook.com/groups/3Dtechtalk/
- Marlin Configuration https://www.youtube.com/results?search_query=marlin+configuration on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a Bug Report at https://github.com/MarlinFirmware/Marlin/issues/new/choose).

toophast2catch, 2020-06-01:
[Marlin 2053-TT-SKR1.4T-V2.zip](https://github.com/MarlinFirmware/Marlin/files/4712813/Marlin.2053-TT-SKR1.4T-V2.zip)

When T1 tries to purge, it actually retracts 15 and then moves the correct direction on the second line for 30. But this doesnt allow me to purge it well because it first retracted 15. 

Any fix or work around for this? Im running Marlin 2.5.3

boelle, 2020-06-21:
@toophast2catch still an issue?

toophast2catch, 2020-06-21:
No. Please feel free to close. I figured it out through the start gcode

closed 2020-06-21 by boelle