MarlinFirmware/Marlin #18171: Marlin with SKR GTR not compiling

SevCr, 2020-06-01:
Hello Everyone. 

Trying to compile a firmware. Haven't done too many changes, just the enviroment, board, serial ports and enabled steppers.

SKR GTR v.1.0

When compiling it gives an error straight away.
I would much appreciate if someone more experienced can point me to the issue.
Thanks a lot.
________________________________________________________________
________________________________________________________________

> > Executing task: C:\Users\admin\.platformio\penv\Scripts\platformio.exe run <

Processing BIGTREE_GTR_V1_0 (platform: ststm32@>=5.7.0; board: BigTree_GTR_v1; framework: arduino)  
----------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/BigTree_GTR_v1.html
PLATFORM: ST STM32 6.1.0 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10700.200103 (1.7.0)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
'arm-none-eabi-g++' is not recognized as an internal or external command,
operable program or batch file.
*** Error 1
AssertionError: :
  File "C:\Users\admin\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:    
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\admin\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\admin\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\admin\.platformio\platforms\ststm32\builder\main.py", line 98:
    target_elf = env.BuildProgram()
  File "C:\Users\admin\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 
219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\admin\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\admin\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 
219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\admin\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 118:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\admin\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 
219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\admin\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 327:
    env.ConvertInoToCpp()
  File "C:\Users\admin\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 
219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\admin\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 240:
    out_file = c.convert(ino_nodes)
  File "C:\Users\admin\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 94:
    return self.process(contents)
  File "C:\Users\admin\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 115:
    assert self._gcc_preprocess(contents, out_file)
==================================== [FAILED] Took 1.99 seconds ====================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
MightyBoard1280           IGNORED
MightyBoard2560           IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
FYSETC_F6_14              IGNORED
sanguino644p              IGNORED
sanguino1284p             IGNORED
melzi                     IGNORED
melzi_optiboot            IGNORED
at90usb1286_cdc           IGNORED
at90usb1286_dfu           IGNORED
DUE                       IGNORED
DUE_USB                   IGNORED
DUE_debug                 IGNORED
DUE_archim                IGNORED
DUE_archim_debug          IGNORED
LPC1768                   IGNORED
LPC1769                   IGNORED
STM32F103RC               IGNORED
STM32F103RC_fysetc        IGNORED
STM32F103RC_btt           IGNORED
STM32F103RC_btt_USB       IGNORED
STM32F103RC_btt_512K      IGNORED
STM32F103RC_btt_512K_USB  IGNORED
STM32F103RE               IGNORED
STM32F103RE_btt           IGNORED
STM32F103RE_btt_USB       IGNORED
STM32F4                   IGNORED
STM32F7                   IGNORED
ARMED                     IGNORED
STM32F103VE_GTM32         IGNORED
STM32F103VE_longer        IGNORED
mks_robin_mini            IGNORED
mks_robin_nano            IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_e3              IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
STM32F070RB_malyan        IGNORED
malyan_M300               IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          FAILED    00:00:01.995
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
============================== 1 failed, 0 succeeded in 00:00:01.995 ==============================

ellensp, 2020-06-02:
Not a marlin issue, but a issue with platformio or your operating system.

Firstly check you machine is clean of malware. Get malwarebytes and scan and clean your system. 

Then open vscode click on platformio home, scroll down to platforms and uninstall ST STM32. 
Try to compile again, this will reinstall ST STM32 framework.

It it still doesn't work, perhaps your virus checker is to paranoid and not allowing  arm-none-eabi-g++ to be run.

SevCr, 2020-06-02:
Hi. Thanks a lot for advice. 
I will do as instucted. Out of interest, can you tell me what arm-none-eabi-g++ does?

ellensp, 2020-06-03:
Its the main compiler. Converts all the Code into binary

SevCr, 2020-06-08:
Thank you for your help, this is sorted. Tried differrnt PC and it is compiling well.

closed 2020-06-08 by SevCr