MarlinFirmware/Marlin #18172: [BUG] (static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");short description)

sthgardner, 2020-06-01:
### Bug Description

Keep getting DISTINCT_E_FACTORS Errors when compiling. I have uncommented that line to enable. I can list all 5 stepper drivers in Pronterface. But Z steppers move at different step rates. One is correct at 400 and other defaults to E0 at 381.50. Also max feedrate and acceeration will only list 4 driver settings. I need seperate Z steppers to do initial homing/leveling with dual z endstops. Also both recognized in Pronterface. Also tried with latest BugFix version
[Config files.zip](https://github.com/MarlinFirmware/Marlin/files/4713409/Config.files.zip)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step] Compile
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen] Success

**Actual behavior:** [What actually happens] Errors
[Config files.zip](https://github.com/MarlinFirmware/Marlin/files/4713417/Config.files.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-06-02:
You can't set steps/unit for Z2
change #define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 400, 381.5, 400 }
to #define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 400, 381.5 }

From reading the code, there is no steps/unit for Z2 It just uses the steps from Z.
in particular the line 
#define Z_APPLY_STEP(v,Q) do{ Z_STEP_WRITE(v); Z2_STEP_WRITE(v); }while(0)
So when ever Z_APPLY_STEP is called both Z and Z2 is stepped.

What I think is happening is you simply need to up you Z and Z2 stepper driver current 
ie #define Z_CURRENT       800
and #define Z2_CURRENT       800

sjasonsmith, 2020-06-20:
@sthgardner did the suggestion from @ellensp work for you? If we don't hear back we will assume this was a configuration issue and is resolved.

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle