MarlinFirmware/Marlin #18173: [FR] Setting EXTRUDERS to 0 should remove the "Preheat PLA/ABS Conf" from LCD Advanced menu

gminadak, 2020-06-01:
### Description

When setting the EXTRUDERS to 0 the "Preheat PLA/ABS Conf" menu on LCD "Advance" menu should be removed and the "Temperature" from main Menu.

### Feature Workflow

1. #define EXTRUDERS 0

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-06-23:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

gminadak, 2020-06-23:
@boelle I figured out that the menu appears because of the fan and not the extruder (pictures included).
Well maybe for particular setups this makes sense to have a controllable fan but not extruder (laser etc).
But i think it only needs to be in one menu or have an option to complete disable the fans from the screen for other setups.

[ config.zip](https://github.com/MarlinFirmware/Marlin/files/4822291/config.zip)

[image]
[image]
[image]
[image]
[image]

Thank you.

thinkyhead, 2020-06-23:
Apologies for the completely inappropriate reply and label. This is a bug and I am working on it now.

gminadak, 2020-06-26:
Probably you already know this but after the last commits

`Compiling .pio\build\LPC1768\src\src\module\stepper.cpp.o
Compiling .pio\build\LPC1768\src\src\module\stepper\L64xx.cpp.o
Marlin\src\lcd\ultralcd.cpp:90:3: error: 'preheat_t' does not name a type
   90 |   preheat_t MarlinUI::material_preset[PREHEAT_COUNT];  // Initialized by settings.load()
      |   ^~~~~~~~~
*** [.pio\build\LPC1768\src\src\lcd\ultralcd.cpp.o] Error 1`

gminadak, 2020-06-26:
With the latest commit (9013580) everything is ok as you can see from the pictures.
The fan still appears on the status screen but maybe this is from my configuration/board/pins.

I'm closing this.

Thank you.

[image]
[image]
[image]

closed 2020-06-26 by gminadak