MarlinFirmware/Marlin #18174: [BUG] UBL Fine Tune Screen Not Displayed Correctly

bill-orange, 2020-06-01:
Bug Description: The top half of the UBL fine tune screen is cut off.  I am using the RepRap Discount Full Graphics display attached to an SKR1.3.  All other Menus/Screens are fine. See photo.

My Configurations: Attached

Steps to Reproduce:

1. Do the P1 and P3 steps.
2. Using the LCD display and encoder, move through the menus until you get to the Fine Tune Menu.
3. Note the cut off text.

Expected behavior:  Fully legible text

Actual behavior:  Cut off text.  This is cosmetic.  Functionality appears fine.

Additional Information:  This is the only menu that I have found to have a problem.

[image]
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4713487/Marlin.zip)

qwewer0, 2020-06-01:
Works for me.

SKR Mini E3 v1.2, Latest Bugfix, 7x7 UBL mesh.

bill-orange, 2020-06-01:
@qwewer0 What display board are you using?

qwewer0, 2020-06-01:
> @qwewer0 What display board are you using? Remember the problem is the display not functionality. Go to the mesh editor and check your display.

My bad.

Ender 3 original 12864 display.

Tannoo, 2020-06-01:
Works for me 2.

It is hard to guess on all the options you have enabled or disabled in your copy of the "Latest Bugfix". The "Latest Bugfix" is a loose term. The bugfix branch can be updated several times a day sometimes. So, there is a timeline for the bugfix branch.

Are you looking for support? Please don't post here. Instead please use one of the support links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

This was also skipped:

>**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

Tannoo, 2020-06-01:
Nevermind. Still works for me.

bill-orange, 2020-06-01:
I suspect this problem relates to the RepRap Discount Full Graphics Display only.  I would try to solve the problem myself but I have no clue about which file contains this  particular piece of code.

Tannoo, 2020-06-02:
I am not so sure on that. I have that exact display and it's working fine with the Mega2560.

Tannoo, 2020-06-02:
Here is my UBL mesh edit screen from `bugfix-2.0.x` that I downloaded on 5/25/20.

Are you sure you are using the latest?

You also didn't state that you were using UBL either. My bad.

[image]

Tannoo, 2020-06-02:
Here is the closest I have is the babystepping / Z-offset:
[image]

Tannoo, 2020-06-02:
There were some timing lines you could add to see if it helps.

Tannoo, 2020-06-02:
My suggestion is to download a copy of `bugfix-2.0.x` at your earliest convenience, unzip, configure, and upload it.

See if that fixes it as I do not have any of the special timings in my config. But, that can just be the controller not being able to handle the normal timings.

bill-orange, 2020-06-02:
Yea, testing the bug fix is the next logical thing to try. Thanks for uploading the image. I did not know what the menu was supposed to look like. What is odd, is that only this menu is affected. My info screen looks normal. I don’t see what could be different about the graphics in this menu bs the info screen.  Perhaps @thinkyhead could tell us what might be different.

Tannoo, 2020-06-02:
I had a similar effect when I was updating from a year ago and after I posted about it on discord, I found the PR's that "fixed" it and another issue, I found that those PR's were applied AFTER I downloaded my copy. I applied those PR's to my copy and it corrected the issues I had.
So... I downloaded the latest (5-25-20) to update to any new features that I might be missing out on that I didn't know if I could live without or not.

So... always try a fresh copy of the bugfix branch as that is where all of the bugs are fixed. lol

bill-orange, 2020-06-02:
Sadly, same results with Bugfix.

Tannoo, 2020-06-02:
Okay. What exact menu option are you selecting?

Tannoo, 2020-06-03:
Menu > Motion > United Bed Leveling > Mesh Edit
That is what got me to the "Radar Map" edit screen.

I don't know where to find the edit screen you have. Especially when your config has `AUTO_BED_LEVELING_UBL` enabled.

bill-orange, 2020-06-03:
I got to it by:

Motion > United Bed Leveling  > UBL Tools > Edit Mesh > Fine Tune Closest

I think this menu is "reused"  several places in UBL.

Tannoo, 2020-06-03:
Here is my point edit screen with `USE_BIG_EDIT_FONT` disabled:

bill-orange, 2020-06-03:
> Did you just "drag and drop" your old configs into the newest bugfix?  I confess, I did.  I will compare the configs as time permits.  Even with good tools it can take a while. I generally use the compare add-on in Notepad++.  Its easier on my old eye.

Tannoo, 2020-06-04:
10 mins tops unless your old configs are _really_ old.

Tannoo, 2020-06-04:
I see that you have `MESH_EDIT_GFX_OVERLAY` enabled. I am trying that now.

bill-orange, 2020-06-04:
I did a fresh build on Bugfix and experienced the same problem.  Oddly, I am a little more cut-off than you are.  If you can hunt down the problem, that would be great.  I had a hard time understanding the structure of the menu building code, when I had a look last week.  

@thinkyhead you might want to look at this, since we now have confirmation of this minor bug.

Tannoo, 2020-06-04:
Well, I have tried and tried.... I am at a loss. I have jacked up my code so much that I likely need to reload the bugfix branch into my printer.

For the time being, disable `MESH_EDIT_GFX_OVERLAY` so you can see the numbers.

Tannoo, 2020-06-04:
`BABYSTEP_ZPROBE_GFX_OVERLAY` option and the babystepping screen works just fine.

`MESH_EDIT_GFX_OVERLAY` option and it's screen doesn't work properly.

boelle, 2020-06-21:
@bill-orange still an issue?

bill-orange, 2020-06-21:
Yes, this is still an issue. It appears to be a confirmed bug. I am not working on it because tracking down the error exceeds my skill. I am hoping that someone with greater knowledge of the structure of the LCD components of Marlin will pick this up. Mercifully, the errors in generating the LCD screen do not affect functionality.

thinkyhead, 2020-06-21:
For screens that self-manage from within the menu system I've taken care to ensure that when you click a button to go to a new screen that the screen change happens at the right time to ensure at least one complete screen update. This is not a screen that I implemented, and the screen change event seems to be done external to the regular screen update loop. So, I'll need to find a clever way to ensure that screen changes external to the screen update loop also have the correct timing, or at least do an extra update.

bill-orange, 2020-06-21:
@thinkyhead Thanks. When you need me to test, I have a Tronxy X3A and an Anet A8 that exhibit the problem with the discount full graphics display and  SKR1.3.

Tannoo, 2020-06-21:
Just to note, I am using a Mega2560 with RAMPS.

thinkyhead, 2020-06-21:
Please give this a try and see if it helps. #18373

bill-orange, 2020-06-21:
Sorry, no difference in display behavior.

With `//#define MESH_EDIT_GFX_OVERLAY` commented in or out the display looks like the picture in my first post on this issue.  
Did you do anything that would change the encoder timing?  It seems better, no corruption in the numbers and no missed encoder ticks.

bill-orange, 2020-06-24:
on my system, the latest changes in bugfix have not affected the problem.  However, the encoder's behavior is greatly improved.  It is interesting that the mesh editor  screen is the only one with this problem.

Perhaps @Tannoo should retest also.

bill-orange, 2020-06-28:
@thinkyhead I tested the branch Marlin-bf2_simpler_mesh_edit_PR this morning and the EDIT display now functions normally.  Thanks.  If I turn the encoder wheel quickly, I can still corrupt the digits.  To get the digit corruption the encoder wheel needs to be turned faster than a wise person would actually turn it in real use.  There is therefore a pretty minor issue.

boelle, 2020-07-02:
@bill-orange so as soon @thinkyhead merges #18373 we can swing the axe on this one?

bill-orange, 2020-07-02:
I believe so.  I briefly tested the un-merged working version and it corrected the display problem on my machine.

sjasonsmith, 2020-07-03:
@bill-orange, can you close this once you are able to test the merged change? If we don't hear back in a couple days we will likely presume it is fixed and close it for you.

bill-orange, 2020-07-03:
I see no reason not to close it now. If a problem is found later it probably should start as a new issue anyway.

closed 2020-07-03 by bill-orange