MarlinFirmware/Marlin #18180: [BUG] MKS MINI 12864 + SKR 1.3 - all black pixels from time to time.

Pajacyk0v, 2020-06-02:
### Bug Description

From time to time my MKS MINI 12864 goes black. I mean all pixels on screen are black. Print form SD card continues, no reaction for encoder. LCD returns to normal operation after some time. Boot screen always appear, no issues with contrast.

Same behavior has been reported here https://github.com/MarlinFirmware/Marlin/issues/13550#issuecomment-544490824

### My Configurations

Marlin 2.0.5.3 

#define MKS_MINI_12864
#define SDSUPPORT
Rest of LCD section settings are defaults.

### Steps to Reproduce

It occurs randomly. During print or not.

#### Additional Information

[image]

Mainboard BTT SRK 1.3

Parkrcz, 2020-06-14:
What I have found in my case are basically two type of glitches. First = all black pixels(only restart works), and second =  empty screen (kind of "all white pixels"). To get rid of second one, removing SD card works. Once I remove SD card from MKS_Mini_12864 v2 panel, "all white pixels" glitch is gone.

Pajacyk0v, 2020-06-15:
> What I have found in my case are basically two type of glitches. First = all black pixels(only restart works), and second = empty screen (kind of "all white pixels"). To get rid of second one, [...]

I have same issues.

boelle, 2020-06-23:
@Pajacyk0v 
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

Parkrcz, 2020-06-24:
Hello, yesterday I compiled the latest bugfix-2.0.x branch, but glitches are still present.

ils15, 2020-06-25:
same problem here

sjasonsmith, 2020-07-04:
Can one or more of you attach your configuration files which reproduce this issue?

Also, are there any ideas to make this reproduce easily, or just power-on and wait?

gerleimarci, 2020-07-14:
Same issue here with the latest bugfix-2.0.x
It produces various glitches and gives freezed blank display during SD read.
I tried `#define FORCE_SOFT_SPI` but it just make things worse. With this option the display freezes at the moment I plug in the SD card.

MKS Robin E3D
CR20 display (Basically a MKS MINI12864 without SD socket)

The display uses the same SPI pins as the onboard SD card reader.

```
#define DOGLCD_SCK                      PB13
#define DOGLCD_MOSI                     PB15

#define SCK_PIN                             PB13
#define MOSI_PIN                            PB15```

anomalchik, 2020-07-24:
MKS_SGEN_L (LPC1768) with MKS MINI12864 have same issue too

anomalchik, 2020-07-27:
> Can one or more of you attach your configuration files which reproduce this issue?
> 
> Also, are there any ideas to make this reproduce easily, or just power-on and wait?

Choose file from sdcard and wait...

It also breaks read from sdcard in printing state. i choose big gcode file (12mb) with infill 100% and i get constant execution of the same buffer of code (~ after 2 hours / 10). 
Same gcode file print succeful through serial port. 
I tried to change sdcard from onboard to lcd but it didn't fix the situation. 
Small files can be printed without problems. 
[image]

anomalchik, 2020-07-27:
My config https://drive.google.com/file/d/17ZGIfQ7rwg7q7aJQr0mfHvPpLTpVBg2Y/view?usp=sharing

leftover, 2020-08-16:
Same here. SKR 1.3 + TMC2209 + MKS MINI 12864 V1.0. 
Actually there are 2 types of glitch: "black screen" and "white screen". Both happen randomly. White screen disappears with knob turn while black one can be "fixed" with reset only.
My current configs:
[Marlin_cfgs.zip](https://github.com/MarlinFirmware/Marlin/files/5080178/Marlin_cfgs.zip)

Pajacyk0v, 2020-09-19:
I had same issue on 2.0.6.1 :(

Pajacyk0v, 2020-10-10:
In 2.0.7 problem still exists.

closed 2020-11-18 by github-actions[bot]