MarlinFirmware/Marlin #18181: If you change micro stepping from 16 to 32 all the stepper motors run reverse and will not home

Crunch69, 2020-06-02:
**Bug Description**
If you change micro stepping from 16 to 32 all the stepper motors run reverse and will not home

**My Configurations**
Marlin 2.0.x bugfix
SKR 1.3 TMC2209 Sensor less Homing

[32Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4720025/32Configuration.zip)

**Steps to Reproduce**
Change micro stepping from  16 to 32
Expected behavior:
Finer micro stepping

**Actual behavior:**
All steppers run in reverse

boelle, 2020-06-21:
@Crunch69 still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

thinkyhead, 2020-06-23:
Trinamic drivers must be set to 16x micro-stepping unless there is a means to change the micro-steps provided on the board.

closed 2020-06-23 by thinkyhead