MarlinFirmware/Marlin #18182: Selectng Improve Homing Reliability disables all steppers

Crunch69, 2020-06-02:
**Bug Description**
Selecting Improve Homing Reliability disables all steppers

**My Configurations**
Marlin 2.0.x bugfix
SKR 1.3 TMC2209 Sensor less Homing

**Steps to Reproduce**
Uncomment #define Improve Homing reliability

**Expected behavior**:
Tuning of acceleration and jerk when homing and add a guard period for end stop triggering.

**Actual behavior:**
All steppers lock and will not run.

boelle, 2020-06-21:
@Crunch69 still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle