MarlinFirmware/Marlin #18186: [BUG] Calculation of junction_cos_theta in JD returns inaccurateresults

XDA-Bam, 2020-06-03:
### Bug Description

While debugging the JD LUT implementation, I used a couple of Octave/MATLAB scripts to predict and check junction angles, including the values of `junction_cos_theta` and `junction_theta`. I noticed that my predictions for `junction_theta` were a couple of percent different from the Marlin results. I then realized, that the actual difference came not from the `acos()`-implementation, but from surprisingly different values for `junction_cos_theta`. Here's an example, using [`JD_single_arc_test_1x.gcode`](https://github.com/MarlinFirmware/Marlin/files/4722807/JD_single_arc_test_1x.zip) (excerpt from [Teaching Tech speed test](https://www.thingiverse.com/thing:3071464/files)):
[image]
And here's the resulting error in percent:
[image]

In the plots, `Theta = 180° - junction_theta` is shown. Note that I "dumbed down" Octave to single precision fp for this calculation, to give both contenders a fighting chance. I exported both cosine results, capped to 6 digits after the decimal point, ran those through the precise Octave `acos()` and plotted them. So differences are not due to errors in Marlin `acos()`, more digits for Octave, or anything similar.

Considering we recently fought hard to bring down the error in Marlin's `acos()`-approximation to below 0.7%, it seems unreasonable to have an error of ~2% introduced from other sources. Also, mind that the error resulting in `limit_sqr` is much, much larger, because we're using `1/(180°-Theta)` to calculate that. For example, junction 36 has a precise (Octave) angle of 176.83°, while Marlin computes 173.37°. That means, `limit_sqr` is multiplied by a factor of 0.1508 instead of the correct factor of 0.3155. That's a whopping decrease of **52.2%** in junction speed!

Consequently, I see this problem as another source of stutter in the current JD implementation. There's systematic problems when using `limit_sqr`, which are not easy to fix. And there's this, which should be avoidable, right?

After taking a look in the code, it seems like we're converting precise GCode to steps (-> rounding) store them as `int32_t`, then calculate the difference vectors for each move @ https://github.com/MarlinFirmware/Marlin/blob/8477dbf65db4572d6aac07f5c732130607f31545/Marlin/src/module/planner.cpp#L1709 (-> further loosing precision due to full step calculations) and then we re-convert to float @ https://github.com/MarlinFirmware/Marlin/blob/8477dbf65db4572d6aac07f5c732130607f31545/Marlin/src/module/planner.cpp#L1857
(-> further slight loss in precision) which we then use for all following calculations, including everything in JD.

I think I need a little help with this one. Is there a reason we're not directly calculating our JD unit vector from the precise GCode coordinates (after calculating the difference vector for each move)?

### My Configurations

Ender 3 Pro, stock Creality Melzi board, ATMega 1284p
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4722896/Configs.zip)

### Steps to Reproduce

1. Download current bugfix 2.0.x @ 8477dbf
2. Add `SERIAL_ECHOLNPAIR_F("Cos(Theta) [-]: ", junction_cos_theta, 6);` after https://github.com/MarlinFirmware/Marlin/blob/8477dbf65db4572d6aac07f5c732130607f31545/Marlin/src/module/planner.cpp#L2288
3. Print [`JD_single_arc_test_1x.gcode`](https://github.com/MarlinFirmware/Marlin/files/4722807/JD_single_arc_test_1x.zip) while running a serial monitor
4. Compare Marlin cosine to precise result (using software of your choice - Excel may work, MATLAB or Octave recommended)

**Expected behavior:** Results should be close to identical within single precision fp limits.

**Actual behavior:** Results deviate starting at the third to fourth decimal, resulting in deviations of `junction_cos_theta` of up to 2% (more may be possible with different input GCode).

AnHardt, 2020-06-03:
Converting to steps and back to mm seems to be right to me.
Because of the necessary discretization to steps we can't follow the theoretically correct g-code. We should really calculate upon what speeds/accelerations/... we plan to really do, not what we are ordered do (what we can't do).

Small segments are poison. That's the nature of discretization. 
Assuming our segment would only be 1 (or  sqrt(2) ) step long we have only 8 different  surrounding target points. The angle between of them is 45° - the potential error up to 22.5°. The length can differ even more. Because of rounding every point is the center of a square from -1/2 to +1/2 steps around it the error may be up to 1 step in length (per axis. (diagonally sqrt(2) ) ). If the segment is only 1 step long that's up to 100%.
When the segments length grow the relative error in length and the absolut error in angle are shrinking rapidly.

AnHardt, 2020-06-03:
We can only come out of the above dilemma when the slicer/cam does the rasterization for us. But then the slicer/cam has the same problem.
Doubling the resolution shrinks the absolute errors by half. But then the users tend to send even shorter segments :-O .
In my experience only a (ideally hidden) setting in the slicer, limiting the segments length, does help.

XDA-Bam, 2020-06-03:
Is the step size Marlin handles the size of one full motor step, or one micro step?

AnHardt, 2020-06-03:
Micro! One pulse on the steppers step_pin. What you configure in DEFAULT_AXIS_STEPS_PER_UNIT or over rule with M92.
```cpp
/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2...]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 500 }
```

closed 2020-06-05 by XDA-Bam