MarlinFirmware/Marlin #18191: [BUG] G-code M25 does not work

DrumClock, 2020-06-03:
If there is an M25 command in the G-code or it is sent via the terminal, the printer will not pause printing correctly and the process will freeze - the printer RESTART is required.
Pause printing is performed correctly only through the menu on the LCD panel.

**Behavior description:**

  with M25 command in G code
1) print from SD card
2) G-code M25 (in file)
3) parking is performed at the specified position X, Y, Z
4) The print head returns above the object
5) The LCD panel "freezes" and the printer must RESTART

with the M25 command from the terminal
1) print from SD card
2) sent G-code M25 from the terminal
3) parking is performed at the specified position X, Y, Z
4) The print head returns "very slowly" above the object
5) The LCD panel "freezes" and the printer must RESTART

[configuration 3.6.2020.zip](https://github.com/MarlinFirmware/Marlin/files/4725302/configuration.3.6.2020.zip)

ellensp, 2020-06-04:
You need to enable HOST_ACTION_COMMANDS so you can un pause it from console.

DrumClock, 2020-06-04:
Hi,
**I enabled HOST_ACTION_COMMANDS and nothing has changed.**

Interesting that in CONFIGURATION_ADV_H_VERSION 020000 of STRING_DISTRIBUTION_DATE "2019-08-26" the function was disabled and it worked without problems from the terminal or from the file.

I am currently using CONFIGURATION_ADV_H_VERSION 020005 of
STRING_DISTRIBUTION_DATE "2019-07-10"

DrumClock, 2020-06-04:
I tried the latest version of bugfix 2.0.x and **nothing has changed.**
- CONFIGURATION_H_VERSION 020006
- CONFIGURATION_ADV_H_VERSION 020006

Gornerrrrr, 2020-06-08:
I have same problem

qwewer0, 2020-06-08:
I have this too.
SKR Mini E3 v1.2
Latest Bugfix.

DrumClock, 2020-06-08:
**Thank you for the response, we have to wait for the programmers we will not solve it, 
I hope it will be soon.**

boelle, 2020-06-23:
@DrumClock 
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

thinkyhead, 2020-06-24:
An `M25` inside of a G-code file is interesting. Does `M125` work any better? I wish I had a working debugger, or logging in exactly the part of the code where it could tell us what is happening. Without more detailed data it is hard to know what's going on. At some point, myself or one of the community members with the requisite time and talent will investigate the issue and try to narrow it down.

DrumClock, 2020-06-24:
**I will be happy to help you find bugs in Marlin.**
Unfortunately, I'm not a programmer and I'm happy to set up configuration files and upload the FW on the board with using Visual Studio Code, forgive me. I can describe in the behavior of the printer for a certain situation.

**G-codes M25 or M125 behave exactly the same. As described above.**
Printing is interrupted, the head moves to the park position, then returns above printing and the entire printer freezes.

Example G-code in file:
[image]

Gornerrrrr, 2020-06-24:
use balck_stm32f407ve,Latest ver marlin2.0
I changed the program
in pause.cpp line458 change to
`  do{
    planner.synchronize();
    idle();
    queue.advance();
  }
  while(queue.has_commands_queued());`
It's now to pause correctly in my printer.
So I think the bug was caused by instructions that didn't properly process the cache queue.

DrumClock, 2020-06-24:
**My friend helped me edit the pause.cpp program.**

[image]

### Parking still does not work and the printer freezes.

1 print
2 M25 command in G-code
3 parking heads
4 return above the print position
**5 --- filament extrusion ---**
6 print lock - only printer RESTART works

**So it doesn't work for me :(**
I have a RUMBA ATmega2560 board.

qwewer0, 2020-06-27:
Pause via LCD works for me, but M125 is the same as M25.

> do{ planner.synchronize(); queue.advance(); } while(queue.has_commands_queued());

Didn't help.

DrumClock, 2020-07-02:
Due to minimal responses to solving this problem, I closed this issues. The bug is still at 1.7.2020 in Marlin FW.

DrumClock, 2020-07-13:
```
* Marlin-bugfix-2.0.x 
* Downloaded:	13.7.2020
```
### G-code M25 or M125 in the file still does not work.

Test: https://www.youtube.com/watch?v=XJfbxOgrXVM

thinkyhead, 2020-07-20:
We've recently updated this area of the code.  Please test the latest [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands.

qwewer0, 2020-07-20:
@thinkyhead Tested the latest Bugfix-2.0.x cea097d, while it did change the behavior from slowly drifting back to the printing XYZ spot, but now after parking the head, it goes back printing (XY) with `NOZZLE_PARK_XY_FEEDRATE` speed, on the parking z height, and it still freezes after a couple printing moves.

DrumClock, 2020-08-03:
Hi @thinkyhead 
Tested with the version distributed on 2020-08-02
### Still the same behavior error still uncorrected !!
G-code M25 or M125 in file = print pause does not work, printer freezes.

qwewer0, 2020-08-03:
It seems that it works for Chuck Hellebuyck (CHEP) on an Ender 3 v2, [video](https://www.youtube.com/watch?v=dLZm2itXJkg&feature=youtu.be&t=124), but it is still not working for me, 5e34110.

qwewer0, 2020-08-07:
It works for Chuck Hellebuyck (CHEP) on the same board as mine (SKR Mini E3 v1.2, [video](https://www.youtube.com/watch?v=CN6dOI6QxAs&feature=youtu.be&t=171)), but then why could it not working for me?

I will recheck and compare all the settings on the BTT firmware configs.

DrumClock, 2020-08-19:
### Bug still not fixed.

DrumClock, 2020-08-28:
@thinkyhead 
Bugfix 2.0.x - DISTRIBUTION DATE "2020-08-27"
### Bug still not fixed.

qwewer0, 2020-09-21:
On the latest Bugfix-2.0.x https://github.com/MarlinFirmware/Marlin/commit/fb48cd0ab47f3ed981d4c74814e850ff7dcd13ba, the problem is still present.

[M25_Test.zip](https://github.com/MarlinFirmware/Marlin/files/5254684/M25_Test.zip)
Made a (30 line long) dummy `M25` test gcode file, and the results are the same as with a "real" gcode file.
When running the `M25_Test.gcode` and the it reaching `M25` line, the printer parks the head, then slowly drifts back, but not to the XYZ position before M25, but rather to the next gcode lines XY position with the parking Z height, then after it executes one more line it just freezes, and all this while the LCD says "Parking...", and the nozzle kept at temperature.
With `HOTEND_IDLE_TIMEOUT` enabled and `HOTEND_IDLE_TIMEOUT_SEC` set to 5*60, after the printer stayed frozen for 5 min the printer switched off the heating.

DrumClock, 2020-09-28:
Hi, Looks like no one cares about this bug!
Is anyone solving this question or should I close it?

thisiskeithb, 2020-09-28:
Keep in mind that all the code here is contributed by unpaid volunteers who work on it in their free time.

I've added a Help Wanted label to hopefully attract attention of someone who has the hardware, interest, and ability to fix/implement the change.

DrumClock, 2020-10-17:
Hi @thinkyhead 
In the bugfix distribution from "2020-10-16" this bug is still not fixed.
There is a big bug in GitHub not feedback on whether someone is working on the removal.
The problem is reported from June 3, 2020.

Can you look at it, please?

thank you very much

rhapsodyv, 2020-10-17:
@DrumClock, can you do a simple test?

Use `M125 P`, instead of `M25` or `M125`.

Can you test it?

qwewer0, 2020-10-18:
`M125 P` does the [same](https://github.com/MarlinFirmware/Marlin/issues/18191#issuecomment-696027789) as M25 for me.

rhapsodyv, 2020-10-18:
> `M125 P` does the [same](https://github.com/MarlinFirmware/Marlin/issues/18191#issuecomment-696027789) as M25 for me.

M125 without P only wait for the user input if printing using serial. 
It needs P (plus lcd menus) to wait for user input.

So, if we use M125 without P while printing with SD, it will be forever paused, and never goes back to print. I will try unpaused using serial.

rhapsodyv, 2020-10-18:
The code:
```cpp
  const bool sd_printing = TERN0(SDSUPPORT, IS_SD_PRINTING());
  TERN_(HAS_LCD_MENU, lcd_pause_show_message(PAUSE_MESSAGE_PARKING, PAUSE_MODE_PAUSE_PRINT));
  const bool show_lcd = TERN0(HAS_LCD_MENU, parser.seenval('P'));

  if (pause_print(retract, park_point, 0, show_lcd)) {
    TERN_(POWER_LOSS_RECOVERY, if (recovery.enabled) recovery.save(true));
    if (ENABLED(EXTENSIBLE_UI) || !sd_printing || show_lcd) {  // <<<<<<<<<<<<<----
      wait_for_confirmation(false, 0);   // <<<<<<<<<<<<<----
      resume_print(0, 0, -retract, 0);   // <<<<<<<<<<<<<----
    }
  }
```

DrumClock, 2020-10-18:
Hi @rhapsodyv 
nice code, unfortunately not telling me anything because I'm not a programmer.
I can test it and it will be fixed in FW bugfix 2.0.x.
I won't do anything about it until then :)

qwewer0, 2020-10-18:
@rhapsodyv You got it. I run my M25 and M125 P test file through serial, and both file (M25, M125 P) paused, waited and I was able to resume correctly.

rhapsodyv, 2020-10-18:
> Hi @rhapsodyv 
> 
> nice code, unfortunately not telling me anything because I'm not a programmer.
> 
> I can test it and it will be fixed in FW bugfix 2.0.x.
> 
> I won't do anything about it until then :)

The `P` were always there. This is not a fix that I did. I just discovered it, doing some debugging while trying to fix your issue.

So, did `M125 P` works as expected and fix your problem?

qwewer0, 2020-10-18:
Not sure why, but with https://github.com/MarlinFirmware/Marlin/pull/19803, both M25 and M125 P works, just like from serial.

rhapsodyv, 2020-10-18:
There's no fix needed. Just use `M125 P` in the gcode file. Nothing more.

rhapsodyv, 2020-10-18:
The only issue here, is the `P` parameter for `M25` and `M125` is not documented.....

qwewer0, 2020-10-18:
> `M125 P` does the [same](https://github.com/MarlinFirmware/Marlin/issues/18191#issuecomment-696027789) as M25 for me. So, not working.

Like I wrote it, M125 P wasn't working.

rhapsodyv, 2020-10-18:
`M125 P1` will fix

rhapsodyv, 2020-10-18:
`seenval` needs a number. Yesterday was late and I didn't see it. `seenbool` don't need a value. 
Just try with `M125 P1` what it will work.

Sorry about the "only" `P`.

qwewer0, 2020-10-18:
> seenval needs a number. Yesterday was late and I didn't see it. seenbool don't need a value.
> Just try with M125 P1 what it will work.
> 
> Sorry about the "only" P.

@rhapsodyv Got it. M125 P1 works without any change, but why could M25 work with https://github.com/MarlinFirmware/Marlin/pull/19803?
And M125 P1 doesn't fixes the problem with M25 only bypasses it.

qwewer0, 2020-10-18:
In M125.cpp, why did it have !sd_printing, shouldn't it be without exclamation mark to work on LCD call?

rhapsodyv, 2020-10-18:
> @rhapsodyv Got it. M125 P1 works without any change, but why could M25 work with #19803?
> And M125 P1 doesn't fixes the problem with M25 only bypasses it.

M25 and M125 are two tweens that don't like much each other.

When `PARK_HEAD_ON_PAUSE` is **enabled**, `M25`===`M125`. So, `P1` will work for both `M125` and `M25`.

When `PARK_HEAD_ON_PAUSE` is **disabled**, `M25` have it's own code that pause the print, don't accept `P` parameter, will not wait for any user interaction, and need a new command to unpause (`M24`). 

So, for the use case in this issue, the correct is to use `M125`, as it is the only pause gcode that allow interaction with the user.

rhapsodyv, 2020-10-18:
> In M125.cpp, why did it have `!sd_printing`, shouldn't it be without exclamation mark to work on LCD call?

It's designed to work in two ways: wait for the user interaction OR wait for a new gcode to resume printing.

This is the reason we have a case that it just stay forever, waiting for the resume gcode. I didn't tested if the M24 resume the print... but that is the intention.

qwewer0, 2020-10-18:
The problem is that it doesn't stays parked:

> When running the M25_Test.gcode and the it reaching M25 line, the printer parks the head, then slowly drifts back, but not to the XYZ position before M25, but rather to the next gcode lines XY [...]

rhapsodyv, 2020-10-18:
Did `M125 P1` fixed the problem?

qwewer0, 2020-10-18:
Without https://github.com/MarlinFirmware/Marlin/pull/19803, M25 does not work (https://github.com/MarlinFirmware/Marlin/issues/18191#issuecomment-696027789), and M125 P1 works, but with https://github.com/MarlinFirmware/Marlin/pull/19803 both M25 and M125 P1 works.

If I read the issue title right, the problem isn't with M125, but with M25. So M25 doesn't calls M125 the right way with `PARK_HEAD_ON_PAUSE` enabled.

rhapsodyv, 2020-10-18:
M25/M24 aren't  intended to use inside Gcode file, as it stop reading the file and don't ask user any action. It's designed to work in a control serial. To control the printer, not inside the printing file.

`M125 P1` do works inside a gcode file, because it have an option to wait for the user input.
And, as you said, it works.

#19803 may break existing usages, as it change the behaviour that is already supported by `P` option.

So, this issue is fixed, right?

qwewer0, 2020-10-18:
> `M125 P1` do works inside a gcode file, because it have an option to wait for the user input.
> And, as you said, it works.

What do you mean? M25 with PARK_HEAD_ON_PAUSE enabled is practically only this:

```
/**
 * M25: Pause SD Print
 */
void GcodeSuite::M25() {

  #if ENABLED(PARK_HEAD_ON_PAUSE)

    M125();

  #endif
}
```

So M25 just passes all the work to M125, but in a way that right now isn't working for SD printing.

rhapsodyv, 2020-10-18:
I just tested `M25/M125` without the `P1`. And it waits until I send a `M24`, that is the correct behaviour.
I tested `M25 P1` and `M125 P1` and it waits for the user interaction in the LCD.

Behaviour (1) Pause (M125/M25) / Unpause (M24) sending using a control serial.

Behaviour (2) Pause (M25 P1 / M125 P1) from anywhere and use the LCD to unpause.

Both working.

qwewer0, 2020-10-18:
`M25 P1` does work, so if `M25` without `P1` is intend to not stay parked and to freeze the printer so it is unusable without a restart, then maybe we just need to update [M25 - Pause SD print](https://marlinfw.org/docs/gcode/M025.html) page. (sarcasm)

Behavior (1): Pause (`M125`/`M25`) -> Printer returns immediately, without user interaction and to the wrong position on the printing area, and freezes and needs a restart.

Behavior (2) Pause (M25 P1/M125 P1) from anywhere and use the LCD to unpause.

Without https://github.com/MarlinFirmware/Marlin/pull/19803 only those work with `P1`.

rhapsodyv, 2020-10-18:
> `M25 P1` does work, so if `M25` without `P1` is intend to not stay parked and to freeze the printer so it is unusable without a restart, then maybe we just need to update [M25 - Pause SD print](https://marlinfw.org/docs/gcode/M025.html) page. (sarcasm)

`M25` without P will wait forever for a `M24` in the command serial.... As I said above. There're two distinct operations: pause/resume using lcd and pause/resuming with gcode.

When using `M25` without P, did you try sending a M24 using serial to unpause it??

thinkyhead, 2020-10-18:
Reference: #12893

qwewer0, 2020-10-18:
I tried to send M24 to the printer through serial after the it "parked" from M25 (https://github.com/MarlinFirmware/Marlin/issues/18191#issuecomment-711418514), and M24 makes the printer to come out of the "freeze" and to show Purge More or to Continue. After I hit Continue it goes back to the point (XYZ) where it was paused, but then it "skips" to that gcode line where it stopped in the printing area after it drifted back and executed 1-2 more gcode lines before it decides to just stop on top of the print.

But at lease we made some progress, as now I know why it "froze".

If needed I can make a step by step explanation on how it "works" on my printer.

rhapsodyv, 2020-10-18:
This issue can be closed, as it is not a bug, just a missing parameter.

---
About the M25 / M24 combo not working for you:

I lost a few hours to setup a env, test, debug, to find that `P1` parameter, trying to help with this issue.

Then I was totally discredited and even need to read sarcasms about what I was saying.

@thinkyhead needed to came here and just show what I was saying for the last hour, to you be convinced.

Sorry, but I will not help you with that. 

I'm not doing free unpaid work to receive any rudeness in exchange.

Good luck getting help.

qwewer0, 2020-10-18:
I'm sorry that I hurt you, didn't meant to do any harm.

qwewer0, 2020-10-18:
@rhapsodyv And I really appreciate your help even if I seem ungrateful.

qwewer0, 2020-10-18:
In my surroundings a sarcazm is a "great" way to explain things to the other.

qwewer0, 2020-10-18:
If @DrumClock says that @rhapsodyv great tip to use P1 fixes the problem then I'm behind the idea to closed this issue.

PS: If to change your mind @rhapsodyv would mean that I wouldn't able to help with marlin, then so be it. Your work worth way more than mine. So please don't leave Marlin just because my idiocy.

thisiskeithb, 2020-10-18:
@DrumClock / @qwewer0 Please open an issue in the [Marlin Documentation repo](https://github.com/MarlinFirmware/MarlinDocumentation) and reference the missing `P1` parameter for `M25`/`M125` so it can be documented.

closed 2020-10-18 by thisiskeithb

DrumClock, 2020-10-19:
Hi @thisiskeithb 
I will not mention the `P1 `parameter anywhere because it is not my idea.
And I don't know how and who figured it out @rhapsodyv .

**I just pointed out the G-code M25 in file which worked without problems 
until the bugfix 1.1.x ............ then something went wrong in the FW !!!!**

### I am willing to continue testing the FW to fix bugs that are there and have not been before.

Petr

qwewer0, 2020-10-19:
@DrumClock Could you take a video of the M25 problem?

[14 more replies]