MarlinFirmware/Marlin #18192: [BUG] POWER_LOSS_RECOVERY

DrumClock, 2020-06-03:
printer configuration:
UPS backup power supply
two extruders T0 and T1

**Behavior description:**

1) prints the T0 tool
2) power off
3) heating HB, HE T0 and HE T1 are switched off
4) the Z axis moves away by 20 mm

**BUG
POWER_LOSS_RETRACT_LEN 10 (mm) not performed
so that it does not solidify in the HotEnd coolers of tools T0 and T1**

5) power ON
6) resume printing from the LCD panel
7) setting position X, Y min (endstops)
8) heat HB
9) heat HE tools T0
9) heat HE tools T1
10) filament to extract  by 20 mm (tool T1)
11) filament to retract by 10 mm (tool T1)
12) move to the last print position
13) filament to extract  by 10 mm (tool T1)
14) **BUG - printing with tool T1 into the material of tool T0**

if printing is interrupted by the T1 tool, it continues with the T1 tool

[configuration 3.6.2020.zip](https://github.com/MarlinFirmware/Marlin/files/4725546/configuration.3.6.2020.zip)

DrumClock, 2020-06-18:
Hi, what about the error of printing tool T1 to the position where printing was tool T0?

sjasonsmith, 2020-06-20:
I am guessing from your configuration file name that it is not the latest Marlin?
Can you try this from bugfix-2.0.x, and report whether it is still an issue? Posting configurations which work with the latest bugfix will make it easier for people to use for debugging.

If you disable your second extruder (set EXTRUDERS to 1 in Configuration.h), can you still reproduce the problem on T0? Requiring only one extruder will also make it easier for people to help.

DrumClock, 2020-06-21:
The latest version of bugfix 2.0.x cannot be compiled. Downloaded 21.6.2020 at 13:18 (Prague) compiled with Arduino 1.8.9.
[configuration 21.6.2020.zip](https://github.com/MarlinFirmware/Marlin/files/4809366/configuration.21.6.2020.zip)
Error:
avr-gcc: error: CreateProcess: No such file or directory
exit status 1

This version from 4.6.2020 can be compiled and the error is as follows:
[Configuration_4.6.2020.zip](https://github.com/MarlinFirmware/Marlin/files/4809367/Configuration_4.6.2020.zip)

My Configurations

Build version bugfix-2.0.x
Compiled with Arduino 1.8.9

Backup power supply
Extruders T0 and T1

-------- If the G-code for tools is T0 or T1:  --------

1. T0 printing in progress
2. blackout (power off)
3. HB heating off
4. T0 heating off 
5. Z lift on 20 mm

BUG ?
- not off Part-Cooling Fan
- not retraction filament

RESTART PRINTER

1.  resume printing from the LCD
2.  T0 set position X and Y min
3.  HB heating
4.  T0 heating 
5.  T0 extrude 20 mm filament
6.  T0 retraction 10 mm filament
7.  T0 move to last print X,Y position
8.  lift Z to last ptint position
9.  T0 extrude 10 mm filament 
10. T0 printing continues correctly to T0 material

For tool T1 the procedure is the same as for T0

--------  If the G-code for tools T0 and T1 ---------
This error occurs when printing resumes.

1. If printing is interrupted while printing T0.
   When restored, tool T1 is printed in the material for T0.

2. If printing is interrupted while the T1 tool is printing.
   When restoring, printing is OK.

boelle, 2020-06-21:
> Downloaded 21.6.2020 at 13:18 (Prague) compiled with Arduino 1.8.9.

have you tried vscode+platform.io?

DrumClock, 2020-06-21:
Hi, I only use Arduino.
I never used vscode and platform.io.

boelle, 2020-06-21:
try it

arduino is old and no one use it anymore

DrumClock, 2020-06-21:
Hi, 
I'm trying Visual Studio Code + PlatformIO but I have a problem:
**Cannot find d: \ Projects \ Marlin FW \ _TEST_ \ Marlin-bugfix-2.0.x \ include.**

DrumClock, 2020-06-21:
**Compilation OK**
My Configurations

Build version bugfix-2.0.x
Downloaded 21.6.2020 at 13:18 (Prague) 
Compiled with Visual Studio Code + PlatformIO

Backup power supply
Extruders T0 and T1

**-------- If the G-code for tools is T0 or T1:  --------**

1. T0 printing in progress
2. blackout (power off)
3. HB heating off
4. T0 heating off 
5. Z lift on 20 mm

**BUG 
not off Part-Cooling Fan
not retraction filament**

RESTART PRINTER

1.  resume printing from the LCD
2.  T0 set position X and Y min
3.  HB heating
4.  T0 heating 
5.  T0 extrude 20 mm filament
6.  T0 retraction 10 mm filament
7.  T0 move to last print X,Y position
8.  lift Z to last ptint position
9.  T0 extrude 10 mm filament 
10. T0 printing continues correctly to T0 material

For tool T1 the procedure is the same as for T0

***************************************************************

**--------  If the G-code for tools T0 and T1 ---------**

1. **T0 printing in progress**
2. blackout (power off)
3. HB heating off
4. T0 and T1 heating off 
5. Z lift on 20 mm

**BUG 
not off Part-Cooling Fan
not retraction T0 filament
not retraction T1 filament**

RESTART PRINTER

1.  resume printing from the LCD
2.  set position X and Y min
3.  HB heating
4.  T0 heating 

**BUG 
not extrude 20 mm filament T0**

5.  T1 heating 
6.  T1 extrude 20 mm filament
7.  T1 retraction 10 mm filament
8.  T1 move to last print X,Y position
9.  lift Z to last ptint position

**BUG
HOTEND Z offset T1 is not performed**

10. T1 extrude 10 mm filament 

**BUG 
tool T1 is printed in the material for T0**

***************************************************************

1. **T1 printing in progress**
2. blackout (power off)
3. HB heating off
4. T0 and T1 heating off 
5. Z lift on 20 mm

**BUG 
 not off Part-Cooling Fan
 not retraction T0 filament
 not retraction T1 filament**

RESTART PRINTER

1.  resume printing from the LCD
2.  set position X and Y min
3.  HB heating
4.  T0 heating 

**BUG 
not extrude 20 mm filament T0**

5.  T1 heating 
6.  T1 extrude 20 mm filament
7.  T1 retraction 10 mm filament
8.  T1 move to last print X,Y position
9.  lift Z to last ptint position
    - HOTEND Z offset T1 is OK
10. T1 extrude 10 mm filament 
11. T1 printing continues correctly in the material for T1

sjasonsmith, 2020-06-21:
@DrumClock, did you try reproducing the error with only one extruder in your configuration file? That will greatly expand the number of people who can try to replicate the issue.

DrumClock, 2020-06-21:
What mistake do you think?
As described above, there are more errors.
When only one extruder (T0 or T1) is used in the G-code, there are fewer errors (description above)

DrumClock, 2020-06-21:
When printing with one or the other extruder, only this is:
**- does not turn off Part-Cooling Fan
- not retraction filament**
Printing resumes without error after recovery

DrumClock, 2020-07-02:
Due to minimal responses to solving this problem, I closed this issues. The bug is still at 1.7.2020 in Marlin FW.

sjasonsmith, 2020-07-06:
@DrumClock the PR referenced above should fix the retract in the single-extruder case. It would be great if you are able to try it.

You will probably still have problems when using two extruders. I would like to hear what is current behavior when you test my fix. I do not currently have hardware I can use for multi-tool testing, but I might be able to assemble something if there are still issues.

DrumClock, 2020-07-06:
@sjasonsmith What version should I use for testing?
Marlin-2.0.x or Marlin-bugfix-2.0.x
Is there a different powerloss.cpp file in each?

DrumClock, 2020-07-06:
@sjasonsmith 👍 
I edited the powerloss.cpp file according to # 18558 - red lines erased green added  [powerloss.zip](https://github.com/MarlinFirmware/Marlin/files/4879192/powerloss.zip)

### Actual behavior single extruder test :

1. printing in progress     			`nozzle T0`
2. blackout (power OFF)      			`POWER_LOSS_PIN  `
3. HotBed heating off
4. HotEnd heating off
5. **does not turn off Part-Cooling Fan**
6. retraction filament 10 mm   		        `POWER_LOSS_RETRACT_LEN ` 
7. Z lift on 20 mm           			`POWER_LOSS_ZRAISE`

RESTART PRINTER (power ON)

1. resume printing from the LCD
2. axis X,Y moving to 0 position 
3. waiting for HotBed heating
4. waiting for HotEnd heating
5. turns on Part-Cooling Fan 
6. extrude 20 mm filament      			`POWER_LOSS_PURGE_LEN `  
7. **retraction 10 mm filament**  		
8. move axis X,Y to last print position
9. lift Z to last prtint position
10. **extrude 10 mm filament**  		
11. printing continues correctly 

### Why is a 10 mm filament used in steps 7 and 10 ?

boelle, 2020-07-06:
> @sjasonsmith What version should I use for testing?
> Marlin-2.0.x or Marlin-bugfix-2.0.x

always Marlin-bugfix-2.0.x

sjasonsmith, 2020-07-06:
> does not turn off Part-Cooling Fan

Turning off fans might contribute to heat creep and clogs, if the wrong fans are disabled. I don't think it is always save to assume that all the numbered fans controlled by M106/M107 are only part-cooling fans. There may be other fans that need to run as long as power remains. Something can surely be done here, but the correct fix is less obvious than the missed retract on power-loss.

> retraction 10 mm filament
move axis X,Y to last print position
lift Z to last prtint position
extrude 10 mm filament

After purging `POWER_LOSS_PURGE_LEN` this then retracts by `POWER_LOSS_RETRACT_LEN` prior to moving to the print.
In your case this excessive, but another variable would have to be added to make it something shorter.

Does the long retract cause problems for you, or is it something you can live with? Since I do not actually use this feature, I would rather leave adding new features to somebody who does.

DrumClock, 2020-07-06:
Yes, this solution is suitable for one extruder.

For the two extruders, continue here : https://github.com/MarlinFirmware/Marlin/pull/18558

closed 2020-07-10 by DrumClock