MarlinFirmware/Marlin #18193: [BUG] While Calibration of the Lin_ADVANCE E-Motor  stops at 2. and following lines

TailBoom, 2020-06-03:
### Bug Description

I was trying to calibrate the K-faktor in Marlin bugfix 2.0 for the LIN_ADVANCE at my Ender 3 pro. 
Therefore after enabling the function I created a configuration file and leveld the bed. The first line was printed without problems an then the E-Motor just stoped . so I tried another K-Factor-range and the result was the same. 
If the motor doesn't stop becouse of other settings I tried then all the lines are looking as the same as the first line in the picture.
(I istalled the motherboard 1.1.5 before using Marlin and disabled a few functions as per that website: https://www.crosslink.io/marlin-2-0-memory-usage-by-feature/ )

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4725469/Configuration.zip)

### Steps to Reproduce

1. (once: leveling bed using MESH-BED_LEVELING)
2. create a configuration file using online tool
3. print

**Expected behavior:** 
Different lines as in the youtube videos about Linear Advance calibration

**Actual behavior:** 
after the first line the E-Motor stops and thats it....

#### Additional Information

[image]

thisiskeithb, 2020-06-03:
This is a known issue with TMC2208s, especially on Creality boards. See https://github.com/MarlinFirmware/Marlin/issues/16076 and https://github.com/MarlinFirmware/Marlin/issues/17306 (probably others, but I didn't search too hard).

You can try disabling `STEALTHCHOP_E`, but if that doesn't work, you'll need to disable `LIN_ADVANCE`.

TailBoom, 2020-06-04:
Thanks for the quick answer. 
Its a problem with the drivers.... Damit 
I will disable linear advance until a solution is available.
(btw. I tried to disable STEALTHCHOP_E and the result was the same.)

closed 2020-06-04 by TailBoom

thisiskeithb, 2020-06-04:
You could always upgrade to a 32bit drop-in replacement board with TMC2209s 🙂

thisiskeithb, 2020-06-04:
Just as silent and linear advance works with no fuss.

ipartola, 2020-06-06:
@thisiskeithb can you recommend a specific board? Seems like the SKR 1.3 is hilariously badly constructed.