MarlinFirmware/Marlin #18196: [FR] option to disable stepper Auto-Assignment

dshield55, 2020-06-04:
### Description

I think there should be a #define argument to active the Auto-Assignment for Dual X, Dual Y, Multi-Z Steppers section in pins.h. It should continue to be defined by default of course, but I ran into a project where auto-assignments were very troublesome and my options were to either comment out 300 lines (not easy to change or unchange) or delete the whole 300 line section (but then it's gone and troublesome to bring back.) Would have been super helpful if I could have just commented out a define argument and it disables the whole section.

### Feature Workflow

#### Additional Information

sjasonsmith, 2020-06-20:
I tend to agree with this. I personally prefer to know exactly how I'm configuring things as opposed to trying to keep track of some internal rules for how things get reassigned.

@dshield55 could you provide some more detail about your exact scenario?
Posting configs will likely be useful, along with your pins file since I'm assuming that will be needed.

I'd like a clear understanding of what you were trying to do that it caused problems.

InsanityAutomation, 2020-07-14:
It wont auto assign anything that already has pins defined. Auto assignment only kicks in when you tell it to use more than what it has. Explicitly define Z2 and it wont re-assign an E stepper to it. This can be done in the pins file for the board or configuration.h

closed 2020-07-15 by thinkyhead