MarlinFirmware/Marlin #18197: [BUG] ubl not work, deep check result

setar, 2020-06-04:
The UBL function works very strange for my configuration. 
The height map looks correct, but the virtual space is rotated counterclockwise relative to the Y axis ( 0<x<1/2 X_BED_SIZE : is airprinting  ; 1/2 X_BED_SIZE < x < X_BED_SIZE : nozzle hit on the bed surface).

The first time I discovered this bug using the bugfix-2.0.x branch.
Next i try check release branch and test 2.0.4.3 : similar behavior.

I made a test code that reproduces 100% error for my configuration. This is the repeated activation and deactivation autolevel at the edge points of the bed.

The test for 2.0.4.3
test in 0,0 - all OK
https://youtu.be/d3VOv2bHC44
test in  400,0 - BUG (shift Z level)
https://youtu.be/AxNSX5V2sM4

---- detail ----
bed  0,0:400,355

Cold system: Bed, head, ambient ~25C

G28
M851 Z0
G29 D
G29 P0
G29 T1
Bed Topography Report for CSV:
0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000
0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000
0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000
0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000
0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000
0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000
0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000
0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000
0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000
0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000  0.000

G29 P1
...
SENDING:G29 T1
Bed Topography Report for CSV:
1.606  1.618  1.568  1.509  1.474  1.423  1.374  1.305  1.172  0.970
1.797  1.835  1.830  1.768  1.719  1.674  1.609  1.537  1.375  1.165
1.941  1.985  1.968  1.932  1.887  1.839  1.761  1.661  1.503  1.295
1.984  1.991  1.988  1.920  1.874  1.832  1.753  1.668  1.539  1.372
1.965  1.982  1.957  1.915  1.860  1.800  1.738  1.639  1.534  1.362
1.953  1.971  1.941  1.905  1.865  1.799  1.733  1.659  1.541  1.395
1.876  1.882  1.845  1.818  1.787  1.740  1.697  1.633  1.545  1.398
1.697  1.743  1.739  1.708  1.694  1.665  1.636  1.583  1.495  1.374
1.565  1.601  1.590  1.594  1.584  1.576  1.553  1.508  1.446  1.345
1.498  1.497  1.488  1.464  1.471  1.465  1.464  1.457  1.411  1.338

G29 S2

SENDING:G29 P5
of samples: 100
Mean Mesh Height: 1.650071
Standard Deviation: 0.215662

G29 P6 C-1.65

SENDING:G29 T1
Bed Topography Report for CSV:
-0.044  -0.032  -0.082  -0.141  -0.176  -0.227  -0.276  -0.345  -0.478  -0.680
0.147  0.185  0.180  0.118  0.069  0.024  -0.041  -0.113  -0.275  -0.485
0.291  0.335  0.318  0.282  0.237  0.189  0.111  0.011  -0.147  -0.355
0.334  0.341  0.338  0.270  0.224  0.182  0.103  0.018  -0.111  -0.278
0.315  0.332  0.307  0.265  0.210  0.150  0.088  -0.011  -0.116  -0.288
0.303  0.321  0.291  0.255  0.215  0.149  0.083  0.009  -0.109  -0.255
0.226  0.232  0.195  0.168  0.137  0.090  0.047  -0.017  -0.105  -0.252
0.047  0.093  0.089  0.058  0.044  0.015  -0.014  -0.067  -0.155  -0.276
-0.085  -0.049  -0.060  -0.056  -0.066  -0.074  -0.097  -0.142  -0.204  -0.305
-0.152  -0.153  -0.162  -0.186  -0.179  -0.185  -0.186  -0.193  -0.239  -0.312

G29 S1

SENDING:M111 S246
echo:DEBUG:INFO,ERRORS,COMMUNICATION,LEVELING

SENDING:G0 X0 Y0 Z0
SENDING:M114 D
X:0.00 Y:0.00 Z:0.15 E:0.00 Count A:0 B:0 Z:0
SENDING:G0 X0 Y0 Z10
SENDING:M114 D
X:0.00 Y:0.00 Z:10.00 E:0.00 Count A:0 B:0 Z:38460

SENDING:G29 A
  current_position= X0.00 Y0.00 Z0.00 : Leveling OFF
 raw get_z_correction(0.00,0.00) = -0.152496 >>>---> -0.152496
  current_position= X0.00 Y0.00 Z0.15 : ...Now ON
  current_position= X0.00 Y0.00 Z0.15 : sync_plan_position
 raw get_z_correction(0.00,0.00) = -0.152496 >>>---> -0.152496
Unified Bed Leveling System v1.01 active

SENDING:M114 D
X:0.00 Y:0.00 Z:0.15 E:0.00 Count A:0 B:0 Z:0

SENDING:G0 X0 Y0 Z10
SENDING:M114 D
X:0.00 Y:0.00 Z:10.00 E:0.00 Count A:0 B:0 Z:37874

(37874-38460)/3846 = -0.152366 // OK

SENDING:G0 X400 Y0 Z10 F6000 // head do jump in last 1mm of path
SENDING:M114 D
X:400.00 Y:0.00 Z:10.00 E:0.00 Count A:85333 B:85333 Z:38460  // why 38460 ??? Unified Bed Leveling System v1.01 active , must be 37261
SENDING:G29 D
  current_position= X400.00 Y0.00 Z10.00 : Leveling ON
 raw get_z_correction(400.00,0.00) = -0.311778 >>>---> -0.311778
  current_position= X400.00 Y0.00 Z9.69 : ...Now OFF
  current_position= X400.00 Y0.00 Z9.69 : sync_plan_position
Unified Bed Leveling System v1.01 inactive
SENDING:M114 D
X:400.00 Y:0.00 Z:9.69 E:0.00 Count A:85333 B:85333 Z:37261

P.S. tomorrow I will repeat the test for current branch bugfix-2.0.x

[test_gcode.zip](https://github.com/MarlinFirmware/Marlin/files/4728522/test_gcode.zip)
[config_2.0.4.3.zip](https://github.com/MarlinFirmware/Marlin/files/4728543/config_2.0.4.3.zip)

setar, 2020-06-04:
checked on bugfix-2.0.x firmware, the problem is absolutely identical

[setar_conf_bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4732625/setar_conf_bugfix-2.0.x.zip)

setar, 2020-06-05:
point X_BED_SIZE -1 (399,1) is not affected by this effect.
Try #define MESH_INSET 1
[x399_y1_ubl_test.zip](https://github.com/MarlinFirmware/Marlin/files/4734060/x399_y1_ubl_test.zip)

setar, 2020-06-05:
with MESH_INSET 1 the bug shift inside, and point 399,1 now affected.

boelle, 2020-06-21:
@setar still an issue?

setar, 2020-06-22:
I have not done any tests since the last post. Were there any changes in the auto level code?

boelle, 2020-06-23:
you have to be active and check commits and changes yourself

and then retest with [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) as its updated daily

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle