MarlinFirmware/Marlin #18203: [FR] Get SPEAKER working on SKR Mini E3 Configs

thisiskeithb, 2020-06-05:
### Description

Due to timer conflicts, `SPEAKER` can’t be enabled on SKR Mini E3 boards. This Feature Request is a continuation of comments from https://github.com/MarlinFirmware/Marlin/pull/18011.

sjasonsmith, 2020-06-05:
The limited timers and inefficiency with which they are used by Arduino-based firmware causes a shortage of timers when using all the possible features on these boards.

For example, on the SKR Mini E3 boards they are allocated as below:

1 - Used for Fan PWM
2 - Temperature ISR
3 - Disabled because its pins are used by the SD slot. This might not truly be necessary, but it is what happens today.
4 - Software Serial (Needed for Mini E3 1.2 and E3 DIP boards)
5 - Step ISR
6 - Basic Timer (Currently unusable)
7 - Basic Timer (Unusable by most libraries, but might be used by _something_
8 - Servo / BLTouch

This accounts for every timer on a STM32F103 MPU. As can be seen, there is nothing left for the TONE_TIMER needed by the SPEAKER feature, unless we can use the Basic Timers.

Tone, Servo, and the Marlin ISRs are not written to work with a basic timer. SoftwareSerialM _might_ work on timer 6, but I have not yet tried it.

I have changes in progress which allow moving the temperature ISR to timer 6, freeing up timer 2. I have been unable to use Timer 7. Attempting to use it fails. I suspect it is being used by something already, but I do not know what.

xC0000005, 2020-06-07:
Something similar to STM32GENERIC's PWM would work - a single timer provides PWM -

https://github.com/huaweiwx/STM32GENERIC/blob/master/STM32/cores/arduino/stm32/stm32_PWM.c

sjasonsmith, 2020-06-08:
I had all this working and figured out some timer conflict detection mechanisms and everything, but it kind of fell apart once I started trying to make it general enough to apply to all STM32F1 boards.

I'm having to remove that conflict detection logic (for now), due to the complications involved in knowing which timers will be used for TONE and FANs. Even if you specify TONE_TIMER in the platformio.ini file, the Maple tone.cpp library will auto-select a different timer if the pin aligns with a timer channel.

In light of the information thinkyhead added to this PR regarding the removal of Maple support in PlatformIO, it may be more valuable to focus on moving these boards to `HAL/STM32`, rather than investing in any improvements in `HAL/STM32F1`.
https://github.com/MarlinFirmware/Marlin/pull/17970#issuecomment-640299776

sjasonsmith, 2020-06-08:
@xC0000005 how is your timer auto-select code developing for `HAL/STM32`? Is that something you anticipate contributing any time soon/

xC0000005, 2020-06-08:
Yes, timer select is working. I have a few bugs to shake out and time’s hard to come by during the week, but it’s reasonable. I really also think I’m going to try and retrofit STM32GENERIC’s PWM method into Marlin, going through a PWM class. One timer for <n> PWM IOs seems like a decent trade.

thinkyhead, 2020-06-20:
I seriously need a cheat-sheet for all these timers and things, and which platforms and libraries are stealing them….

thinkyhead, 2021-02-15:
Please give #21079 a test to see whether it fixes the issue, which is most likely a timer conflict.

qwewer0, 2021-02-15:
Latest bugfix-2.0.x, SKR Mini E3 v1.2, (Ender 3 LCD)

**Without** https://github.com/MarlinFirmware/Marlin/pull/21079:

- `SPEAKER` defined
   - `M300` frequency and duration is kind of working, but definitely not as we would expect:
      - It seems that the speaker on/off tone switching time isn't as good as with `BUZZER`. The tone switching time seems to be longer than expected (?), which is really noticeable at low frequencies, and on high frequencies the tone switching seems to interfere with the M300 frequency witch produces lower or higher pithed sound depending on the extent of interference. (But this is just a guess from the quick test)

- `BUZZER` defined
   - `M300` sounds are merges together (no pause in between `M300` sounds) (bug?)
   - `M300` frequency control is still not working **as it is to be expected**
   - `M300` duration works

**With** https://github.com/MarlinFirmware/Marlin/pull/21079:

- `SPEAKER` defined
   - The "speaker" is now really quiet and the sounds are crunchy (maybe it is crunchy because of the low sound level?)
   - `M300` sound duration is now "frequency" dependent, which means, both `P` and `S` parameter somewhat changes the sound duration
   - Frequency control isn't working

- `BUZZER` defined
   - `M300` sounds are merges together (no pause in between `M300` sounds) (bug?)
   - `M300` frequency control is still not working **as it is to be expected**
   - `M300` duration works

So on the latest bugfix-2.0.x, for some reason M300 now does more than is did before, but it is still not useable.

I really hope that others too can test the speaker functionality, as it is really hard to describe how it is working right now.

qwewer0, 2021-02-15:
When BUZZER enabled the sound is the same on https://github.com/MarlinFirmware/Marlin/pull/21079 as on bugfix, but with SPEAKER the sound is so quiet that I wasn't able to record it.
With https://github.com/MarlinFirmware/Marlin/pull/21079 and SPEAKER enable, the **board restarts** when selecting Restore Defaults.

STM32F103RC_btt, SKR Mini E3 v1.2 configs:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5983435/Configuration.zip)

Here are 2-2 audio examples to the difference of SPEAKER and BUZZER on the latest bugfix-2.0.x.

[Audio_semples_SPEAKER_bugfix2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/5983400/Audio_semples_SPEAKER_bugfix2.0.x.zip)
[Audio_semples_BUZZER_bugfix2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/5983401/Audio_semples_BUZZER_bugfix2.0.x.zip)
(selecting restore defaults and store settings)

Misterke, 2021-02-22:
At some point I had the SPEAKER working with the change in tone.cpp (TONE_CHANNEL from 8 to 4), but somewhere down the line it seems to have broken again.  I'm unsure whether I had it working before I added a 3DTOUCH (and now my own opto-servo probe) and that this affected it or that just more recent builds (with newer libraries / framework) broke it, but when I now try to play a tone, I do notice some sort of dependency on the M300 S parameter, but it is very weird - always pretty low in frequency and slightly going up in frequency as I have the S parameter go down.

So, is there any progress here?  Especially being able to use a single timer for all PWM pins mentioned in one of the above comments sounds like the ideal solution, right?

Misterke, 2021-02-22:
I've also tried out the -DTONE_CHANNEL=4 -DTONE_TIMER=4 build_flags change of  #21079 .  Unfortunately it didn't fix my problems with SPEAKER.  Using M300 S1000 gives a tone far below 1000Hz, going to S2000 and higher does now seem to play a higher tone, but weirdly enough, going down to S900 or lower also gives a higher tone (with quite some distortion).

Note that I'm on an SKR Mini E3 v2.0 and if I understand the definitions in $HOME/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103r/board.cpp, some of the pins using this timer are different between v1.2 and v2.0.  But I'm confused as the values in that board.cpp do not seem to match the stuff mentioned by @sjasonsmith :

> 1 - Used for Fan PWM
> 2 - Temperature ISR
> 3 - Disabled because its pins are used by the SD slot. This might not truly be necessary, but it is what happens today.
> 4 - Software Serial (Needed for Mini [...]

So, @sjasonsmith , can you educate me on how your table can be determined from the code and how it matches the pin-to-timer mapping in Maple's board.cpp?  I'd love to understand this better - even if just for education ...  Thanks in advance!

thisiskeithb, 2021-06-05:
With all of the affected BTT SKR Mini E3 boards now running in the STM32 HAL by default & STM32F1/maple going away soon, I'm going to consider this resolved.

closed 2021-06-05 by thisiskeithb