MarlinFirmware/Marlin #18205: [BUG] (short description)

radek8, 2020-06-05:
By disabling the option
#define LCD_TIMEOUT_TO_STATUS
it does not turn off the return from the menu to the status screen, but the default value of 15s is used.
I would expect that disabling this option will turn off automatic return from the menu to the status screen.

When using the TFT display, there is a problem when using the encoder in touch mode. Turning off automatic return from the menu to the status screen would partially solve this.
see:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/757

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/680

Lord-Quake, 2020-06-05:
The default when disabled it 15 seconds.
Activate the option and set to x seconds for a different delay to menu.

radek8, 2020-06-05:
I don't want another delay, I want to turn off this feature.
I expected disabling this option to turn off the status screen return feature, but it's not.
When I select 0, the function is deactivated?

Lord-Quake, 2020-06-05:
> When I select 0 or-1, the function is deactivated?

Well, have you tried it?

closed 2020-06-05 by radek8