MarlinFirmware/Marlin #18208: [BUG] 3D-Touch probe does not deploy only on G29

huso4life, 2020-06-05:
### Bug Description
When trying to build a cold mesh the probe does not deploy. The z-axis does continues to move which results in a crash.
On static testing and homing the probe works just fine. Electrical connection is fine - probe deploys and stows fine in test mode while moving the xy-axes by hand.

### My Configurations
Marlin 2.0.5.3 config files attached. I am trying for days to get the probe to work with different settings enabled/disabled. At this point im not sure if the attached cfg works correctly. This was my latest try bevore giving up and posting this report.
[bltouch.zip](https://github.com/MarlinFirmware/Marlin/files/4737695/bltouch.zip)
Probe is connected to the z_min endstop pin. The SKR PROs bltouch probe pins were also tried but did not work either.

### Steps to Reproduce
1. select "build cold mesh" on lcd
2. x and y axes are homing
3. probe deploys and z-axis homes
4. actual bed leveling starts - probe wont deploy für 1st grid point resulting in a crash

**Expected behavior:** [What you expect to happen]
either
a) probe does not deploy - z-axis movement is blocked in z- direction and/or printer shuts down because probe is already triggered, 
or
b) probe deploys prior to z-axis movement allowing for bed leveling

**Actual behavior:** [What actually happens]
Probe does not deploy in any situation other than static tests via lcd menu, terminal or on G28 homing moves

#### Additional Information
I've reset the controller prior to actually crashing my printer. When resetting I've tried testing the deployment of the probe at the exact xy-coordinates it faild prior - the probe works perfectly fine. I can only reproduce the error on G29 directly after the initial homing

Tynik, 2020-06-16:
The same situation!

boelle, 2020-06-21:
@huso4life still an issue?

huso4life, 2020-06-21:
Its still an issue for me. I was not able to get the probe to work correctly and switched to manual bed leveling for the time beeing.

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle