MarlinFirmware/Marlin #18213: [BUG] (short description)

Coffee0297, 2020-06-06:
### Bug Description
My printer allways tagets 10c lower than setpoint. I have done an autotune as sugested in 17223 and 17547.

and im writing here bd your discord have not answered in a day.. kinda shitty imo

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4740067/config.zip)

[image]

thisiskeithb, 2020-06-06:
> And im writing here bc your discord have not answered in a day.. kinda shitty imo.

Marlin is _free_ software created and supported by volunteers. Adjust your expectations.

Coffee0297, 2020-06-06:
And so is klipper and we are less ppl but we answer 99% of the time within 10 min. So Idk.

I ogt the issue fixd by going to 2.0.3. Closing

closed 2020-06-06 by Coffee0297