MarlinFirmware/Marlin #18214: SD card issues with ReARM/RAMPS 1.4 with RepRap Full Graphics display

scott-pett, 2020-06-06:
Trying to use the LCD SD card on the RR Full Graphics display with ReARM/RAMPS 1.4
with suitably modified adaptor cable to obtain 5v from the ReArm board.

SD card appears to be detected, but no SD content appears - often causing long delays and occaisional grahics corruption.

All other functions work OK on the Graphics controller.

Expected functionality is to see a list of files on the LCD SD card after detection.

Switching to the Onboard SD card works as expected apart from a persistent "Media Inserted" status message.

Marlin Version is Bugfix 2.0.5.3 downloaded 6/6/2020 although similar problem was seen on 2.0.5.2

Similar situation also seen with a 2004 LCD with the necessary adaptor cable mods.

Both 2004 LCD and Full Graphics displays work OK on Marlin 1.1.9 with Arduino Mega and RAMPS 1.4 configuration. (with standard adaptor cable).

[Configuration - Copy.h.txt](https://github.com/MarlinFirmware/Marlin/files/4740064/Configuration.-.Copy.h.txt)

[Configuration_adv - Copy.h.txt](https://github.com/MarlinFirmware/Marlin/files/4740065/Configuration_adv.-.Copy.h.txt)

boelle, 2020-06-21:
@scott-pett still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle