MarlinFirmware/Marlin #18216: [BUG] Babystepping menu is lost if G29 in start gcode

wikolii, 2020-06-06:
### Bug Description

The babystepping menu (in my case with BABYSTEP_ZPROBE_OFFSET activated) is lost if i use G29 in the start gcode. Marlin version"2.0.5.3"

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[FW-DMS-DP5.zip](https://github.com/MarlinFirmware/Marlin/files/4741008/FW-DMS-DP5.zip)

### Steps to Reproduce

1. Generate Gcode with G28 and G29 commands
2. Save to SD card
3. Doesn't matter if the printer is just reset or has been powered for a while.
3. Start the print
4. Babystepping is accessible by doubleclick up until the G29 routine finishes.
5. Start gcode finishes and the actual print starts.

**Expected behavior:** [What you expect to happen]
Doubleclicking should bring up the babystepping (Z-offset) menu.
This behavior works when executing G29 from the motion menu, before starting the print and without G29 in the start gcode.
**Actual behavior:** [What actually happens]
Doubleclicking brings up the "Move Z" menu
Changing Z-offset from the Tune menu doesn't seem to have an effect during the print.
#### Additional Information
Configuration file is set to spanish language, please consider this if uploading the sketch for testing.
Arduino mega+ RAMPS + Inductive sensor + 2004 smart controller.
* Provide pictures or links to videos that clearly demonstrate the issue.

[WhatsApp Video 2020-06-06 at 18.32.07.zip](https://github.com/MarlinFirmware/Marlin/files/4741014/WhatsApp.Video.2020-06-06.at.18.32.07.zip)

* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Crunch69, 2020-06-07:
confirmed on latest bugfix

ellensp, 2020-06-07:
is related to having `if (should_babystep && can_babystep)` with `#define MOVE_Z_WHEN_IDLE                // Jump to the move Z menu on doubleclick when printer is idle.`
If the test fails it shows z menu...   
So either` should_babystep = printer_busy()` or `can_babystep = all_axes_known()` is failing...
Since it happens after a G29, I would guess the latter....    
But its 4am.. so have to stop looking for now...

ellensp, 2020-06-07:
if you define BABYSTEP_WITHOUT_HOMING does it still happen? as that sets `constexpr bool can_babystep = true;`

wikolii, 2020-06-08:
> if you define BABYSTEP_WITHOUT_HOMING does it still happen? as that sets `constexpr bool can_babystep = true;` so it ignores all_axes_known() function.

I don't know why I checked this at 2am.
I'll check in the morning and get back to you
Thanks

wikolii, 2020-06-08:
> if you define BABYSTEP_WITHOUT_HOMING does it still happen? as that sets `constexpr bool can_babystep = true;` so it ignores all_axes_known() function.

Ok, I couldn't sleep so I tried this and seems to be working. NoI, what does BABYSTEP_WITHOUT_HOMING mean? There is no explanation for it.

Thanks again!

ellensp, 2020-06-08:
it does a it says, it enables babysteps whithout having to home first...
more technically it disables  the all_axes_known() check.  

It not really a fix, its a work around of your issue.

boelle, 2020-06-21:
@wikolii still an issue?

wikolii, 2020-06-21:
I was able to enable the menu by defining  BABYSTEP_WITHOUT_HOMING as the suggested workaround. 

I haven't updated marlin since, if that's your question. I don't know if a fix was implemented other that the workaround. Plus i no longer have the printer with me. It was a client's.

Thanks everyone for the assistance.

closed 2020-06-23 by boelle