MarlinFirmware/Marlin #18217: [BUG] MKS Robin Nano Pins, ONBOARD_SD_CS_PIN

zillarob, 2020-06-07:
Been a while since I played with this board, but ran into the not in scope error with the current pins file.
Adding the pin to the pins file fixed it:
`//
// SD Card
//
#ifndef SDCARD_CONNECTION
  #define SDCARD_CONNECTION              ONBOARD
#endif

#define SDIO_SUPPORT
#define SD_DETECT_PIN                       PD12
#define ONBOARD_SD_CS_PIN                   PC11`

And that SDIO thing reminds me, had a ton of ghost moves with the SDIO clock at the default value. Changing it as shown here immediately cleared it up. Not sure how to go about tuning this one up without editing manually.
[https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/5](url)

zillarob, 2020-06-08:
That was quick, looks like it should be all tuned up now =D

https://github.com/MarlinFirmware/Marlin/pull/18223

thisiskeithb, 2020-06-18:
We had some chats about this outside this issue, but see PR https://github.com/MarlinFirmware/Marlin/pull/18347.

closed 2020-06-18 by zillarob