MarlinFirmware/Marlin #18220: [QUESTION] CONTROLLER_FAN Is not defined anywhere???

Amachete, 2020-06-07:
Hello

I am triying to learn how marlin works....

When I search for "CONTROLLER_FAN"  with MATCH WHOLE WORD on the whole marlin folder, Visual  Studio Code just finds 4 matches, any of them defines the CONTROLLER_FAN "word"

Is this right? 

Sorry If I am making you losing your time....

ellensp, 2020-06-07:
your search is to limiting.
You have things like CONTROLLER_FAN_PIN and #if PIN_EXISTS(CONTROLLER_FAN) the latter appends _PIN and checks it was defined....
there is also #define HAS_CONTROLLER_FAN 1

ellensp, 2020-06-07:
take a look in 
macros.h:#define PIN_EXISTS(PN)      (defined(PN##_PIN) && PN##_PIN >= 0)
this is a macro.   PN is the passed string, replaces iPN## in the second half
so you get (defined(CONTROLLER_FAN_PIN) && CONTROLLER_FAN_PIN >= 0)

Amachete, 2020-06-07:
Ok, i understand....there are so many macros! Thank you.

closed 2020-06-07 by Amachete