MarlinFirmware/Marlin #18225: HELP! Need help with marlin code to hook up REPRAP discount full graphic smart controller to stock Ender 3 motherboard

Wiketti, 2020-06-08:
Cant seem to get any graphics out of it.

I have added u8glib library to firmware and uncommented reprap discount full graphic smart controller from the code. Also commented ( // ) the CR-10 stock display from firmware.

It just shines blue and "whines".

ellensp, 2020-06-08:
This is mostly a hardware issue
2 * 10 pin plugs into 1 * 10 socket..   you can't have everything. 

the reprap discount full graphics display has 
encoder (3 pins) 
buzzer (1 pin)
12864 lcd (3 or 4 pins)   
sd card  (3 additional pins, shares some with lcd) 
stop button (1 pin)
power  (2 pins)

So 12 pins pins going to a 10 pin plug?
This is why the CR-10 display doesn't have sdcard.

Wiketti, 2020-06-08:
Can i modify one connector so that it matches ender's own pins without SD reader?

ellensp, 2020-06-08:
you have   (pins numbers are reversed on some boards) 
```
 *               _____                                              _____
 *           NC | 1 2 | GND                                     5V | 1 2 | GND
 *        RESET | 3 4 | 1.31 (SD_DETECT)             (LCD_D7) 1.23 | 3 4 | 1.22 (LCD_D6)
 *  (MOSI) 0.18 | 5 6   3.25 (BTN_EN2)               (LCD_D5) 1.21 | 5 6   1.20 (LCD_D4)
 * (SD_SS) 0.16 | 7 8 | 3.26 (BTN_EN1)               (LCD_RS) 1.19 | 7 8 | 1.18 (LCD_EN)
 *   (SCK) 0.15 | 9 10| 0.17 (MISO)                 (BTN_ENC) 0.28 | 9 10| 1.30 (BEEPER)
 *               -----                                              -----
 *               EXP2                                               EXP1
```

you need to use these pins  (marked with a X) 

```
 *               _____                                              _____
 *           NC | 1 2 | GND                                     5V | X X | GND
 *        RESET | 3 4 | 1.31 (SD_DETECT)             (LCD_D7) 1.23 | 3 4 | 1.22 (LCD_D6)
 *  (MOSI) 0.18 | X X   3.25 (BTN_EN2)               (LCD_D5) 1.21 | 5 6   1.20 (LCD_D4)
 * (SD_SS) 0.16 | 7 X | 3.26 (BTN_EN1)               (LCD_RS) 1.19 | 7 X | 1.18 (LCD_EN)
 *   (SCK) 0.15 | X 10| 0.17 (MISO)                 (BTN_ENC) 0.28 | X 10| 1.30 (BEEPER)
 *               -----                                              -----
 *               EXP2                                               EXP1
```
Everything with a X need to go to your 10 pin connector.
I have no idea on the wiring of that connector. but  MOSI, SCK and power pins cannot be moved to other pins.

Wiketti, 2020-06-08:
I'll give it a go soon as i get home. Thank you for the replys.

I'll get back to you as soon as i get some results.

ellensp, 2020-06-08:
From circuit diagrams I found https://github.com/Creality3DPrinting/CR10-Melzi-1.1.2/blob/master/Circuit%20diagram/Motherboard/CR-10%20Schematic.pdf

It looks like the EXP1 connector is as follows

```
/*  EXP1 Connector
 *                   _____
 *               5V | 1 2 | PC1 EXT-SDA
 *       EXT-A1 PA1 | 3 4 | RESET
 *      EXT-SCL PC0 | 5 6   PA4 EXT-A4
 *      EXT-TX1 PD3 | 7 8 | PD2 EXT-RX1
 *       EXT-A3 PA3 | 9 10| GND
 *                   -----
 */

/*  EXP1 Connector when used as CR-10 LCD
 *                   _____
 *               5V | 1 2 | DATA
 *              CLK | 3 4 | RESET
 *          BTN_ENC | 5 6   BEEPER_PIN
 *          BTN_EN1 | 7 8 | BTN_EN2
 *               CS | 9 10| GND
 *                   -----
 */
```

ellensp, 2020-06-08:
This seems to be the connector on your controller.  (please verify) 

```
/*   EXP1 Connector       As CR10 LCD Connector
 *        _____                                   _____
 *   PA4 |10 9 | PC0                  BEEPER_PIN |10 9 | BTN_ENC
 *   PD3 | 8 7 | RESET                   BTN_EN1 | 8 7 | RESET
 *   PD2   6 5 | PA1                     BTN_EN2   6 5 | LCD_PINS_D4 (ST9720 CLK)
 *   PA3 | 4 3 | PC1     (ST9720 CS) LCD_PINS_RS | 4 3 | LCD_PINS_ENABLE (ST9720 DAT)
 *   GND | 2 1 | 5V                          GND | 2 1 | 5V
 *        -----                                   -----
 */

```

ellensp, 2020-06-08:
Re check previous posts, I have updated the pins a bunch

thisiskeithb, 2020-06-09:
This isn't a Marlin bug, but someone has mapped out the pins & created a custom cable for a full graphic LCD on an SKR Mini E3 which uses the same EXP3/connector for a `CR10_STOCKDISPLAY`: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/159

[image]

Wiketti, 2020-06-09:
Took me ages yesterday to make that 10 pin cable 😂  as i dont have the right crimping tool. So gonna give it a go today. 

At first i just assumed it was marlin prob... 🙄 So sry to post this on wrong channel. Altho, i am super gratefull for all the help! You guys are the best!
I think i will be asking more help in near future as i'm fairly new to all this.

I'll let you know how this turns out soon 😊👌

ellensp, 2020-06-09:
I would have used a bunch of singe wire DuPont cables. same as you get with arduino boards and modules, then you can easily change things. Only make up a cable when you know its going to work.

Wiketti, 2020-06-09:
They did'nt have those single duponts (or the single connectors) in the local shop 🙄 so i had to make the cable on my own. Really need to order some for future. Those arduino boards keep piling up and i dont have decent tools to go with them 😑

Wiketti, 2020-06-10:
Okay, got those single duponts after some digging and a lot of driving 😅
Hooked the screen up like shown above (also tried reversed).

It just flickers and lights up but nothing else.

Wiketti, 2020-06-10:
Can i link a video?

ellensp, 2020-06-10:
yes and images... on how you wired it, and confirm you set CR10_STOCKDISPLAY and disabled reprap discount full graphic smart controller

ellensp, 2020-06-10:
also do you have a multimeter...  so you can confirm which pins are gnd and 5v on both ends

Wiketti, 2020-06-10:
[image]
[image]
[image]

Wiketti, 2020-06-10:
[image]

Wiketti, 2020-06-10:
do i need  to install some firmware into the screen?

ellensp, 2020-06-10:
This https://github.com/Creality3DPrinting/Ender-3/blob/master/Ender-3%20PCB/Ender3_pcb_parts.PDF tells me that the its not wired correct, gnd and 5v are at other end. gnd on the inside 5v on the outside 

try my pins
/*   EXP1 Connector       As CR10 LCD Connector
 *        _____                                   _____
 *   PA4 | 6 5 | PC0                  BEEPER_PIN | 6 5 | BTN_ENC
 *   PD3 | 7 4 | RESET                   BTN_EN1 | 7 4 | RESET
 *   PD2   8 3 | PA1                     BTN_EN2   8 3 | LCD_PINS_D4 (ST9720 CLK)
 *   PA3 | 9 2 | PC1     (ST9720 CS) LCD_PINS_RS | 9 2 | LCD_PINS_ENABLE (ST9720 DAT)
 *   GND |10 1 | 5V                          GND |10 1 | 5V
 *        -----                                   -----
 */

no there is no firmware on the display. its a dumb device.
firmware on the controller.

ellensp, 2020-06-10:
to simply things you don't need exp2 at all just yet. It is for encoder and reset pins. So you can remove those for now if you wish...