MarlinFirmware/Marlin #18227: [BUG] Can't move probe down during temperature probe / PINDA v2 calibration using G76

rudihorn, 2020-06-08:
### Bug Description

I'm trying to get the temperature probe calibration for a pinda v2 compatible probe to work on my printer. During calibration my probe is slightly too far from the bed, so it doesn't heat up that well. I'm trying to reduce that distance so it heats up better, and it does it for the first heat up phase, but as soon as the probe runs it refuses to go down to that distance again.

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/4744937/Config.zip)

Line 137 in `G76_m871.cpp` needs to be changed from:
```C++
  const xyz_pos_t parkpos = temp_comp.park_point,
            probe_pos_xyz = temp_comp.measure_point + xyz_pos_t({ 0.0f, 0.0f, 0.5f }),
```
to
```C++
  const xyz_pos_t parkpos = temp_comp.park_point,
            probe_pos_xyz = xyz_pos_t(temp_comp.park_point) + xyz_pos_t({ 0.0f, 0.0f, <some negative height> }),
```
(note the cast of `temp_comp.park_point` to `xyz_pos_t`, this is necessary otherwise the z axis value is ignored)

### Steps to Reproduce

Set some height as above in the calibration G-code file, and run `G76 P`.

**Expected behavior:** It should go down to `<some negative height> - probe.offset_xy[2]`, but doesn't do this after a probe

**Actual behavior:** The probe moves down until the probe triggers, and refuses to move any further down.

#### Additional Information

I've added some debugging output running:
```C++
      // before move
      SERIAL_ECHOLNPAIR("Move to X:", (noz_pos_xyz[0]), "Y:", (noz_pos_xyz[1]),"Z:", (noz_pos_xyz[2]));

      // after move
      SERIAL_ECHOLNPAIR("Now at   X:", (current_position[0]), " Y:", (current_position[1])," Z:", (current_position[2]), " ZP:", planner.position[2], " ZA:", (planner.get_axis_position_mm(Z_AXIS)));

      // after probe
      SERIAL_ECHOLNPAIR("Now at   X:", (current_position[0]), " Y:", (current_position[1])," Z:", (current_position[2]), " ZP:", planner.position[2], " ZA:", (planner.get_axis_position_mm(Z_AXIS)));
```

It results in a log such as:
```
Recv: Now at   X:55.00 Y:100.00 Z:0.10 ZP:40 ZA:0.10
Recv: Waiting for probe heating. Bed:110 Probe:30
...
Recv: Measured: 1.19
Recv: After at X:55.00 Y:100.00 Z:2.49 ZP:998 ZA:2.49
Recv: Move to X:55.00Y:100.00Z:0.10
Recv: Now at   X:55.00 Y:100.00 Z:0.10 ZP:40 ZA:2.49
Recv: Waiting for probe heating. Bed:110 Probe:35
```

So `planner.get_axids_position_mm(Z_AXIS)` does not update while the motor steps and current z axis values do. The only call happening inbetween is the following probe call, which should not stop the motor from moving below the probe point:

```C++
    do_blocking_move_to_z(5.0); // Raise nozzle before probing
    const float measured_z = probe.probe_at_point(nozpos, PROBE_PT_NONE, 0, false);  // verbose=0, probe_relative=false
```

rudihorn, 2020-06-08:
Looking at the debug logs I see 

```
Recv:   current_position= X55.00 Y100.00 Z5.00 : Probe::set_deployed
Recv: deploy: 1
```

Is called, but `set_deployed` is never disabled. I will investigate a little more and report back.

rudihorn, 2020-06-08:
Indeed adding a `probe.stow()` seems to solve this problem, but I'm not sure it is the right solution to this problem. To me it seems that the `probe_at_point` function should be restoring the state it was in, rather than leaving this up to the caller. If the `probe_at_point` needs to deploy to perform the point probe, then it should stow it again, and if the caller would like the probe to not be stowed then the caller should be responsible for deploying and stowing the probe.

In fact ideally one would try to define an object that manages this using a constructor / destructor which simultaniously acts as a proof obligation:
```
class ProbeDeployed {
  ProbeDeployed() { 
    if (probe.is_deployed) { fail }
    probe.deploy();
  }
  ~ProbeDeployed() {
    if (!probe.is_deployed) { fail }
    probe.stow();
  }
}
```

Then your code for a multi probe be something like:
```
void multi_probe() {
  ProbeDeployed deploy;
  probe.probe_at_point(xy, deploy);
  probe.probe_at_point(xy2, deploy);
}
```

While a single probe could implicitly create this object:
```
void single_probe() {
  probe.probe_at_point(xy, ProbeDeployed());
}
```

rudihorn, 2020-06-08:
I am unfortunately now receiving probe triggered early messages as the temperature rises though:
```
Recv: SLOW Probe fail! - Triggered early.
Recv:   current_position= X55.00 Y100.00 Z3.60 : <<< run_z_probe
```

Will need to debug this later.

rudihorn, 2020-06-08:
The slow probe seems to have been an issue with the probe firing slow after waiting for a long heat up time. I've also had to adjust the temperature steps for any useful values. My setup yields the following graph and shows that it is necessary to have far more granular controls over the temperature, as my setup cannot get to such high temperatures and there is no need to extrapolate higher, but instead more knowledge about the offsets at lower temperatures is crucial.

[image]

I'll be submitting a PR in the coming days for this.

boelle, 2020-06-21:
@rudihorn still an issue?

rudihorn, 2020-06-21:
This can be closed.

closed 2020-06-21 by rudihorn