MarlinFirmware/Marlin #18229: [BUG] EEPROM Index Error; Initialize (Reset) does not fix

warrenfalk, 2020-06-08:
### Bug Description

As of the latest commit on bugfix-2.0.x, bac760207c3e85b9ac97a36c37aeb4ad601102a3, on boot it prompts "EEPROM Index Error, Initialize EEPROM?" with options "Ignore" and "Reset".  Selecting "Reset" shows "Settings Stored" but on reboot the same error comes up.

This does not happen on the previous commit.

### My Configurations

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4747028/configuration.zip)

### Steps to Reproduce

Happens every time on boot as described.
1. Boot
2. See error
3. Select "Reset"
4. Boot
5. See error again

**Expected behavior:** [What you expect to happen]

Boot, no error
-or-
Boot, see error, reset, no error

**Actual behavior:** [What actually happens]

Reset does not work

nife87, 2020-06-08:
Also getting these errors with the latest commit (not only on boot). Really annoying because they revert the EEPROM settings to the defined limits, so for instance my Z offset and steps/mm are reverted. Tried reinitializing EEPROM and storing the new values again with the same result.

Slapsider, 2020-06-08:
Im also having the same issue on start up. with bug fix 2.0.x

allanchen2019, 2020-06-08:
Same issue with latest bugfix-2.0.x.
Anet A8

SirYasko, 2020-06-08:
Saving EEPROM from menu or M500 (from .gcode file) does not work.
Version: trunk (bugfix 2.0.x)
Printer: Ender 5

closed 2020-06-09 by thinkyhead