MarlinFirmware/Marlin #18231: [FR] Secondary Serial Output for LCD Display

bill-orange, 2020-06-08:
Description: 

The USB output on my SKR1.3 board provides a serial output viewable via Octopi or many other devices.  I would like to have a second serial output of the terminal information at a lower baud rate (9600) and 5v level to feed into an Arduino for display on my printers stock 20 X 4 display.

Feature Workflow:

1. select something like this in configuration_adv.h: 
```
#define Secondary_Serial
#if ENABLED (Secondary_Serial)
  #define Secondary_Serial_Baud 9600
  #define Secondary_Serial_Pin P2_04
#endif
```
2. program Arduino for serial in at 9600 baud and LCD out and wire (easy)
3. view serial data stream at the printer

Additional Information:  I have added an SKR1.3 board to my Anet A8.  This leaves the existing Anet 2004 LCD display unused.  I could hook an Arduino to it to measure voltage, ambient temperature, etc, but the serial data stream would be much more useful.

borland1, 2020-07-11:
There is a g-code M575 supported by Marlin 2.0.x that can set the baud rate for the secondary serial.  Custom g-code can be initiated with Startup commands found in `Configuration_adv.h` file. Look for....

```
 * Startup commands
 *
 * Execute certain G-code commands immediately after power-on.
 */
//#define STARTUP_COMMANDS "M17 Z"
```

Hope that helps.

bill-orange, 2020-07-11:
@borland1 That is very helpful.  Thank you.

closed 2020-07-11 by bill-orange