MarlinFirmware/Marlin #18232: [BUG] MAXTEMP: Bed error

Heimdoulila, 2020-06-08:
Hi,
I have an SKR 1.4 with TMC2208 and TFT24. My printer is CR10s and I use Marlin 2.0.5.3. Also I transform my "CR10s 12v" into 24v printer (bed/fan/hotend)

All the time I get this error. Without connecting the bed, I start the card via the computer, it beeps then the message "MAXTEMP : bed error" appears.

I try to delete my .PIO folder and updating again but nothing change.

Maybe my card is faulty?

Thanks you for your help, I've been looking for the solution all day on internet, I don't understand the problème.

[configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/4747608/configuration.h.txt)
[configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/4747609/configuration_adv.h.txt)
[platformio.txt](https://github.com/MarlinFirmware/Marlin/files/4747610/platformio.txt)

ellensp, 2020-06-09:
The firmware is looking at your bed thermistor connector.
If this is not plugged in you should not get a MAXTEMP error it should read a very low temp instead.
So yes, if bed thermistor is unplugged and your getting this error. you may have a board fault.
If the thermistor is plugged in, it may be shorted. this would give a MAXTEMP error

Heimdoulila, 2020-06-09:
That's what I thought. The same error is displayed with the thermistor pluggedc and unplugged.
Thanks you for your help

XDA-Bam, 2020-06-09:
Before you send it back, maybe take a look under the board and check if the solder pads on the thermistor connector are - by accident - shorted by a small string of solder. If so, that should be easy to fix.

Heimdoulila, 2020-06-09:
> Before you send it back, maybe take a look under the board and check if the solder pads on the thermistor connector are - by accident - shorted by a small string of solder. If so, that should be easy to fix.

aduction

Heimdoulila, 2020-06-09:
No there is nothing, the card is like new.
Is it impossible that it comes from Firmeware?

XDA-Bam, 2020-06-09:
It's not impossible. Maybe try the latest bugfix-2.0.x version and set up a fresh config for your printer / board based on the default one.

Heimdoulila, 2020-06-09:
That's exactly what I just tried, and it's the same problem. I took the preprogrammed version of BTT. I imagine now that there are no more hopes

boelle, 2020-06-21:
@Heimdoulila still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle