MarlinFirmware/Marlin #18233: Unable to pause or stop print from LCD when printing tethered (with octoprint)

Koyodee, 2020-06-08:
When printing with OctoPrint, I am unable to either use a M601 to pause my print or pause it from the LCD Screen. Both the Pause Print and Stop Print buttons don't do anything to the print in progress, yet on the LCD, it shows that the print is stopped, and shows the usual menu when the printer is idle. This means that if a print is failing, the only way I'm able to stop it is by power cycling, or by launching octoprint in order to stop the print from there.

thisiskeithb, 2020-06-08:
You didn’t include your configs like the issue template asked, but you should enable `HOST_ACTION_COMMANDS` and `HOST_PROMPT_SUPPORT`.

Koyodee, 2020-06-09:
Sorry about that. What you suggested worked great, thank you!!

closed 2020-06-09 by Koyodee