MarlinFirmware/Marlin #18235: [BUG] Y Axis not homing on core XY

SevCr, 2020-06-08:
Hi community. 
Please assist. Y axis not homing. Spent everal hours doing small changes to firmware and tinkering with it. I am not sure if I missed someting or if it is a bug, but help will be greatly appreciated. It seams this is the last obstacle, before I can do first calibration prints and tune it. 

### Bug Description

Y axis not homing.
Set all endstops ok. Checked operation of all endstop with M119, all good. 
G28, only homes X - Y doesnt move
G28 X Y only homes X - Y doesnt move
Comands sent both via screen and Pronterface, same result.

It is Core XY, stepping, directions, etc, all working ok. Axis are moving manualy ok in all directions.
Using Z_probe as Z_min (induction probe).
Endstops on X and Y are of the same type and wired the same. Simple mechanical switch, 2 wire connection GND-Input

### My Configurations
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4748268/Configuration.zip)

### Steps to Reproduce
Described above

**Expected behavior:** Home all axis

**Actual behavior:** only X is homing

G4Cab, 2020-06-09:
Do you home to ymin?
Try homing to ymax. Helped me.

Look at my issue https://github.com/MarlinFirmware/Marlin/issues/18212

SevCr, 2020-06-17:
Homing to Y+ seemed to help with this issue. Case closed

closed 2020-06-17 by SevCr