MarlinFirmware/Marlin #18236: [BUG] Anycubic Kossel Delta Plus Provided Example fails to compile

pablopeu, 2020-06-08:
Trying to fix some issues with my Delta printer (bed leveling calibration) downloaded June/8 2.0x bugfix release and the untouched provided example config files for the printer fail to compile with this error:

`
C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `Temperature::start_watching_bed()':

<artificial>:(.text+0x165a): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `Temperature::disable_all_heaters()':

<artificial>:(.text+0x16e8): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `MarlinUI::get_blink()':

<artificial>:(.text+0x1758): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `MarlinUI::set_custom_characters(HD44780CharSet)':

<artificial>:(.text+0x209c): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `MarlinUI::goto_screen(void (*)(), unsigned int, unsigned char, unsigned char)':

<artificial>:(.text+0x213c): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o:<artificial>:(.text+0x4828): more undefined references to `millis' follow

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `Buzzer::tick()':

<artificial>:(.text+0x4a2c): undefined reference to `tone(unsigned char, unsigned int, unsigned long)'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `MarlinUI::set_status(char const*, bool) [clone .part.6]':

<artificial>:(.text+0x4afc): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `MarlinUI::synchronize(char const*)::{lambda()#1}::_FUN()':

<artificial>:(.text+0x60e0): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `menu_advanced_filament()':

<artificial>:(.text+0x97bc): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `kill(char const*, char const*, bool)':

<artificial>:(.text+0xb044): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `menu_preheat_m2()':

<artificial>:(.text+0xc3ac): undefined reference to `delay'

<artificial>:(.text+0xc3ba): undefined reference to `delay'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `menu_preheat_m1()':

<artificial>:(.text+0xc52a): undefined reference to `millis'

<artificial>:(.text+0xc572): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `Temperature::thermal_runaway_protection(Temperature::tr_state_machine_t&, float const&, float const&, heater_ind_t, unsigned int, unsigned int)':

<artificial>:(.text+0xc73e): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `Temperature::manage_heater()':

<artificial>:(.text+0xcde2): undefined reference to `millis'

<artificial>:(.text+0xce30): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o:<artificial>:(.text+0xd900): more undefined references to `millis' follow

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans0.ltrans.o: In function `MarlinUI::completion_feedback(bool) [clone .part.4]':
[... 113 lines cut ...]
<artificial>:(.text+0x424c): undefined reference to `millis'

<artificial>:(.text+0x4276): undefined reference to `millis'

<artificial>:(.text+0x433e): undefined reference to `millis'

<artificial>:(.text+0x4366): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans1.ltrans.o:<artificial>:(.text+0x438c): more undefined references to `millis' follow

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans1.ltrans.o: In function `idle(bool) [clone .constprop.70]':

<artificial>:(.text+0x45b2): undefined reference to `delay'

<artificial>:(.text+0x4706): undefined reference to `millis'

C:\Users\popopopo\AppData\Local\Temp\ccupKD8z.ltrans1.ltrans.o:(.rodata+0x29): undefined reference to `Print::write(unsigned char const*, unsigned int)'

c:/users/popopopo/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/avr6/crtatmega2560.o:(.init9+0x0): undefined reference to `main'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Mega or Mega 2560.
`

thisiskeithb, 2020-06-09:
I can't reproduce the error with the latest `bugfix-2.0.x` and [Kossel Linear Plus configs](https://github.com/MarlinFirmware/Configurations/tree/19ccae84e4466886bb39bdb72ef557f3d45ff0aa/config/examples/delta/Anycubic/Kossel%20Linear%20Plus) with VSCode/PlatformIO:

```prolog
Linking .pio/build/mega2560/firmware.elf
Checking size .pio/build/mega2560/firmware.elf
Building .pio/build/mega2560/firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [======    ]  63.0% (used 5163 bytes from 8192 bytes)
Flash: [=======   ]  72.2% (used 183260 bytes from 253952 bytes)
===================================== [SUCCESS] Took 36.48 seconds =====================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  SUCCESS   00:00:36.481
mega1280                  IGNORED
MightyBoard1280           IGNORED
MightyBoard2560           IGNORED
```

pablopeu, 2020-06-09:
I'm using Arduino IDE 1.8.12

ellensp, 2020-06-09:
Arduino IDE 1.8.12 Configes from Kossel Linear Plus configs 

.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-size -A /tmp/arduino_build_206274/Marlin.ino.elf
Sketch uses 181536 bytes (71%) of program storage space. Maximum is 253952 bytes.
Global variables use 5154 bytes (62%) of dynamic memory, leaving 3038 bytes for local variables. Maximum is 8192 bytes.

No issues.

pablopeu, 2020-06-09:
D@mn well will reinstall ide and try again, will report later. Thanks!

ellensp, 2020-06-09:
The errors your getting are all standard Arduino functions. This show something is fundamentally wrong. My guess is you selected a board other than  "Arduino Mega or Mega 2560'

pablopeu, 2020-06-09:
I compile 2.053 without a hitch, will try again with today bugfix later. Thanks

pablopeu, 2020-06-09:
uninstalled arduino IDE 1.8.12, reinstalled, no dice same problem, uninstalled again, removed userdata/local/arduino15 folder, installed again. now it compiled fine. Downloading PlatformIO to test it too. 
Now testing the firmware in the machine, but in any case, I'm closing this one. Thanks Guys

closed 2020-06-09 by pablopeu