MarlinFirmware/Marlin #18238: [BUG] (short description)

nullassult, 2020-06-09:
### Bug Description
When enabling miniVIKI display with RUMBA board in Marlin 2.0 (including latest bugfix) I am unable to compile sketch. This board and display works fine in Marlin 1.1.x.

Error compiling for board Arduino Mega or Mega 2560 (RUMBA)
Arduino: 1.8.12 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src\lcd\dogm\ultralcd_DOGM.cpp:42:0:

sketch\src\lcd\dogm\ultralcd_DOGM.h:195:23: error: 'DOGLCD_CS' was not declared in this scope

     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI

                       ^

sketch\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'

 U8G_CLASS u8g(U8G_PARAM);

               ^~~~~~~~~

sketch\src\lcd\dogm\ultralcd_DOGM.h:195:23: note: suggested alternative: 'DOGLCD_SCK'

     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI

                       ^

sketch\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'

 U8G_CLASS u8g(U8G_PARAM);

               ^~~~~~~~~

sketch\src\lcd\dogm\ultralcd_DOGM.h:195:34: error: 'DOGLCD_A0' was not declared in this scope

     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI

                                  ^

sketch\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'

 U8G_CLASS u8g(U8G_PARAM);

               ^~~~~~~~~

sketch\src\lcd\dogm\ultralcd_DOGM.h:195:34: note: suggested alternative: 'DOGLCD_SCK'

     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI

                                  ^

sketch\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'

 U8G_CLASS u8g(U8G_PARAM);

               ^~~~~~~~~

exit status 1
Error compiling for board Arduino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4749198/Marlin.zip)

### Steps to Reproduce

1. Start with default Marlin - 2.0.x  or Marlin-bugfix - 2.0.x
2. Uncomment #define VIKI2 or #define miniVIKI (I have a miniVIKI, but either causes issue)
3. Click upload button in Arduino (1.8.12 with all U8glib libraries updated; U8glib 1.19.1, U8g2 2.27.6, LCDMenuLib2 2.2.4, LCDMenuLib 2.3.5)

**Expected behavior:** Complies, uploads, enables display, and works as it does in my Marlin 1.1.x config.

**Actual behavior:** Errors out while compiling sketch.

#### Additional Information

ellensp, 2020-06-09:
I just tried 1.1.9 and 1.1.6 and it also will not compile.

There is no standard place to plug a vikiMINI into a RUMBA board.   So there is no pins setup for it in pin_RUMBA.h  
So how do you have it wired to the RUMBA?
Can you share this older marlin that work so we can see what has been changed?

nullassult, 2020-06-09:
[Marlin-1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/4754403/Marlin-1.1.x.zip)
This complies fine for me.

I checked the pins_RUMBA.h and it contained this (I also included that file in the zip)

//
// Misc. Functions
//
#define SDSS               53
#define LED_PIN            13
#define PS_ON_PIN          45
#define KILL_PIN           -1 //46
#define CASE_LIGHT_PIN     45

//
// LCD / Controller
//
#define SD_DETECT_PIN      49
#define BEEPER_PIN         44
#define LCD_PINS_RS        19
#define LCD_PINS_ENABLE    42
#define LCD_PINS_D4        18
#define LCD_PINS_D5        38
#define LCD_PINS_D6        41
#define LCD_PINS_D7        40
#define BTN_EN1            11
#define BTN_EN2            12
#define BTN_ENC            43
#define DOGLCD_A0          46
#define DOGLCD_CS          44

ellensp, 2020-06-10:
try this replacement pins_RUMBA.h
[pins_RUMBA.zip](https://github.com/MarlinFirmware/Marlin/files/4756101/pins_RUMBA.zip)

ellensp, 2020-06-10:
It would seem you did something incorrectly when replacing pins_RUMBA.h with the contents of the zip file. I know this compiles

nullassult, 2020-06-12:
You are correct! I copied it to the wrong directory (Marlin-2.0x>Marlin) instead of the correct one (Marlin-2.0x>Marlin>src>pins>ramps) for those that come across this in the future.

Would it make sense to add in the comments a need for a custom pins_RUMBA.h file or for any display without a "standard place to plug in"?

Thank you for your help!

ellensp, 2020-06-13:
Please close this issue. If anyone else has should have this issue, we can direct them here.

ellensp, 2020-06-17:
Please close this issue.

closed 2020-06-17 by nullassult