MarlinFirmware/Marlin #18240: [BUG] EXTUI (Malyan LCD) not working on board with LPC1768/69

adams79, 2020-06-09:
### Bug Description

Enabling Maylan LCD into the configuration file, the serial1 (the one on the TFT pins) continue to output unencoded command. Tried on STM32F1 board and able to see the encoded Malyan command when enabled.

### Steps to Reproduce
1. Enable Malyan LCD in configuration.h
2. Build :)
3. connect an usb to ttl onto the Serial1 of the board
4. Monitor the output

**Expected behavior:** 

The serial should send Malyan display commands

**Actual behavior:** 

The serial continue to log the console normally (no change enabling Malyan_lcd on that port)

[Archivio.zip](https://github.com/MarlinFirmware/Marlin/files/4751301/Archivio.zip)

ellensp, 2020-06-09:
The Serial port on the TFT pins of the SKR 1.4 turbo is Serial0

```
Port    | TX    | RX
--      |   --  |   --
Serial  | P0_02 | P0_03
Serial1 | P0_15 | P0_16
Serial2 | P0_10 | P0_11
Serial3 | P0_00 | P0_01

```
Serial1 is on Exp2

adams79, 2020-06-09:
@ellensp thank you, I'll try.. however why I can't set Serial0 into the extra_maylan.cpp? Seems that this serial is not defined.. Those mapping (between pins and serial) are only available into the SanityCheck file?

ellensp, 2020-06-09:
you need the following in  extra_maylan.cpp
`#define LCD_SERIAL Serial
`
and you need the following in Configuration.h 
```
#define SERIAL_PORT -1
//#define SERIAL_PORT_2 0
```

cooldudie2, 2020-06-14:
no issue still persists for me

cooldudie2, 2020-06-14:
you cant use the value of 'Serial' for the malyan cpp file it throws this error compiling 

.pio\build\LPC1768\src\src\sd\usb_flashdrive\lib-uhs2\parsetools.cpp.o
Compiling .pio\build\LPC1768\src\src\sd\usb_flashdrive\lib-uhs2\usbhost.cpp.o
Linking .pio\build\LPC1768\firmware.elf
.pio\build\LPC1768\libFrameworkArduino.a(HardwareSerial0.cpp.o): In function `UART0_IRQHandler':
C:\Users\Tim\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial0.cpp:37: multiple definition of `UART0_IRQHandler'
.pio/build/LPC1768/src/src/HAL/LPC1768/MarlinSerial.cpp.o:C:\Users\Tim\Desktop\new skr 1.3 board for monoprice select mini/Marlin\src\HAL\LPC1768/MarlinSerial.cpp:30: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1768\firmware.elf] Error 1

cooldudie2, 2020-06-14:
im using the skr 1.3 board with same chip lpc1768 chip, and I cant get it to work either on this board. it works fine on the stm32f1 chip envrioment (on the skr 1.2 board) the screen functions as it should.

xC0000005, 2020-06-16:
I don't have an LPC board, but I would expect that changing LCD_SERIAL to MSerial would make this work, but what I've usually done is use a USB->TTL converter. When I can monitor the converter at 7N1, 500000 baud and see the Marlin serial output, I know I have the pins and Serial right.

cooldudie2, 2020-06-16:
ive got a new serial to ttl converter on order. in the meantime MSerial doesnt work either. it doesnt make sense

InsanityAutomation, 2020-06-16:
Define the DGUS serial and let it pass it through that interface. I had to do the same with the creality displays on LPC chips after the dwinos stuff got merged with mixed extui / native functions. Some conflict was brought in there but I havnt had time to figure out what.

cooldudie2, 2020-06-16:
so just to be clear so i can wrap my head around it, are you saying set the first define serial_port as -1 for the usb conection.

Then uncomment serial port 2 in the config and add the line instead

#define DGUS_SERIAL_PORT 0

then in the malyan.cpp file set it as 

#define LCD_SERIAL MSerial

if thats the case it doesnt work either

cooldudie2, 2020-06-16:
okay i have a sort of breakthough

if you define the following items the screen recognises the board and the version number v99.160.2 shows on the screen to confirm that its connected correctly. However thats it, no further comms, the screen doesn't display any numbers for thermistor values and no commands like print etc work either. if you load pronterface and use the jog controls or home button, nothing is being sent or received and you cannot view the sd card, it just hangs on please wait.

anyway to get it to even recognise the screen i connected it to the T0 R0 G on the TFT header and the 3.3v required from the swd header for power. 

then in marlin for the config file you need to set 
#define SERIAL_PORT -1
and
#define SERIAL_PORT_2 3

then in the extui_Malyan_lcd.cpp

you need to define it as       
Serial

thats it. also note the rx and tx pins seemed to also be flipped around to what i thought they should be so if it doesnt work try flipping them over. 

also the screen will display the version number for a few microseconds at startup before disapearing so it can be hard to spot. but if you press the reset button on the board it will show up and stay on the screen. 

if anyone can shed some more light as to why there is no comms working after initialization i would be greatful, thanks.

cooldudie2, 2020-06-16:
As a side note if you use the same settings i suggested,

 but change the 'Serial' in malyan cpp file 
to 'Serial1' then you can use the pins on the exp2 port as suggested by ellensp 
These pins
Serial1 | P0_15 | P0_16

the same behaviour occurs as the Tft header pins on serial 0 you can see the version but thats it.

InsanityAutomation, 2020-06-17:
While setting 
#define DGUS_SERIAL_PORT #
instead of SERIAL_PORT_2
In maylan you would set
#define LCD_SERIAL DGUS_SERIAL

cooldudie2, 2020-06-17:
okay i have tried this as well, it displays the version number the same but thats it no communication at all so the problem is still there, needs further investigation.

boelle, 2020-06-21:
@adams79 still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle