MarlinFirmware/Marlin #18241: Enhancement To Z Steppers Auto-Alignment

paulluby, 2020-06-09:
Hi Guys

Generally I use a manual 4 corner bed level on my 3d printers as I feel I can get a better level given my beds at 420 x 420mm are usually larger than the average printer. Also because I usually use multiple Z Steppers with individual Z Microswitches there is zero chance of a Nozzle striking the bed.

Recently I have been experimenting with a BLTouch and as I use usually use multiple Z-Stepper Motors I've also been using Z Stepper Auto-Alignment, all with no Z Microswitches.

This seems to work rather well however if the initial angle between the BLTouch Probe and the Nozzle is too great the the Nozzle will strike the bed before the BLTouch can be activated.

I'm not a fan of crashing my Nozzle into my bed so prior to starting a print, if one of my Z Steppers has sagged a lot, I have to adjust a Z-Stepper manually to bring my bed to a rough level so as the Nozzle doesn't strike the bed before the BLTouch can activate.

Then a thought came to mind, can we use a sensor of some sort to provide a rough bed level prior to the Z-Stepper Auto-Alignment routine being carried out therefore enhancing the function.

I design, build an program some devices for Radio Control Model Aircraft that use similar types of 3-axis Accelerometers as shown in the picture below along with an Arduino.

I think I've illustrated what I mean in the picture below but I'll try to describe what I think the enhancement should do with reference to a 2 Z Stepper set up.

On initiation of the Z Steppers Auto-Aligment function the enhancement function would look at the accelerometer vertical error and work out in which direction it has to move the Z Steppers individually to get the bed in a level position using the bed centre as the centre rotation. This level position does not have to be very accurate, it just has to be within a few degrees of level to prevent the Nozzle striking the bed when the Z Steppers Auto-Alignment is carried out. Obviously the Accelerometer would have to be mounted under the bed in some way but the additional circuitry would be minimal for the reduction in the chance of the Nozzle striking the bed when the Z Stepper Auto-Alignment function is being carried out.

Whilst I'm capable of programming a small device I design, build and program from scratch, jumping into a peice of firmware as complex as Marlin is I feel beyond my abilities at the moment.

Your thoughts on this would be appreciated as to its usefulnes.

Cheers guys.

[image]

[image]

XDA-Bam, 2020-06-09:
This looks like a very interesting idea. When using MEMS accelerometers, I see three potential problems:
1. Temporal drift: As far as I know, MEMS accs drift over time, so you need to calibrate them - maybe daily, maybe weekly. Each time you calibrate them, you need to re-level your bed manually.
2. Temperature drift: Pretty sure MEMS accs also drift with temperature. If you have a heated bed, you will probably only be able to level it with the acc when it is either hot, or cold - depending on how you initially calibrated the acc. The heating/cooling cycles of the bed could also result in the gyro being de-calibrated after each cycle (drift hysteresis - the deviation before heating up may be different from the one after cooling down again). Could be mitigated by very good insulation.
3. Level does not mean level: If you level your bed using the acc, it will be level relative to the ground/gravitational force. Your X/Y-gantry, however, may not be level to the ground. This could potantially be calibrated by leveling the bed corretly relative to the X/Y-gantry once. But if the orientation of your X/Y-gantry changes (slack in the mechanics, wear, different belt tension, etc.), you will likely have the same problem again you do have now.

I don't know if this can work correctly. Maybe. But wouldn't it be easier to just mount the BLTouch a milimeter lower and adapt your Z-offset? I'm not too familiar with the BLTouch, maybe that is not possible - let me know.

paulluby, 2020-06-10:
I'm pleased someone thinks it's an interesting idea.

I grabbed the picture of the Adafruit LIS3DH acceleromter in a rush and I agree they're not the best and have issues.

The accelerometers I like to use in applications like this are the ones below.

https://www.analog.com/media/en/technical-documentation/data-sheets/ADXL335.pdf

Drift is so insignificant with these items its not even mentioned on the datasheet. In my experience drift is more of an issue with MEMS gyros and less of an issue in MEMS accelerometers.

Also it gives out an analogue voltage, so should be easy to integrate into the firmware. Lets not mention that it uses so little power you can power it through one of the digital pins on an Arduino. See tutorial below. Also the GY-61 breakout board used in the tutorial is readily available and easy to mount.

https://www.arduino.cc/en/Tutorial/ADXL3xx

Regards level does not mean level, I agree totally with this but it could be adjusted on the mount to give you the "level" you need or the value you need to acheive to obtain "your" printers level could be set in your printers firmware. Lets remember it doesn't have to be accurate to the N th degree as all we need in my opinion is a course level, the Z Steppers Auto-Alignment routine will do the fine alignment.

The operating temp range is stated as -40 to 85 DegC so I reckon unless you put it directly on your bed it should be ok.

I may have this wrong but I expect most printers that will be using multiple Z Axis Steppers to level the bed will be of the CoreXY variety.

Because of board constraints (number of drivers, etc) I reckon most guys will be using 2 Z Steppers and Microswitch sockets could be at a premium for other things such as Filament Runout Sensing. I'm using BTT SKR V1.1 Pro and BTT GTR V1.0 with M5 V1.0 boards so resources are a constraint on one printer but not on the other.

I must admit the more I think about this the more I like it. Though not initially a BLTouch fan I do like the idea of using a single sensor such as BLTouch to level your multi stepper Z steppers  and having a method of doing that without running the risk of your nozzle striking the bed is very attractive I reckon.

As I said I only wish I had the skills to do this but diving into Marlin does hold a lot of apprehension for me.

I have a few GY-61 boards in my pile of "stuff" so I may have to hook one up at the weekend and see if it does drift noticeably and if bed temperature does affect it more than we'd like.

paulluby, 2020-06-11:
Whilst not an issue for some the bed on one of my large printers is of the floating variety and I've spent much time and effort ensuring it has a very free and easy movement.

When power is off any of the z steppers can sag dramatically and easily leave me with at least a 25mm discrepency from one side to the other.

I've tried home to Z-Max with no avail when tied in to Z Stepper Auto-Alignment.

I am contemplating trying 2209 steppers and driving it past max height till stallguard kicks in on both and then winding back up.

But to be honest the amount of time taken to do a Z Stepper Auto Alignment and especially the ABL routine required afterwards on such a large bed almost renders the whole thing a complete waste of time as I'm incurring nearly a 30 minute delay on the start of a print.

I print on 6mm toughened glass beds and the ABL Bi-Linear routine is only throwing up 0.2 mm variation. So the question comes down to do I need Z Stepper Auto-Alignment and ABL at all or should I go back to good old safe micro-switches given I'm now using GTR V1.0 boards on my printers so board resources are not and issue.

The use of an accelerometer would I reckon have made it pretty nozzle strike proof but given KISS is the principle I tend to go by should I over complicate my printers just so they look gucci?

Though I will continue to play with this idea for the time being, whether I try to impliment it and use it in anger, given my apprehension regards delving into Marlin, is a question for another time.

Marlor, 2020-06-11:
For your "Status quo" to have such ab big skew, your frame must be very crooked itself it may be better to fix this first.

You are right about the wasted ABL time, my take:
If your bed is even then Z Auto Align is enough

If your bed is very uneven (like mine) have a look at UBL
https://marlinfw.org/docs/features/unified_bed_leveling.html
one time mesh probing and storing it instead of probing before every print

paulluby, 2020-06-12:
Think there's a little confusion here, the bed on the printer in question isn't crooked nor is it bent and neither is it wavy. The frame is square using a large set square and corner to corner measurements match.

If you look at the picture of the printer below you can see the two bed stepper motor assemblies are not connected mechanically across the bed. So when power is off one side of the bed can sag dramatically.

I spent lots of time making the z movement smooth and free.

The print quality is extremely good and I only added the BLTouch just to try it out and immediately came across the issue I've described.

Lowering the BLTouch will only increase the angle than can be tolerated before the nozzle strikes the bed during Z Steppers Auto-Alignment if the bed has sagged at one side quite dramatically. Whereas an accelerometer operating in the manner I described earlier should in theory alleviate the nozzle strike issue.

I was hoping given my previously described apprehension at diving into firmware as complicated as Marlin some assistance would be offered to help me at least understand the commands within the routines required to move the Z Stepers.

But the more I look the more I am thinking going back to microswitches is the way to go.

[image]

pszeder, 2020-06-15:
First of all, your printer looks cool! 
I'm not familiar with the code, so just talking about the theory: 

Could you put something to stop the Z from going into heights that would mean that the bed is at an angle that could cause collisions? The process would be: move ahead to the middle, elevate the bed evenly until stallguard kicks in on either side or BLTouch triggers. If the stallguard kicked in, then elevate the other side until BL touch triggers, then do the normal auto align.
So if in your case the nozzle hits the bed when there is 15mm difference in Z, then I would put two blocks to stop the Z sides lets say 5mm above the nozzle. So it does not interfere with the auto-alignment at all and would not be needed to be changed if you change nozzles.

paulluby, 2020-06-21:
Hi guys

To stop the sagging I've modified the bed on the 3d printer with the issue so its now a rigid type bed frame. All adds weight so not my first choice but there you go as the sagging is no longer an issue. Pictures of modded frame and printer below.

During the problem solving process I've tried using the BLTouch and Bi-Linear ABL alongside both Z-Min microswitches. This was a complete failure as the Bi-Linear ABL routine would start and freeze at the first probe point. Tried homing to Z-Max with same issue. Tried using 2209's with Stallguard triggering at both Z-Min and Z-Max and same problem occured. Have scoured code and no idea why this is happening. Is there an issue with ABL and multiple Z microswitches that I'm not aware of?

Getting rid of microswitches and using Z Stepper Auto-Align and Bi-Linear ABL all works fine and once the bed frame was modded to rigid type the sagging issue was negated.

Still think the Accelerometer idea is a good one and when I feel like expanding my code knowledge I'll have a go at it.

[image]
[image]

sjasonsmith, 2020-07-20:
@paulluby if your G34 is taking 30 minutes, you need to adjust the amplification factor so that it accounts for the distance between your screws and your probe points. If well-tuned I imagine you could get a good G34 alignment in one or two iterations.