MarlinFirmware/Marlin #18244: (Did you forget to enable DISTINCT_E_FACTORS?)[BUG] (short description)

iamthebest22, 2020-06-09:
Tried the newest bugfix from last week and still have the same issue.

My configuration.h AND configuration_adv.h is attached below in the zip file.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4754342/Marlin.zip)

### Bug Description

I'm building a cheap 3D printer for a friend that wants a portable one, it's using a MKS Sbase 1.3 and I'm basically going to use one E3D Titan, and a Cr-10 style MK8 extruders I have extra laying around for a one hotend only, but a splitter for a non mixing hotend for dual color printing. 

The problem is, even with #define DISTINCT_E_FACTORS enabled, I get this issue 

In file included from Marlin\src\HAL\LPC1768\../../core/../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:26:
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:2434:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2434 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:2441:35: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2441 | static_assert(COUNT(sanity_arr_2) <= XYZE_N, "DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:25:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2434:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2434 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2441:35: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2441 | static_assert(COUNT(sanity_arr_2) <= XYZE_N, "DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2434:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2434 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2441:35: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2441 | static_assert(COUNT(sanity_arr_2) <= XYZE_N, "DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2434:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2434 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2441:35: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2441 | static_assert(COUNT(sanity_arr_2) <= XYZE_N, "DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\eeprom_sdcard.cpp:25:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2434:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2434 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2441:35: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2441 | static_assert(COUNT(sanity_arr_2) <= XYZE_N, "DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_flash.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o] Error 1
In file included from Marlin\src\HAL\LPC1768\../../sd/../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\../../sd/cardreader.h:24,
                 from Marlin\src\HAL\LPC1768\main.cpp:38:
Marlin\src\HAL\LPC1768\../../sd/../inc/SanityCheck.h:2434:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2434 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
Marlin\src\HAL\LPC1768\../../sd/../inc/SanityCheck.h:2441:35: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2441 | static_assert(COUNT(sanity_arr_2) <= XYZE_N, "DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\main.cpp.o] Error 1
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\timers.cpp:31:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2434:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2434 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2441:35: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2441 | static_assert(COUNT(sanity_arr_2) <= XYZE_N, "DEFAULT_MAX_FEEDRATE has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\timers.cpp.o] Error 1

### My Configurations

Here just in case again
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4754345/Marlin.zip)

### Steps to Reproduce

configured the settings to make sure define distinct_E_factors is enabled, but still gives that bug

**Expected behavior:** [What you expect to happen]
Expected to build/compile with no problems
**Actual behavior:** [What actually happens]
The above error happens
#### Additional Information
This is the non bugfix 2.0.5.3, BUT I have tried it with the bugfix version downloaded from last week with the same result.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-06-10:
The configurations you attached are not the ones being used by marlin.
The Configuration files must bin in Marlin/Marlin directory. same place as the Marlin.ino file
Not in Marlin/config directory

iamthebest22, 2020-06-10:
Huh? I just checked, the files I provided are from the Marlin/Marlin directory, same place as the Marlin.ino file, I'm confused. 

Here are the pictures for proof, I zipped those two files and you can see the zip there.
[image]
[image]

ellensp, 2020-06-10:
You have #define EXTRUDERS 1 when you have DISTINCT_E_FACTORS this is in conflict.

You also need #define SINGLENOZZLE  for a single hotend with > 1 extruder

iamthebest22, 2020-06-10:
ah that fixed it, I rely on error messages alot to solve config issues, so that didn't pop up and I was confused, thank you very much! Lesson learned!

closed 2020-06-10 by iamthebest22