MarlinFirmware/Marlin #18247: Add M114 polling to M155 command function

G2Barbour, 2020-06-10:
### Description

I have just set up firmware that allows M600 to be used without marlin Mode, and unfortunately a new but separate quirk of the printing process has reared its head. In order for the TFT to keep track of XYZ positions, I need them to be auto-reported while printing from Octoprint otherwise I have to interrupt the active serial port with the TFT commanding M114. A 'P' parameter for position in the M155 command should do, as shown below:

//M155.cpp

void GcodeSuite::M155() {

  if (parser.seen('P')){//automatically report_current_position_projected(); /*new*/
    gcode.autoreport_position = true; 
  }
  else gcode.autoreport_position = false; 

  if (parser.seenval('S'))
    thermalManager.set_auto_report_interval(parser.value_byte());

}

//temperature.cpp

    void Temperature::auto_report_temperatures() {
      if (auto_report_temp_interval && ELAPSED(millis(), next_temp_report_ms)) {
        next_temp_report_ms = millis() + 1000UL * auto_report_temp_interval;
        PORT_REDIRECT(SERIAL_BOTH);
        print_heater_states(active_extruder);
        if(gcode.autoreport_position){ SERIAL_CHAR(' '); report_current_position_projected(); } /*new*/
        else SERIAL_EOL();
      }
    }

//gcode.cpp
#if HAS_AUTO_REPORTING || ENABLED(HOST_KEEPALIVE_FEATURE)
  bool GcodeSuite::autoreport_paused; // = false
  bool GcodeSuite::autoreport_position; // = false /*new*/
#endif

//gcode.h
  #if HAS_AUTO_REPORTING || ENABLED(HOST_KEEPALIVE_FEATURE)
    static bool autoreport_position;  /*new*/
    static bool autoreport_paused;
    static inline bool set_autoreport_paused(const bool p) {
      const bool was = autoreport_paused;
      autoreport_paused = p;
      return was;
    }

### Feature Workflow

1. M155 S1 ;Normal temperature reporting every 1 sec
2. M155 S1 P ;Temperature AND position reported every 1 sec
3. [and so on...]

#### Additional Information
This code has been tested working. I'm new to Marlin and not that familiar with your coding conventions, so you may want to clean it up for your liking should you choose to adopt it.

shitcreek, 2020-06-26:
This is a good idea. I've created a PR for it:
https://github.com/MarlinFirmware/Marlin/pull/18427
Thanks.

thisiskeithb, 2021-05-15:
https://github.com/MarlinFirmware/Marlin/pull/18427 was merged. Closing.

closed 2021-05-15 by thisiskeithb