MarlinFirmware/Marlin #18248: EEPROM Index error on Marlin-bugfix-2.0.x dated 08062020

SteenAndreassen, 2020-06-10:
Setup: Prusa MK3S clone with Bigtree SKR 1.3 board (LPC1768)
Compiled on: Vs Code with Platform IO

Everything working as expected in release 2.0.5.3 except the new feature G76 that I need due to major problems adjusting my Z-offset due to temperature variations on the Pinda V2 probe

BTW!! This new, experimental, Pinda Temperature compensation is, in my opinion, one of the most important features that have been added in the past.
Looking on the major temp errors both I and other on the internet get from the G76 command is showing, that this feature is highly necessary for a useful 1.st layer 

But in hotfix 2.0.x downloaded 8'th of June 2020, I get a EEPROM Index Error when I upgrade my board, and this error can't be removed issuing the normal M502 followed by a M500
Every time I reset or power on  the printer, I get this EEPROM Index error on the LCD 
Even choosing the Reset option from the error 

screen on the LCD do not fix this issue

Please find config files from both releases attached

Br Steen Andreassen
Denmark

[Marlin-Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4757549/Marlin-Configs.zip)

XDA-Bam, 2020-06-10:
This should already be fixed by #18234. Please test today's bugfix and close this issue, if the error is gone.

SteenAndreassen, 2020-06-10:
I can confirm, that this EEPROM Index error is fixed in Hotfix-2.0.x dated 10-06-2020

closed 2020-06-10 by SteenAndreassen