MarlinFirmware/Marlin #18257: [BUG] Error compiling Bugfix-2.0.x

frank-pian, 2020-06-11:
### Bug Description

Board: BIGTREE_SKR_PRO（STM32 F407）

The code is normal for 2020-05-07.There was a compilation error when I updated to the latest version.
I think this commit may have led to a dependency error?
 [8f67451](https://github.com/MarlinFirmware/Marlin/commit/33d1e77e2e252f3a3825941630718ea7d8f67451)

### My Configurations

[Configure.h](https://drive.google.com/file/d/19nu-SL1qwPyRcAHeAmf2gUdP5KDefNUJ/view?usp=sharing)
[Configuration_adv.h](https://drive.google.com/file/d/1rjHI5c2BgwgxYlvaTuSKDReZEXewtT-5/view?usp=sharing)

### Steps to Reproduce

1. Build

#### Additional Information

In file included from Marlin\src\module\../inc/MarlinConfig.h:33:0,
                 from Marlin\src\module\stepper.h:44,
                 from Marlin\src\module\stepper.cpp:80:
Marlin\src\module\stepper.cpp: In static member function 'static void Stepper::do_babystep(AxisEnum, bool)':
Marlin\src\module\../inc/../HAL/STM32/timers.h:50:34: error: 'HAL_TIMER_RATE' was not declared in this scope
 #define STEPPER_TIMER_PRESCALE ((HAL_TIMER_RATE)/(STEPPER_TIMER_RATE))
                                  ^
Marlin\src\module\../inc/../HAL/STM32/timers.h:54:30: note: in expansion of macro 'STEPPER_TIMER_PRESCALE'
 #define PULSE_TIMER_PRESCALE STEPPER_TIMER_PRESCALE
                              ^~~~~~~~~~~~~~~~~~~~~~
Marlin\src\module\stepper.cpp:2766:124: note: in expansion of macro 'PULSE_TIMER_PRESCALE'
     #define _PULSE_WAIT() while (EXTRA_CYCLES_BABYSTEP > (uint32_t)(HAL_timer_get_count(PULSE_TIMER_NUM) - pulse_start) * (PULSE_TIMER_PRESCALE)) { /* nada */ }
                                                                                                                            ^~~~~~~~~~~~~~~~~~~~
Marlin\src\module\stepper.cpp:2894:11: note: in expansion of macro '_PULSE_WAIT'
           _PULSE_WAIT();
           ^~~~~~~~~~~
Marlin\src\module\../inc/../HAL/STM32/timers.h:50:34: note: suggested alternative: 'PULSE_TIMER_RATE'
 #define STEPPER_TIMER_PRESCALE ((HAL_TIMER_RATE)/(STEPPER_TIMER_RATE))
                                  ^
Marlin\src\module\../inc/../HAL/STM32/timers.h:54:30: note: in expansion of macro 'STEPPER_TIMER_PRESCALE'
 #define PULSE_TIMER_PRESCALE STEPPER_TIMER_PRESCALE
                              ^~~~~~~~~~~~~~~~~~~~~~
Marlin\src\module\stepper.cpp:2766:124: note: in expansion of macro 'PULSE_TIMER_PRESCALE'
     #define _PULSE_WAIT() while (EXTRA_CYCLES_BABYSTEP > (uint32_t)(HAL_timer_get_count(PULSE_TIMER_NUM) - pulse_start) * (PULSE_TIMER_PRESCALE)) { /* nada */ }
                                                                                                                            ^~~~~~~~~~~~~~~~~~~~
Marlin\src\module\stepper.cpp:2894:11: note: in expansion of macro '_PULSE_WAIT'
           _PULSE_WAIT();
           ^~~~~~~~~~~
*** [.pio\build\BIGTREE_SKR_PRO\src\src\module\stepper.cpp.o] Error 1

sjasonsmith, 2020-06-20:
@frank-pian, I can't get things into the same state because it seems you have probably modified other files as well?

Could you reproduce this in the bugfix-2.0.x branch, then post a zip file including all the files you have modified? I suspect you have probably modified pins and maybe even sanity check files.

frank-pian, 2020-06-24:
@sjasonsmith 
you need open BABYSTEPPING in 'Configuration_adv.h'.

sjasonsmith, 2020-06-24:
Please provide me updated files to reproduce with bugfix-2.0.x.

thinkyhead, 2020-06-27:
The `bugfix-2.0.x` branch compiles perfectly with the supplied configurations (once SPINDLE_LASER pins are defined).

closed 2020-06-27 by thinkyhead